motioneye-project/motioneye #2329: Eufy camera not working

fxtv1060, 2022-03-11:
I installed ME on a Cubox i running Armbian (Bulleye) When trying to connect one of my Eufy 2 cameras here is what I get: 

[image]

I got the info from the Eufy app:

[image]

Any advice?

François-Xavier

starbasessd, 2022-03-11:
If you use the http:// URL try using 
http://192.168.1.9:554
If you want to use the RTSP feed, use
rtsp://192.168.1.9/live1
both with username: admin
and password: admin
Let us know...

fxtv1060, 2022-03-11:
Thanks! The RSTP option works. Great!

fx

starbasessd, 2022-03-11:
Don't forget to close this.

closed 2022-03-11 by fxtv1060