motioneye-project/motioneye #2331: No image

fxtv1060, 2022-03-11:
I have got two Eufy 2 cameras connected thru rtsp to ME. Here is the screen I get via the brower:

[image]

Most probably I did not configure ME the right way. Help!

François-Xavier

starbasessd, 2022-03-11:
I can't read the message in the top left of the 2 views, what does it say?

fxtv1060, 2022-03-11:
It says: Unable to open video device since 2022-03-11.

fx

starbasessd, 2022-03-11:
what happens when you reboot it?

fxtv1060, 2022-03-11:
Nothing. Unfortunately.

fx

starbasessd, 2022-03-11:
I'm going to leave this here, maybe one of the new developers can help, if they know Armbian.

fxtv1060, 2022-03-11:
The camera are not streaming continuously. They are shooting only in cas of an "event".

starbasessd, 2022-03-11:
One other option, though I can suggest:
delete the cameras, then re-add them, but in the bottom Camera box, select RTSP/UDP instead of RTSP/TCP

fxtv1060, 2022-03-11:
Now there is a redish frame around the grey picture...

[image]

fx

MichaIng, 2022-03-11:
Armbian is basically Debian, just with a custom, mostly mainline kernel to run on ARM SBCs. I'll try to find a public RTSP stream to generally test it. To narrow it down to whether the motionEye or its host is the issue, or the camera/RTSP stream itself, could you test it elsewhere, e.g. via VLC? https://support.ipconfigure.com/hc/en-us/articles/115005588503-Using-VLC-to-test-camera-stream

fxtv1060, 2022-03-11:
SOmetime is works sometime not...

[image]

MichaIng, 2022-03-11:
Something I recognised is that the browser energy savings (automatically setting "unused" tabs to sleep and such) and pause the stream. I use Opera with energy savings enabled, and when switching tabs back and forth, the camera image is at first missing and then reappears after 1-3 seconds, like if the RTSP stream was stopped and re-established, which somehow even makes sense.

Please test in VLC player whether the stream is reliable and stable there. Probably we can tell the browser via modern browser APIs to keep the stream or tab active 🤔.

And of course, checking logs is always a good idea:
```sh
journalctl -u motioneye
```

fxtv1060, 2022-03-11:
Many thanks. I will try what you suggest affer the kids' dinner...
fx

fxtv1060, 2022-03-12:
Many thanks for your help. 

What does this mean: "With motion v4.3 one shouldn't face any errors, i.e. on Debian Bullseye or when pulling related older releases from motion GitHub repo" ?

fx

MichaIng, 2022-03-12:
> What does this mean: "With motion v4.3 one shouldn't face any errors, i.e. on Debian Bullseye or when pulling related older releases from motion GitHub repo" ?

Ah sorry, I forgot to remove that when shifting the find to a new issue: #2333
However, just to rule it out, you could try to use motion v4.3 which is best supported by motionEye currently, regarding the default motion configs it applies.

I had the RSTP stream running here for a while, shutdown the machine over night, now booted and check back again, and the stream usually works fine. Indeed sometimes on first motionEye UI access I saw it grey (inaccessible) or green (some intermediate state, I guess) for a short while, but it fixed itself quickly, so seems to be due to an intermediate network issue or timeout when motionEye tries to connect the first time.

closed 2022-03-12 by fxtv1060