motioneye-project/motioneye #2333: Support multiple motion versions

MichaIng, 2022-03-11:
I tested motionEye on Debian Buster with native repo `motion` v4.1, and a bunch of settings motionEye applies are not supported (yet):
```
Mar 11 18:55:33 VM-Buster motion[3878]: [-1660513328:motion] [NTC] [ALL] config_camera: Processing camera config file camera-1.conf
Mar 11 18:55:33 VM-Buster motion[3878]: [-1660513328:motion] [ALR] [ALL] conf_cmdparse: Unknown config option "threshold_maximum"
Mar 11 18:55:33 VM-Buster motion[3878]: [-1660513328:motion] [ALR] [ALL] conf_cmdparse: Unknown config option "movie_passthrough"
Mar 11 18:55:33 VM-Buster motion[3878]: [-1660513328:motion] [ALR] [ALL] conf_cmdparse: Unknown config option "vid_control_params"
Mar 11 18:55:33 VM-Buster motion[3878]: [-1660513328:motion] [NTC] [ALL] config_camera: Processing camera config file camera-2.conf
Mar 11 18:55:33 VM-Buster motion[3878]: [-1660513328:motion] [ALR] [ALL] conf_cmdparse: Unknown config option "threshold_maximum"
Mar 11 18:55:33 VM-Buster motion[3878]: [-1660513328:motion] [ALR] [ALL] conf_cmdparse: Unknown config option "movie_passthrough"
```
And those options indeed have changed from v4.1 to v4.3 and have changed again with v4.4: https://motion-project.github.io/motion_config.html#configfiles

I would actually like to add support for a range of motion versions instead of a single one only. We have two options:
1. Detect the motion version and set the respective setting(s).
2. Set all settings and leave motion ignore the invalid ones.

The first option is of course more elegant and avoids these nasty errors messages, but it is more difficult to apply. The second option is simpler to implement, but unclean.

I'd of course prefer method 1. We could add arrays which map motion versions to config keys. But it is not so trivial to reliably get the motion version, at least I didn't fine one so far. Only way was to parse `motion -h` output:
```sh
motion -h | awk -F'[ ,]' 'NR==1{print $3}'
```

zagrim, 2022-03-12:
My 2 cents:
It might not be possible to get very good UX with backward compatibility unless also the UI was modified to only show options that are available with the installed version of Motion. E.g. movie passthrough mode (`movie_passthrough`) and maximum change threshold (`threshold_maximum`) seem like they were not available at all in 4.1, even though I'm not quite sure if passthrough was somehow supported in 4.1 or not. To implement that, there would need to be digging around the history of Motion option docs though.

But maybe it is not necessary to rebuild 4.1 support at this point? Starting support from 4.2 would require less redoing of stuff that has already been removed/changed for newer versions.I have no idea how common 4.1 still is, though.

I do agree that having wider supported version range of Motion would reduce bug reports somewhat, especially if the support for new versions would get implemented early.

Regarding the options, I'd favour method 1, too, if somehow possible. It would also allow having more complicated intra-version mapping than just having the option key renamed (if that ever is needed). I fear method 2 would cause issues at some point and at least make debugging of the configuration more tedious.

MichaIng, 2022-03-12:
Not sure whether it makes sense to add support for older `motion` config keys, but for newer ones at least. And you are right, when settings are not only renamed but added or removed (and we had or want dedicated options for them in UI), then the UI needs to be made version-specific as well.

MichaIng, 2022-03-12:
Nice, there is some pre/post v4.2 motion config mapping done already: https://github.com/motioneye-project/motioneye/blob/c58ad0f/motioneye/config.py#L138-L167

We can build on this and add v4.4 support just the same way 👍.

fxtv1060, 2022-03-12:
I am currently running motion 4.3.2. Would you please help me going back to 4.3?

fx

MichaIng, 2022-03-12:
v4.3.2 is v4.3 😄. The patch version (last digit) never includes any breaking changes or renamed settings.

fxtv1060, 2022-03-12:
Oooups. So what does "However, just to rule it out, **you could try to use motion v4.3** which is best supported by motionEye currently, regarding the default motion configs it applies" mean??!

fx

MichaIng, 2022-03-12:
This was just in case you use a different motion version, e.g. the newest v4.4 available upstream or the older v4.1 shipped by Debian Buster and alike. Since you are on v4.3 already, this is not related to your issue. As I commented in the other topic, I found similar symptoms by times, on initial IP camera stream connection, so bet it is due to temporary or initial network/connection issues which sort themselves within some seconds or on page reload.

zagrim, 2022-03-12:
@MichaIng I didn't remember there were such intra-version mappings. And there's actually some sort of motion version detection in https://github.com/motioneye-project/motioneye/blob/c58ad0fccb063183f454435632b2423d9348a27d/motioneye/motionctl.py#L340

MichaIng, 2022-03-12:
> And there's actually some sort of motion version detection in

Otherwise the mapping wouldn't be possible. It uses exactly the `motion -h` parsing I found as a way to obtain the version 😄. Though it does it differently, not via `awk` but via native Python string parsing, which makes sense.

I'm currently going through all the open PRs, but afterwards I mind find time to add motion v4.4 support properly with the existing methods.

MichaIng, 2022-05-03:
PR up but I need help to finish it: #2462

zagrim, 2023-12-07:
Since https://github.com/motioneye-project/motioneye/pull/2462 has already been merged this issue should've been fixed long time ago, so closing this. The PR mentions one issue with changed Motion default behaviour in some case with MJPEG netcams (if I understood correctly) but I don't think that should be a reason to keep also this one open.

closed 2023-12-07 by zagrim