motioneye-project/motioneye #2336: Cannot mount network share

jtmpush18, 2022-03-11:
Hello.
I'm having an issue with writing to my samba network share.
Does anyone understand these error messages:
[image]

It's somewhat of an unusual setup:
motionEye Version: 0.42.1
Motion Version: 4.2.2
OS Version Ubuntu 20.04

but, it's not really Ubuntu. It's really Ubuntu on a X86 TrueNAS server (Core not Scale) in a virtual machine. Motioneye was installed properly without any error messages, and I can get in the web interface without anby issues whatsoever.

MichaIng, 2022-03-11:
Duck has lots of finds but difficult to get the essence of the issue: https://duckduckgo.com/?q=SMB+signiture+verification+error&t=opera&ia=web

It seems to try authenticating as "guest". Did you verify user/password and that the SMB protocol version matches the server?

JamBalaya56562, 2026-11-10:
Closing, as this got no reply to MichaIng's questions and was with motionEye 0.42.1. The log shows the mount trying to log in as guest, so the share's username and password and an SMB version the server accepts are the first things to check. Please open a new issue if mounting still fails with a current release.

closed 2026-11-10 by JamBalaya56562