motioneye-project/motioneye #2338: How do I set the time and date (internal network)

Badwolf15002, 2022-03-12:
Hey so I have the Motioneye OS on a raspberry Pi 4 model B. 

My computer has 2 Ethernet ports. I have my main one connected to the internet, and I have the raspberry Pi connected to the other port. I have set that port as 192.168.2.2 as a separate network from the rest of my internet. But because of this, the Motioneye Os can't get internet access or NTP data to set the correct time and dates.

I have tried pointing the NTP server to my computer, even http but i must be doing something wrong for it cant get the data it wants.

I'm running Windows 10 desktop computer.

MichaIng, 2022-03-12:
I don't now motionEyeOS, but generally you either need internet access from that device or access to the router (which often can serve as NTP server as well).

Another option is to install chrony on the RPi and another reachable host, since multiple chrony instances can synchronise with each other.

Badwolf15002, 2022-03-12:
So after trying everything I found my solution. Its a simple one and I feel stupid for it. 

If you has a Raspberry Pi with Motioneye Os on it and you want to use it of grid (Not connected to the internet) but you still want the correct date & time on the video feed. 

Simply set a static Ip address on it but using your home network address. Example: Home network - 192.168.1.xxx set your Pi to 192.168.1.xxx 

Once the static Ip address is set simply remove the ethernet cable and connect it to different computer or device you can access the Pi from but not on your home network. (DO NOT unplug the power to the Pi otherwise it will attempt to gather new NTP data when it doesn't have internet access and use its default time and date info) 

Once you connect it to the new device (make sure the Nic on that device is set to the same network (I think at least I set it ahead of time not wanting to mess with it further. So who knows leaving the Nic to be assigned an Ip automatically may work still )) 

Once you connect to the Pi using that computer or device that is also off grid (or if you have multiple ethernet ports) you will see that the date and time are current. Just if you loose power again, temporally connect it to the internet and just swap it over to the off grid LAN

closed 2022-03-12 by Badwolf15002

MichaIng, 2022-03-12:
Doesn't sound so simple when you need to plug back and forth the Pi 😄. Then I'd prefer to manually set the system clock on the console via `date -s '2022-03-12 13:50:00'` (system clock) and `hwclock -w` (sync system clock to hardware clock), when required. But it depends on individual usage, of course, so I'm happy you found your solution.

Badwolf15002, 2022-03-12:
> Doesn't sound so simple when you need to plug back and forth the Pi 😄. Then I'd prefer to manually set the system clock on the console via `date -s '2022-03-12 13:50:00'` (system clock) and `hwclock [...]

You can do that with Motioneye OS?
And its not that difficult. Just connect it to the internet, let it get its data, then plug it into the offline LAN. As long as it has power it will keep the date and time

MichaIng, 2022-03-12:
`date` and `hwclock` are both very basic UNIX/Linux commands, so I'm pretty sure they are available on motionEyeOS as well.

If you have physical access to the Pi and cables for Internet access nearby, then the method is fine, but I personally would find it annoying when having to plug it back and forth on every boot 😉.