motioneye-project/motioneye #2357: Is there a point to inviting me back?

starbasessd, 2022-03-14:
I am not sure about what is wanted, what is expected, etc., for your clique.

MichaIng, 2022-03-15:
Sorry, I recognised after the invitation that you left the orga intentionally. One can be orga member without being repo collaborator and the other way round,

Btw, I read your posts from the other pull request from mail notifications but wasn't sure whether to react since you removed them. Would be still interesting to understand the issue you had/have, making you leave. To me it seemed like a misunderstanding of what the goal of the individual pull request and GitHub Actions workflow was, probably additionally confusing that we posted on two different pull requests concurrently? Probably we need/can change something in our communication or focus?

Generally you are very welcome to contribute to the future of motionEye in any way you can and want 🙂.

starbasessd, 2022-03-15:
I thought I made it abundantly clear when this new group took over as to my skill set and skill level as to testing. I am not a dev. I am not a programmer.
When I was asked to test things I had no clue about, and stated so, I was not given any directions I could follow. 
I am an experienced beta tester from all the way back to the 80's for everything from Zoom Modems to Diesel engine controllers, to internal and external software. In every case, because each environment is different, I was given a list of what was needed (software and hardware environment), and what parameters were to be tested. 
In this case, when I asked about environments I was told to use an internal github environment. 
For what should have been tested, it was working in those isolated environments, but not in the real world. When I brought it up, I was told it wasn't important.  
Yet, when I asked for some things, I was poo-poo'd about having double effort to fix those things. IOW, never enough time to do it right, but always enough time to go back and fix it (related to python3, but not for python2) 
In internal corporate environments, when helping Dev-Ops build new software, the first thing they do is ask us, is it working as expected in your real world environment. Why? Because Devs may have packages installed in their environments, that the average user / customer doesn't, and although it works perfectly for them, it doesn't really work unless they remove that package dependency, or tell the end user to include it. 
That is why I went off about the Dev mentality most, if not all, the devs here have. And I got tired of tilting at windmills. (Don Quixote reference).

MichaIng, 2022-03-15:
> When I brought it up, I was told it wasn't important.

This is probably the misunderstanding I meant. The two pull requests we were discussion were not about real world tests but about automated build and install tests in a fixed environment, not replacing real world tests, but only for detecting rough issues and things like missing or added dependencies, before changes are merged to `dev`. With #2353 I updated the new [motioneye_init](https://github.com/motioneye-project/motioneye/blob/dev/motioneye/extra/linux_init) command which aims to setup motionEye, once installed, in a streamlined way that works on most Linux distros.

Real world tests are of course always important, so having proper install instructions to use for real world testing is indeed important and something we should do next. The GitHub Actions workflows already contain working install instructions, though probably not containing all dependencies for all features yet, which we should add to `motioneye_init` then, in case.

If you are willing to rejoin or otherwise do testing of the `dev` branch, here are install instructions for Debian/Ubuntu (APT based distros):
```sh
sudo apt update
sudo apt install curl python3-dev libcurl4-openssl-dev gcc libssl-dev
curl -O 'https://bootstrap.pypa.io/get-pip.py'
sudo python3 get-pip.py
rm get-pip.py
sudo pip3 install 'https://github.com/motioneye-project/motioneye/archive/dev.tar.gz'
sudo motioneye_init
```
Makes sense to use those in our Ubuntu install test action and add them to the Wiki for testers as well. This way of installing `pip` works on all distros and gets latest `pip` with the required `setuptools` and `wheel` modules. Only installing Python and the development headers of course depends on the package manager, like `yum` on RPM based distros or `pacman` on Arch Linux, and how especially the development headers packages are called.

The last command installs and starts the motionEye service, so after some seconds startup time the web UI should be reachable. Logs can be checked via `journalctl -u motioneye`.

To update motionEye to the lastest commit in `dev` (as long as the version string isn't incremented):
```sh
pip3 install --force-reinstall --no-deps 'https://github.com/motioneye-project/motioneye/archive/dev.tar.gz'
systemctl restart motioneye
```

cclauss, 2022-03-15:
```sh
curl -O 'https://bootstrap.pypa.io/get-pip.py'
sudo python3 get-pip.py
rm get-pip.py
# —>
sudo apt install curl python3-pip
```

starbasessd, 2022-03-15:
Another issue I have is burying requests for changes and requests for testing and so forth into conversation threads. If something like this occurs (and the reverse), they should be broken out, and linked **_if necessary_**. This conversation had little to nothing to do with the current state of the install instructions, yet some felt it necessary to include [above](https://github.com/motioneye-project/motioneye/issues/2357#issuecomment-1067471957) and [above again](https://github.com/motioneye-project/motioneye/issues/2357#issuecomment-1067610149) in this thread, which, at a minimum should be 2 other threads, one discussing including a 3rd party 'blob' for pip3 vs the repo version for specific distros (wasn't one of the 'goals' to lessen such things?) and the other of testing the current state of installation instructions, IMO.

MichaIng, 2022-03-15:
Yes you are right, I tend to mix or widen topics of conversations, which is not always good. I thought from your post that not having clear install instructions for real world testing was one of your major issue. I'll split off the `pip` install method into a PR now.

> including a 3rd party 'blob' for pip3

`get-pip.py` is the official upstream install method, `python3-pip` is the 3rd party/distro/downstream method 😉: https://pip.pypa.io/en/stable/installation/

However, let me explain/suggest this in a PR.

starbasessd, 2022-03-15:
Nope, there wasn't.

closed 2022-03-15 by starbasessd

starbasessd, 2022-03-15:
This issue, meant for conversation, got hijacked for other things, so I closed it. If there are issues of the software brought up that should still be 'open', then as someone said, open a new issue or PR. I'm not staying involved because of things like this. I can go off on a tangent as well as the best of them, but it's usually because there aren't any decent forums to hold conversations, and before, after Calin dropped support, it was me and 2 programmers and whoever we (I) was conversing with. <sigh>

MichaIng, 2022-03-15:
> This issue, meant for conversation, got hijacked for other things

You said before:

> When I was asked to test things I had no clue about, and stated so, I was not given any directions I could follow.

so I was thinking to solve this right here... If my attempt to help you with the issue you named as reason to leave is causing you so much trouble, because it was not done in a dedicated issue, then I'm sorry. Looks like any attempt I do to make you stay in the team triggers only another problem. There seem to be more things which make you feel not well with us, but I will now give up trying to understand them, since this is causing me bad mood now, which once should avoid as space time volunteer in open source projects 😞.