motioneye-project/motioneye #2359: Issues with new install

starbasessd, 2022-03-15:
If you are willing to rejoin or otherwise do testing of the `dev` branch, here are install instructions for Debian/Ubuntu (APT based distros):
>sh
sudo apt update
sudo apt install curl python3-dev libcurl4-openssl-dev gcc libssl-dev
curl -O 'https://bootstrap.pypa.io/get-pip.py'
sudo python3 get-pip.py
rm get-pip.py
sudo pip3 install 'https://github.com/motioneye-project/motioneye/archive/dev.tar.gz'
sudo motioneye_init
- - - - -
gave me:

sudo motioneye_init
Hit:1 http://deb.debian.org/debian bullseye InRelease
Hit:2 http://security.debian.org/debian-security bullseye-security InRelease
Hit:3 http://deb.debian.org/debian bullseye-updates InRelease
Reading package lists... Done
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
curl is already the newest version (7.74.0-1.3+deb11u1).
The following additional packages will be installed:
  alsa-topology-conf alsa-ucm-conf fontconfig i965-va-driver intel-media-va-driver libaacs0 libasound2 libasound2-data
  libass9 libasyncns0 libavc1394-0 libavcodec58 libavdevice58 libavfilter7 libavformat58 libavresample4 libavutil56
  libbdplus0 libblas3 libbluray2 libbs2b0 libcaca0 libcairo-gobject2 libcairo2 libcdio-cdda2 libcdio-paranoia2
  libcdio19 libchromaprint1 libcodec2-0.9 libdatrie1 libdc1394-25 libdrm-amdgpu1 libdrm-common libdrm-intel1
  libdrm-nouveau2 libdrm-radeon1 libdrm2 libflac8 libflite1 libfribidi0 libgbm1 libgdk-pixbuf-2.0-0
  libgdk-pixbuf2.0-bin libgdk-pixbuf2.0-common libgfortran5 libgl1 libgl1-mesa-dri libglapi-mesa libglvnd0
  libglx-mesa0 libglx0 libgme0 libgraphite2-3 libgsm1 libharfbuzz0b libiec61883-0 libigdgmm11 libjack-jackd2-0
  liblapack3 liblilv-0-0 libllvm11 libmariadb3 libmfx1 libmicrohttpd12 libmp3lame0 libmpg123-0 libmysofa1 libnorm1
  libogg0 libopenal-data libopenal1 libopenmpt0 libopus0 libpango-1.0-0 libpangocairo-1.0-0 libpangoft2-1.0-0
  libpciaccess0 libpgm-5.3-0 libpixman-1-0 libpocketsphinx3 libpostproc55 libpq5 libpulse0 librabbitmq4 libraw1394-11
  librsvg2-2 librsvg2-common librubberband2 libsamplerate0 libsdl2-2.0-0 libsensors-config libsensors5 libserd-0-0
  libshine3 libsnappy1v5 libsndfile1 libsndio7.0 libsodium23 libsord-0-0 libsoxr0 libspeex1 libsphinxbase3
  libsratom-0-0 libsrt1.4-gnutls libssh-gcrypt-4 libswresample3 libswscale5 libthai-data libthai0 libtheora0
  libtwolame0 libudfread0 libv4l-0 libv4l2rds0 libv4lconvert0 libva-drm2 libva-x11-2 libva2 libvdpau-va-gl1 libvdpau1
  libvidstab1.1 libvorbis0a libvorbisenc2 libvorbisfile3 libvpx6 libvulkan1 libwavpack1 libwayland-client0
  libwayland-cursor0 libwayland-egl1 libwayland-server0 libx11-xcb1 libx264-160 libxcb-dri2-0 libxcb-dri3-0
  libxcb-glx0 libxcb-present0 libxcb-randr0 libxcb-render0 libxcb-shape0 libxcb-shm0 libxcb-sync1 libxcb-xfixes0
  libxcursor1 libxdamage1 libxfixes3 libxi6 libxinerama1 libxkbcommon0 libxrandr2 libxrender1 libxshmfence1 libxss1
  libxv1 libxvidcore4 libxxf86vm1 libz3-4 libzmq5 libzvbi-common libzvbi0 mariadb-common mesa-va-drivers
  mesa-vdpau-drivers mesa-vulkan-drivers mysql-common ocl-icd-libopencl1 pocketsphinx-en-us python3-botocore
  python3-dateutil python3-jmespath python3-s3transfer va-driver-all vdpau-driver-all x11-common
Suggested packages:
  ffmpeg-doc i965-va-driver-shaders libasound2-plugins alsa-utils libbluray-bdj jackd2 libportaudio2 opus-tools
  pulseaudio libraw1394-doc librsvg2-bin xdg-utils lm-sensors serdi sndiod sordi speex default-mysql-client
  postgresql-client opencl-icd nvidia-vdpau-driver nvidia-tesla-440-vdpau-driver nvidia-tesla-418-vdpau-driver
  nvidia-legacy-390xx-vdpau-driver nvidia-legacy-340xx-vdpau-driver
The following NEW packages will be installed:
  alsa-topology-conf alsa-ucm-conf ffmpeg fontconfig i965-va-driver intel-media-va-driver libaacs0 libasound2
  libasound2-data libass9 libasyncns0 libavc1394-0 libavcodec58 libavdevice58 libavfilter7 libavformat58
  libavresample4 libavutil56 libbdplus0 libblas3 libbluray2 libbs2b0 libcaca0 libcairo-gobject2 libcairo2
  libcdio-cdda2 libcdio-paranoia2 libcdio19 libchromaprint1 libcodec2-0.9 libdatrie1 libdc1394-25 libdrm-amdgpu1
  libdrm-common libdrm-intel1 libdrm-nouveau2 libdrm-radeon1 libdrm2 libflac8 libflite1 libfribidi0 libgbm1
  libgdk-pixbuf-2.0-0 libgdk-pixbuf2.0-bin libgdk-pixbuf2.0-common libgfortran5 libgl1 libgl1-mesa-dri libglapi-mesa
  libglvnd0 libglx-mesa0 libglx0 libgme0 libgraphite2-3 libgsm1 libharfbuzz0b libiec61883-0 libigdgmm11
  libjack-jackd2-0 liblapack3 liblilv-0-0 libllvm11 libmariadb3 libmfx1 libmicrohttpd12 libmp3lame0 libmpg123-0
  libmysofa1 libnorm1 libogg0 libopenal-data libopenal1 libopenmpt0 libopus0 libpango-1.0-0 libpangocairo-1.0-0
  libpangoft2-1.0-0 libpciaccess0 libpgm-5.3-0 libpixman-1-0 libpocketsphinx3 libpostproc55 libpq5 libpulse0
  librabbitmq4 libraw1394-11 librsvg2-2 librsvg2-common librubberband2 libsamplerate0 libsdl2-2.0-0 libsensors-config
  libsensors5 libserd-0-0 libshine3 libsnappy1v5 libsndfile1 libsndio7.0 libsodium23 libsord-0-0 libsoxr0 libspeex1
  libsphinxbase3 libsratom-0-0 libsrt1.4-gnutls libssh-gcrypt-4 libswresample3 libswscale5 libthai-data libthai0
  libtheora0 libtwolame0 libudfread0 libv4l-0 libv4l2rds0 libv4lconvert0 libva-drm2 libva-x11-2 libva2 libvdpau-va-gl1
  libvdpau1 libvidstab1.1 libvorbis0a libvorbisenc2 libvorbisfile3 libvpx6 libvulkan1 libwavpack1 libwayland-client0
  libwayland-cursor0 libwayland-egl1 libwayland-server0 libx11-xcb1 libx264-160 libxcb-dri2-0 libxcb-dri3-0
  libxcb-glx0 libxcb-present0 libxcb-randr0 libxcb-render0 libxcb-shape0 libxcb-shm0 libxcb-sync1 libxcb-xfixes0
  libxcursor1 libxdamage1 libxfixes3 libxi6 libxinerama1 libxkbcommon0 libxrandr2 libxrender1 libxshmfence1 libxss1
  libxv1 libxvidcore4 libxxf86vm1 libz3-4 libzmq5 libzvbi-common libzvbi0 mariadb-common mesa-va-drivers
  mesa-vdpau-drivers mesa-vulkan-drivers motion mysql-common ocl-icd-libopencl1 pocketsphinx-en-us python3-boto3
  python3-botocore python3-dateutil python3-jmespath python3-s3transfer v4l-utils va-driver-all vdpau-driver-all
  x11-common
0 upgraded, 178 newly installed, 0 to remove and 0 not upgraded.
Need to get 136 MB of archives.
After this operation, 461 MB of additional disk space will be used.

MichaIng, 2022-03-15:
That looks actually good. Especially `ffmpeg` pulls a large number of dependencies, and `motion` at least the database libraries. The packages are automatically installed as well, long with further setup steps, isn't it?

This makes me think that we should add some messages to the script so one knows what is going on.

starbasessd, 2022-03-15:
And for what. The way it was arranged before, you saw what motion and ffmpeg were installing as their dependencies, instead of just one big package. And be sure to either put a pause to be readable, or a wait for input, if you are going to leave it in motion_init (a lot of recently disabled packages are doing the pause so people can notice before continuing on during apt-get upgrade, a Good Thing(tm) IMO)
And are you going to make motion/ffmpeg optional just to use motionEye as a Hub?

MichaIng, 2022-03-15:
> And for what. The way it was arranged before, you saw what motion and ffmpeg were installing as their dependencies, instead of just one big package.

You'll see the exact same output if your do:
```sh
apt install motion curl v4l-utils ffmpeg curl python3-boto3
```
Or did you install every package in an individual call before?

> And be sure to either put a pause to be readable, or a wait for input

The `motioneye_init` command is supposed to be a quick start command, also used in scripts, so I think it isn't the right way do add pauses/input into it. But two things:
- It should print messages about what it is doing
- The documentation/Wiki (when we add it) should clearly state that it does a full installation as system service with via `ffmpeg` for hardware acceleration, a systemd unit etc.
- For admins which want to set it up differently, we should link or list the steps it does, especially where the sample config and systemd unit can be found.

> And are you going to make motion/ffmpeg optional just to use motionEye as a Hub?

motionEye depends strictly on a local `motion`, so this cannot be made optional, and it never was possible. `ffmpeg`, as mentioned is for (hardware accelerated) video encoding/decoding, which is practically required or very reasonable in very most use cases.

But, again I'm mixing/widen topics, we could add a little CLI, like adding a `--skip-ffmpeg` and more importantly `--skip-boto` (required only for AWS S3 uploads, if I understood correctly). However, for testing purpose a complete setup with all dependencies makes sense anyway, isn't it?

starbasessd, 2022-03-15:
After 'completion' of motioneye_init:
- - - - -
Warning: The home dir /var/lib/motion you specified already exists.
Adding system user `motion' (UID 107) ...
Adding new user `motion' (UID 107) with group `motion' ...
The home directory `/var/lib/motion' already exists.  Not copying from `/etc/skel'.
adduser: Warning: The home directory `/var/lib/motion' does not belong to the user you are currently creating.
Adding user `motion' to group `video' ...
Adding user motion to group video
Done.
Created symlink /etc/systemd/system/multi-user.target.wants/motion.service → /lib/systemd/system/motion.service.
Setting up ffmpeg (7:4.3.3-0+deb11u1) ...

Broadcast message from systemd-journald@deb11-me-test2 (Tue 2022-03-15 10:04:35 EDT):

motion[5163]: [0:motion] [EMG] [ALL] motion_startup: Exit motion, cannot create log file /var/log/motion/motion.log: Permission denied

Message from syslogd@deb11-me-test2 at Mar 15 10:04:35 ...
 motion[5163]: [0:motion] [EMG] [ALL] motion_startup: Exit motion, cannot create log file /var/log/motion/motion.log: Permission denied
Processing triggers for man-db (2.9.4-2) ...
Processing triggers for libc-bin (2.31-13+deb11u2) ...
Processing triggers for libgdk-pixbuf-2.0-0:amd64 (2.42.2+dfsg-1) ...
Created symlink /etc/systemd/system/multi-user.target.wants/motioneye.service → /etc/systemd/system/motioneye.service.
- - - - -
netstat -tupln | grep 8765
tcp        0      0 0.0.0.0:8765            0.0.0.0:*               LISTEN      5433/python3
- - - - -
 sudo systemctl status motioneye
● motioneye.service - motionEye Server
     Loaded: loaded (/etc/systemd/system/motioneye.service; enabled; vendor preset: enabled)
     Active: active (running) since Tue 2022-03-15 10:04:40 EDT; 4min 55s ago
   Main PID: 5433 (meyectl)
      Tasks: 5 (limit: 4678)
     Memory: 44.1M
        CPU: 914ms
     CGroup: /system.slice/motioneye.service
             ├─5433 /usr/bin/python3 /usr/local/bin/meyectl startserver -c /etc/motioneye/motioneye.conf
             └─5446 /usr/bin/python3 /usr/local/bin/meyectl startserver -c /etc/motioneye/motioneye.conf

Mar 15 10:04:40 deb11-me-test2 systemd[1]: Started motionEye Server.
Mar 15 10:04:40 deb11-me-test2 meyectl[5433]: configure_logging cmd motioneye: False
Mar 15 10:04:40 deb11-me-test2 meyectl[5433]: configure logging to file: None
Mar 15 10:04:40 deb11-me-test2 meyectl[5433]:     INFO: hello! this is motionEye server 0.43.0
Mar 15 10:04:40 deb11-me-test2 meyectl[5433]:     INFO: main config file /etc/motioneye/motion.conf does not exist, using default values
Mar 15 10:04:40 deb11-me-test2 meyectl[5433]:     INFO: cleanup started
Mar 15 10:04:40 deb11-me-test2 meyectl[5433]:     INFO: wsswitch started
Mar 15 10:04:40 deb11-me-test2 meyectl[5433]:     INFO: tasks started
Mar 15 10:04:40 deb11-me-test2 meyectl[5433]:     INFO: mjpg customer garbage collector has started
Mar 15 10:04:40 deb11-me-test2 meyectl[5433]:     INFO: server started
- - - - -

MichaIng, 2022-03-15:
That is great. Just do mention the value of the GitHub Actions workflows we added: We already know that it works until that point because of the tests. So interesting is indeed whether all features of the web interface are working as expected, and/or whether error messages are produces during usage in `journalctl -u motioneye` (what I already found is failing surveillance user login).

MichaIng, 2022-03-15:
> Why is it setting /var/lib/motioneye as the HOME directory for user motion?

This is done by the `motion` package `postinst` script, outside of our control, and never was different, also is not an issue. I agree that the outputs are confusing and could be changed. But this is a `motion` upstream issue:
- https://github.com/Motion-Project/motion-packaging/issues
- https://github.com/Motion-Project/motion-packaging/blob/master/debian/motion.postinst#L17-L18

> Constantly getting pop-up overlay during sign-in for all fields:

motionEye does not support HTTPS, so that is expected. To resolve this, one needs to setup a proxy with HTTPS. This warning is from your browser, not from motionEye, and hence it is a bug in your browser if that button does not do anything.

Hmm, adding MJPEG and RTSP streams worked fine here when I tested a few days ago. Could you make a dedicated issue out of this? Include the logs, they contain valuable information.

starbasessd, 2022-03-15:
> This is done by the `motion` package `postinst` script, outside of our control, and never was different, also is not an issue. I agree that the outputs are confusing and could be changed. But this [...]

Really? Motion uses a motionEye (a wrapper) folder for its files? Really?  
> motionEye does not support HTTPS, so that is expected. To resolve this, one needs to setup a proxy with HTTPS. This warning is from your browser, not from motionEye, and hence it is a bug in your [...]

Hmmm... Doesn't occur in previous version using the same browser.

> Hmm, adding MJPEG and RTSP streams worked fine here when I tested a few days ago. Could you make a dedicated issue out of this? Include the logs, they contain valuable information.

Will do.

zagrim, 2022-03-15:
This also occurred [here](https://github.com/motioneye-project/motioneye/issues/2359#issuecomment-1068035320):
`motion[5163]: [0:motion] [EMG] [ALL] motion_startup: Exit motion, cannot create log file /var/log/motion/motion.log: Permission denied`

Doesn't look too good to me? Permission issue?

zagrim, 2022-03-15:
Ah, it was the "rogue" motion service! :smile:  I was just being baffled on where on earth those things are coming from since linux_init doesn't create users etc...

starbasessd, 2022-03-15:
@MichaIng 
>Ah sorry I misread this. Why do you think /var/lib/motioneye is made the home dir? It is not in my case an there is no step to touch this user's home dir 🤔.

Heck if I know, but the motion install you showed isn't doing it, something within motioneye (maybe a bad copy into an _init file doing it?) but motion-project would never create a motioneye folder for anything in their system, unless there is a cross-over of members, and I doubt they would allow an oops like that...

MichaIng, 2022-03-15:
> Well, the command `adduser --system --home /var/lib/motion # line 18` is what is setting it

But that is `/var/lib/motion` as intended, not `/var/log/motioneye`, and it is what the `motion` package does, not what we do. So just to clear it up:
- motionEye is a frontend for motion, but a completely dedicated otherwise unrelated project.
- motion on install creates the `motion` user with a home directory `/var/lib/motion`, the `motion.service` and all such things one might want to have or need to run and use motion as standalone application.
- motionEye simply reuses the `motion` command and user, but starts it as own process with own config files in `/etc/motioneye`, hence it does not require the `motion.service` or the default motion configs in `/etc/motion`. But it re-uses the `motion` user to benefit from the assured permissions to access camera devices etc. That way it is also compatible with `motion.service` usage, if for some reason admins want to use if for some cameras while controlling other cameras with motionEye.

The above things are and were always the same, have not changed in `dev`, and I see no reason to change them, it simply makes sense to me the way it is, even that it may cause confusion to possible have two config, data and logs directories:
- `/etc/motioneye` and `/etc/motion`
- `/var/lib/motioneye` and `/var/lib/motion`
- `/var/log/motioneye` and `/var/log/motion`

The first is used for motionEye itself and its own `motion` process. The second is only used by `motion.service`, if admins manually set it up (which as we see does not work OOTB as of the upstream bug, possibly even intentionally).

starbasessd, 2022-03-15:
First, why do you keep calling it an upstream bug?

sudo getent passwd motion:
motion:x:107:115::/var/lib/motion:/usr/sbin/nologin

MichaIng, 2022-03-15:
> First, why do you keep calling it an upstream bug?

Because it is an upstream bug, i.e. a bug or issue in the `motion` package's `postinst` script as stated and linked above. We simply download and install the official `motion` DEB package. If this causes issues, it's out of our control and needs to be fixed by the motion project, not by us.

> motion❌107:115::/var/lib/motion:/usr/sbin/nologin

So all exactly as it should be, `/var/lib/motioneye` is not made the home of the `motion` user, as expected.

starbasessd, 2022-03-15:
Must've been me, or the upstream motion-project, motioneye is perfect, never mad a mistake.

closed 2022-03-15 by starbasessd

MichaIng, 2022-03-15:
> Must've been me, or the upstream motion-project

You ~sound~ read ironic. Whatever you meant with `/var/lib/motioneye`, it was neither you nor motion-project, because nothing is wrong, all is as it should be.

MichaIng, 2022-03-18:
PR opened at motion project to resolve one warning + another unnecessary message during package install, with the option to resolve a second warning as well: https://github.com/Motion-Project/motion-packaging/pull/14

The expectedly failing `motion.service` cannot be so easily resolved since it is added by debhelper packaging tools. Not sure whether it's better to skip that service startup or to have the systemd service pre-creating the missing log directory (which is easy to achieve). I'll open an issue about this question.

zagrim, 2022-03-19:
We could have the init script disable motion.service? Of course that's not nice IF someone has it running already and wants to keep on using it for something.

MichaIng, 2022-03-19:
I was also thinking about this. However, since there is not way to disable the startup attempt during package install (we can only disable the service afterwards), that warning remains.

But it would reduce the overhead of motion service startup attempts on every boot. Actually it is "disabled" via `/etc/default/motion` setting, but this is not interpreted by the systemd unit to skip the attempt gracefully (would be possible as well) but within the `motion` startup phase, obviously after the first logging is tried. Probably it's even a "motion is disabled via /etc/default/motion, aborting..." like message 😄.

However, in any case you have a full process spawn on every boot as long as the service is enabled, which is unnecessary overhead and logs spam. And I think the chance that someone really wants motion service in addition to motionEye service is near to zero, while confusion with a failing motion service startup attempt is more of an issue?