motioneye-project/motioneye #2360: Unable to add RTSP or MJPEG cameras to motioneye 0.43.0

starbasessd, 2022-03-15:
-- Journal begins at Fri 2021-08-27 12:43:03 EDT, ends at Tue 2022-03-15 11:08:36 EDT. --
Mar 15 10:04:40 deb11-me-test2 systemd[1]: Started motionEye Server.
Mar 15 10:04:40 deb11-me-test2 meyectl[5433]: configure_logging cmd motioneye: False
Mar 15 10:04:40 deb11-me-test2 meyectl[5433]: configure logging to file: None
Mar 15 10:04:40 deb11-me-test2 meyectl[5433]:     INFO: hello! this is motionEye server 0.43.0
Mar 15 10:04:40 deb11-me-test2 meyectl[5433]:     INFO: main config file /etc/motioneye/motion.conf does not exist, using default values
Mar 15 10:04:40 deb11-me-test2 meyectl[5433]:     INFO: cleanup started
Mar 15 10:04:40 deb11-me-test2 meyectl[5433]:     INFO: wsswitch started
Mar 15 10:04:40 deb11-me-test2 meyectl[5433]:     INFO: tasks started
Mar 15 10:04:40 deb11-me-test2 meyectl[5433]:     INFO: mjpg customer garbage collector has started
Mar 15 10:04:40 deb11-me-test2 meyectl[5433]:     INFO: server started
Mar 15 10:11:22 deb11-me-test2 meyectl[5433]:  WARNING: 403 GET /config/list/?_=1647353481822 (192.168.2.224) 0.99ms
Mar 15 10:12:03 deb11-me-test2 meyectl[5433]:    ERROR: Exception in callback test_mjpeg_url.<locals>.on_response(<Future finis...route to host>) >
Mar 15 10:12:03 deb11-me-test2 meyectl[5433]: handle: <Handle test_mjpeg_url.<locals>.on_response(<Future finis...route to host>) at /usr/local/li>
Mar 15 10:12:03 deb11-me-test2 meyectl[5433]: Traceback (most recent call last):
Mar 15 10:12:03 deb11-me-test2 meyectl[5433]:   File "/usr/lib/python3.9/asyncio/events.py", line 80, in _run
Mar 15 10:12:03 deb11-me-test2 meyectl[5433]:     self._context.run(self._callback, *self._args)
Mar 15 10:12:03 deb11-me-test2 meyectl[5433]:   File "/usr/local/lib/python3.9/dist-packages/motioneye/utils/mjpeg.py", line 96, in on_response
Mar 15 10:12:03 deb11-me-test2 meyectl[5433]:     if response.code == 401 and auth_modes and url_obj.username:
Mar 15 10:12:03 deb11-me-test2 meyectl[5433]: AttributeError: '_asyncio.Future' object has no attribute 'code'
Mar 15 10:19:05 deb11-me-test2 meyectl[5433]:    ERROR: Future exception was never retrieved
Mar 15 10:19:05 deb11-me-test2 meyectl[5433]: future: <Future finished exception=HTTP 599: Failed to connect to 192.168.2.11 port 80: No route to >
Mar 15 10:19:05 deb11-me-test2 meyectl[5433]: tornado.curl_httpclient.CurlError: HTTP 599: Failed to connect to 192.168.2.11 port 80: No route to >
Mar 15 10:19:06 deb11-me-test2 meyectl[5433]:  WARNING: 403 GET /login/?_=1647353945942&_username=_&_signature=be41fcc3cf04ae0c862b38e57f960e0c0fc>
Mar 15 10:23:12 deb11-me-test2 meyectl[5433]:    ERROR: Exception in callback test_mjpeg_url.<locals>.on_response(<Future finis...tion Required>) >
Mar 15 10:23:12 deb11-me-test2 meyectl[5433]: handle: <Handle test_mjpeg_url.<locals>.on_response(<Future finis...tion Required>) at /usr/local/li>
Mar 15 10:23:12 deb11-me-test2 meyectl[5433]: Traceback (most recent call last):
Mar 15 10:23:12 deb11-me-test2 meyectl[5433]:   File "/usr/lib/python3.9/asyncio/events.py", line 80, in _run
Mar 15 10:23:12 deb11-me-test2 meyectl[5433]:     self._context.run(self._callback, *self._args)
Mar 15 10:23:12 deb11-me-test2 meyectl[5433]:   File "/usr/local/lib/python3.9/dist-packages/motioneye/utils/mjpeg.py", line 96, in on_response
Mar 15 10:23:12 deb11-me-test2 meyectl[5433]:     if response.code == 401 and auth_modes and url_obj.username:
Mar 15 10:23:12 deb11-me-test2 meyectl[5433]: AttributeError: '_asyncio.Future' object has no attribute 'code'
Mar 15 10:28:56 deb11-me-test2 meyectl[5433]:     INFO: adding new mjpeg camera with id 1...
Mar 15 10:28:57 deb11-me-test2 meyectl[5433]:    ERROR: Future exception was never retrieved
Mar 15 10:28:57 deb11-me-test2 meyectl[5433]: future: <Future finished exception=HTTP 599: Operation timed out after 10001 milliseconds with 70913>
Mar 15 10:28:57 deb11-me-test2 meyectl[5433]: tornado.curl_httpclient.CurlError: HTTP 599: Operation timed out after 10001 milliseconds with 70913>
Mar 15 10:29:15 deb11-me-test2 meyectl[5433]:     INFO: main config file /etc/motioneye/motion.conf does not exist, using default values
Mar 15 10:29:15 deb11-me-test2 meyectl[5433]:     INFO: removing camera config file /etc/motioneye/camera-1.conf...
Mar 15 10:29:50 deb11-me-test2 meyectl[5433]:    ERROR: Exception in callback test_mjpeg_url.<locals>.on_response(<Future finis...tion Required>) >
- - - - -
No logs under /var/log/motioneye
- - - - -
ping -c4 192.168.2.11
PING 192.168.2.11 (192.168.2.11) 56(84) bytes of data.
64 bytes from 192.168.2.11: icmp_seq=1 ttl=64 time=3.61 ms
64 bytes from 192.168.2.11: icmp_seq=2 ttl=64 time=5.15 ms
64 bytes from 192.168.2.11: icmp_seq=3 ttl=64 time=3.15 ms
64 bytes from 192.168.2.11: icmp_seq=4 ttl=64 time=4.75 ms

--- 192.168.2.11 ping statistics ---
4 packets transmitted, 4 received, 0% packet loss, time 3016ms
- - - - -
traceroute 192.168.2.11
traceroute to 192.168.2.11 (192.168.2.11), 30 hops max, 60 byte packets
 1  192.168.2.11 (192.168.2.11)  3.732 ms  3.667 ms  4.502 ms
for the no route to host 192.168.2.11
- - - - -
the dlink dcs-5020 was tried as both Network Camera (what works on production previous motioneye) and Simple MJPEG, neither worked.

starbasessd, 2022-03-15:
Environment:
Debian11 server, fully updated
using install instructions posted in 
https://github.com/motioneye-project/motioneye/issues/2357#issuecomment-1067471957
Trying to add MJPEG Camera d-Link dcs-5020-l using same Network Camera and URL settings as motionEye 0.42.1 (production server) and Foscam R2C RTSP using same Network Camera and URL settings as motionEye 0.42.1 (production server).

starbasessd, 2022-03-15:
Must've been me. Software is perfect.

MichaIng, 2022-03-15:
It is a bug and a fix is already on the way: #2362

MichaIng, 2022-03-15:
Fixed now in `dev`, including surveillance user login in general.

closed 2022-03-15 by MichaIng