motioneye-project/motioneye #2365: Motioneye vs motioneyeOS

bulgin, 2022-03-15:
What are the differences between Motioneye and MotionEyeOS?  I ask because I couldn't get apache with signed certificates to work with motioneyeOS so switched to MotionEye debian on Raspberry Pi 4 and now have encrypted connections but now the camera is not recognized.

Using Raspberry Pi 4 with infra-red camera module.

MichaIng, 2022-03-15:
The first is an application you can run on any Linux distro, the second is an own Linux distro (actually an Ubuntu) with motionEye installed.

If I'm not mistaken, motionEyeOS comes with Nginx as proxy installed already, so two webservers on port 80/443 conflict, of course. I.e. you'd need to configure Nginx with the HTTPS certificates.

But I didn't have a closer look at motionEyeOS yet, so probably others can give more details.

bulgin, 2022-03-15:
wow that's helpful I didn't know that.  So I should keep the motionEyeOS (since I prefer Nginx anyway) and make it work with credentials.  I was actually able to get the motionseye version to work with encryoted apache quiet easily.  I'd prefer Nginx.  I'll see what I can do with that, then.

MichaIng, 2022-03-15:
At least you should have a look at the services running in motionEye OS, like with `top` or `htop`, `pgrep nginx`, `ss -tlp` or so. If Nginx is not listed, probably `service nginx start` will do, and the configs in `/etc/nginx`. I see such files in the [motionEyeOS sources](https://github.com/ccrisan/motioneyeos), but haven't tested it 😉.

bulgin, 2022-03-16:
Hmm.  Problems persist.  Burned this image: 

https://github.com/ccrisan/motioneyeos/releases/download/20200606/motioneyeos-raspberrypi4-20200606.img.xz 

for the Raspberry pi4B.  The camera is an "Inland" night vision camera listed as comparable with all makes of raspberry PI.  The burned image loads and I can access the motioneyesOS admin panel - all features seem to be in place but the camera just shows a gray screen.  logs show:

2022-03-15 23:57:06: [motioneye]     INFO: adding new v4l2 camera with id 1...
2022-03-15 23:57:07: [motioneye]     INFO: main config file /data/etc/motion.conf does not exist, using default values
2022-03-15 23:57:26: [motioneye]    ERROR: mjpg client timed out receiving data for camera 1 on port 8081

Running the commands you suggested produce "command not found"  Wonder if the downloaded img file is bogus or what?  Despite it working it does not want to register a camera. . .

MichaIng, 2022-03-16:
Git has the sources, not all OS files 😉. How did you enable HTTPS now, did you check whether Nginx is running, respectively whether an Nginx config file is available? Which command produced `command not found`?

bulgin, 2022-03-16:
> command not found
> y-To: motioneye-project/motioneye <
> motioneye-project/motioneye ***@***.***>Cc: bulgin <
> [motioneye-project/motioneye] Motioneye vs motioneyeOS (Issue
> #2365)Date: Tue, 15 Mar 2022 17:32:58 -0700

Haven't enabled https still trying to get the camera to run.  
***@***.***  etc]# service nginx startNo such service: ***@***.***
  etc]# ss -tlp-sh: ss: command not ***@***.***  etc]# pgrep

MichaIng, 2022-03-16:
Ui, a very minimal OS it seems, without even such a basic command like `pgrep` and no service management at all. Does this directory exist?
```sh
ls -l /etc/nginx
```

bulgin, 2022-03-16:
ls: /etc/nginx: No such file or directory

bulgin, 2022-03-16:
Yet the motioneyeOS panel is accecable.

MichaIng, 2022-03-16:
Indeed, I just had a look into the image, there is no Nginx or other webserver installed on this image. Looks like the GitHub repo contains a lot of software options, but they are not actually installed on the images.

Also there is not package manage (`apt` or such) installed, hence it is not trivial to install one without bootstrapping dpkg and apt first.
________
But back to the grey camera stream: So it is not related to HTTPS? Because in OP you said:
> and now have encrypted connections but now the camera is not recognized.

Not sure where logs are stored, probably `/var/log/motioneye` or somewhere in `/data`.

bulgin, 2022-03-16:
Indeed.  That was with motioneye not mostioneyeOS. Camera didn't work
then either, but I could install apache and use encryption.  Because
camera didn't work I figured I'd switch over to the motioneyeOS.  But
still doesn't 
Regardless, seems like this project was dropped a while ago and perhaps
the newer cameras/OS's can't run it. . . dunno.

#2365)Date: Tue, 15 Mar 2022 17:48:12 -0700

bulgin, 2022-03-16:
Indeed.  That was with motioneye not mostioneyeOS. Camera didn't work then either, but I could install apache and use encryption.  Because camera didn't work I figured I'd switch over to the motioneyeOS.  But still doesn't 

Regardless, seems like this project was dropped a while ago and perhaps the newer cameras/OS's can't run it. . . dunno.

MichaIng, 2022-03-16:
Looks like we need to get the README update merged better sooner than later. Check: https://github.com/motioneye-project/motioneye/pull/2356/files

The project has been transferred to a GitHub organisation with some new maintainers and contributors. Work of course concentrates on motionEye itself at first, probably Docker second (some work has been done on it already) and motionEyeOS afterwards (since it also has not been transferred to this organisation yet). Just to get forward on this point @ccrisan would you mind to transfer it as well, so we generally gain the ability to update/maintain it as well. We'll see whether we get sufficient man power for it, but would be good to have the option if someone wants to start working on it.

ccrisan, 2022-03-16:
motionEyeOS is a BuildRoot-based OS with motionEye (and required deps) installed. It's meant as a "firmware" for a RPi-based camera system and does not allow installing stuff. It runs in a readonly mode so it's more resilient to SD card corruption issues.

motionEyeOS does not (currently) have Nginx but that might be added in the future, including an automated mechanism to register/refresh TLS certificates from Let's Encrypt.

In fact I'm maintaining its parent OS [thingOS](https://github.com/ccrisan/thingos) and I hope that once we have a modernized motionEye, we'll be able to update motionEyeOS accordingly.

Transfer of `ccrisan/motioneyeos` to `motioneye-project/motioneyeos` has now been completed.

MichaIng, 2022-03-16:
Ah nice, looks like I got the wrong impression that it may be Ubuntu based from the Dockerfile. So what is found in `package/` directory are available, but not necessarily installed thingOS packages. This is where I found the Nginx config files when searching the code.

> motionEyeOS does not (currently) have Nginx but that might be added in the future, including an automated mechanism to register/refresh TLS certificates from Let's Encrypt.

Probably we are able to get HTTPS support integrated right into motionEye, after the next modernising release has been done 🙂.

> Transfer of `ccrisan/motioneyeos` to `motioneye-project/motioneyeos` has now been completed.

🚀

bulgin, 2022-03-16:
Great to hear.  I believe there is GREAT promise in this application
and hope it all goes well and wishing everyone involved good luck.
 Keep up the great work!

#2365)Date: Wed, 16 Mar 2022 07:10:29 -0700

JamBalaya56562, 2026-11-10:
Closing, as MichaIng answered this: motionEye is the application that runs on any Linux, motionEyeOS is a whole system image with motionEye built in, and its own nginx is why Apache could not take ports 80/443 there.

closed 2026-11-10 by JamBalaya56562