motioneye-project/motioneye #2368: Frame rate display, and slow capture frame rate...

salzrat, 2022-03-17:
I have the MotionEye addon (0.17.0) installed in Homeassistant, which is running in a Virtualbox on a Macbook Air.

First, I wonder about the overlay display I get when I click the camera: the tooltip says this is "streaming/capture frame rate". However, when I use the "Streaming Frame Rate" slider in the configuration, it directly affects the second number, not the first one (i.e., the one that is supposed to be the "capture" frame rate). Is this just an inconsistent tooltip or something else at play?

Second, when I disable movie passthrough under "Movies", the movies that are generated have really low frame rate (something like 1-2 seconds per frame). Why is this happening? Is the encoder dropping frames because there is not enough CPU power available? How can I debug this? Maybe I can allocate more CPU to the Virtualbox image. 

Third, when I have movie passthrough enabled, the movie starts 5-10 seconds before the actual motion. So it seems somehow the timestamps of the captured video may not be interpreted correctly or something? 

I have managed to ssh into the motioneye docker image in Homeassistant, so I can hack around there as well if it helps...

Please let me know what further information I can provide to debug this...
Thanks!

zagrim, 2022-03-17:
I don't have any experience on HomeAssistant, and don't really know anything of it. So I don't know what changes there may be due to HA compared to how ME works by itself.

1. I looked at the code and while the naming of things in code corresponds with the tooltip, I still think there's a mix-up somewhere, because I have "Motion Optimization" enabled, which drops streaming fps to one when there's no motion detected, and I have the latter fps figure equal to one currently. So, I think there's data being mixed up somewhere, but it would take more effort to actually get to the bottom of it.
2. If CPU load peaks around 100% then that is likely cause for dropping frames. If you have ssh connection to the VM, then you could run e.g. `top` on the command line. Post the output here if you have need assistance on interpreting it. Oh, BTW, what is the "Frame Rate" setting under "Video Device"?
3. Do the movies start at motion start when you have passthrough disabled, then? But I think that it should be the responsibility of Motion to figure out which frames to save, without ME having to sync anything with it. BTW, what are your settings for "Motion Gap" and "Captured Before" under "Motion Detection"?

salzrat, 2022-03-18:
Thanks. 

2. When there is no motion, top shows a CPU usage of about 15-20%. I'll have to check with some motion going on. I have currently set the video frame rate to 4 under video device, but I'll have to play around with it some more. Interestingly, the "Capture Frame Rate" shown in the overlay is not influenced by that at all. So what is the "capture frame rate"? Is it the frame rate at which frames are processed for motion detection? And if motion is detected, only the number of frames specified under "video device" will be put into the video? Shouldn't this setting then rather go under the "Video" section? This is quite confusing for me at the moment...

3. Yes, when passthrough is disabled, the video starts right with the motion. Only when passthrough is enabled, the video starts way before the motion. The settings are 30/1/60/5 for motion gap, captured before/after and minimum motion frames.
4.

zagrim, 2022-03-19:
My initial impression was that "streaming fps" and "capture fps" are just mixed up, but looking now my own production setup I see "5.0 / 1.0" where the latter should correspond to streaming reduced to 1 fps due to Motion Optimization, but the former weirdly doesn't match with the device frame rate which is set to 4 (UI setting for that matches the camera config file setting). I think that is worth a separate issue for investigating and fixing.

3. I cant't think of an explanation for the difference, sorry. Your settings should not cause it, at least.

salzrat, 2022-03-19:
So I tested some more. The reason for the really slow capture seems to have been CPU overload when having "Create Debug Media" on. Without streaming, this already gave >60% CPU utilization in top. Turning it off creates with the frame rate that I set under video device, and it looks fine, and stays below 40% CPU utilization, even for 15fps.

I wanted to see why it captures so many video during the night, but leaving the debug media on seems to only work with more powerful setups...

I have split question 3. in a separate issue:
https://github.com/motioneye-project/motioneye/issues/2372

and I will comment about question 1. in the following issue I have found:
https://github.com/motioneye-project/motioneye/issues/297

closed 2022-03-19 by salzrat