motioneye-project/motioneye #2372: Movie starts too early for Doorbird with Movie Passthrough

salzrat, 2022-03-19:
This is a separate issue for the third item mentioned here:

https://github.com/motioneye-project/motioneye/issues/2368

I have the MotionEye addon (0.17.0) installed in Homeassistant, which is running in a Virtualbox on a Macbook Air connected via LAN.

The camera is a Doorbird D101s also connected via LAN.

When I have movie passthrough enabled, the movie starts 5-10 seconds before the actual motion. So it seems somehow the timestamps of the captured video may not be interpreted correctly or something? This does not occur when disabling passthrough, here the movie starts correctly with the detected motion. The relevant settings are 30/1/60/5 for motion gap, captured before/after and minimum motion frames.

I would like to use passthrough because the video files from the camera are much smaller.

I have managed to ssh into the motioneye docker image in Homeassistant, so I can hack around there as well if it helps...

Please let me know what further information I can provide to debug this...
Thanks!

MichaIng, 2022-03-19:
Would be interesting to know whether this happens the same way with a native installation of the current motionEye `dev` branch: https://github.com/motioneye-project/motioneye/blob/fcf1b1f/README.md#installation

Not sure when I find time to test the HA addon.

salzrat, 2022-03-19:
I don't think I have a device where I can easily do that. But I don't think HA plays a role here, this rather seems an issue with how motion handles the video stream it gets from doorbird...

zagrim, 2022-03-20:
@salzrat Did you already mention the version of Motion you have? That might be relevant, at least if its pretty old. I'm not sure of the HA addon setup if Motion could be upgraded easily if necessary.

salzrat, 2022-03-20:
Hi, according to %{ver} on the text overlay the version is 4.3.2. I can try to upgrade the plugin and see whether it comes with 4.4.0...

zagrim, 2022-03-20:
Yep, 4.3.2 should be fine.

JamBalaya56562, 2026-11-15:
@salzrat Does this still happen with a current version of the add-on? If it does, what keyframe (I-frame) interval is the Doorbird set to? With passthrough motion copies the camera's stream without re-encoding, so a movie may only be able to start at a keyframe, and a long interval would move the start earlier than the pre-capture.

salzrat, 2026-11-15:
I'm sorry, my doorcam died some months ago and I didn't have time to get a replacement so I can't look into this really.

JamBalaya56562, 2026-11-15:
Thanks for letting us know, and sorry about the doorcam. Closing this, since it can no longer be checked and nobody else has reported it. If it comes back with another camera, a comment with the camera's keyframe interval and the motion version would let us reopen it.

closed 2026-11-15 by JamBalaya56562

MichaIng, 2026-11-15:
Interesting. So when grabbing the "current" picture/frame, it is the last keyframe, since an I-frame is a frame within a compressed block, which naturally cannot be the start of a movie (without re-encoding)?

JamBalaya56562, 2026-11-15:
Not quite: motion still decodes every frame for motion detection and pictures; only the passthrough movie copies the camera's packets. An I-frame is the self-contained one, so it is the only frame a copied movie can start at. In motion 4.7.1 the movie starts at the oldest keyframe still in the packet buffer (`ffmpeg_passthru_put`), and that buffer holds about twice the pre-capture images (`netcam_rtsp_pktarray_resize`), so the movie can begin well before the pre-capture even with a short keyframe interval.

MichaIng, 2026-11-15:
Ah, I mixed "I-frame" with those frames within a compression block between keyframes. So yeah, it makes sense that inter-block frames cannot start a copied movie.

And I guess motion has no way to know which frame in the buffer is associated to which frame in its processed stream, so it takes the oldest keyframe to not miss anything.

JamBalaya56562, 2026-11-15:
It does know: each image carries the packet number it was grabbed at (`img_data->idnbr_norm`), and `ffmpeg_passthru_put` writes packets only up to that number. For the start, though, it takes the oldest keyframe at or before that image rather than the nearest one, so the movie can begin earlier than it needs to. The source does not say why.