motioneye-project/motioneye #2373: motionEye v0.43.0 release

MichaIng, 2022-03-20:
Most open pull requests have been merged, or major parts of larger collection PRs. Left are minor or layout/GUI additions, or feature PRs which require a rebase with conflicts resolving, all I'd not see as mandatory for a release.

So the question is how we do it.
- First of all is there anything, especially bugs/issues, any of you want to have addressed first?
- Then we should go through all features and test them, at best once on `x86_64` and once on a Raspberry Pi, where some special features are available. How shall we do it? Adding a ToDo list here that everyone can tick off?
- The Wiki needs an update, at least the install instructions. We have a pretty generic method in the `README.md` now. Not sure whether we should add Fedora/rpm and Arch/pkg Python 3 + development headers instructions to `README.md`, deprecating these Wiki pages, or updating them instead and linking it from the `README.md`?
- The [Docker Wiki page](https://github.com/motioneye-project/motioneye/wiki/Install-In-Docker) needs an update as well. Not sure why it contains the whole Dockerfile instead of just linking the existing one 😄. I don't see much point to list Docker Compose instructions as long as motionEye runs fine in a single container? Finally it doubles with the related [README.md](https://github.com/motioneye-project/motioneye/blob/dev/extra/README.md), so similarly the question is whether we want to keep both, e.g. having install instructions in the Wiki and only container build instructions in the README? We may also want to add instructions about [how to install the container from the GitHub registry](https://github.com/motioneye-project/motioneye/pkgs/container/motioneye), until we have a release at Docker Hub, and keeping it for development releases/testing?
- Then, once we do not find any blocker issues anymore, a pre-release makes sense. I think such is possible on PyPI as well, isn't it? At least there is a `--pre` flag. @ccrisan can a project on PyPI be transferred somehow, so that `pip install motioneye` can be done as before without the need to do keep using your personal PyPI account?

Let me know whether I forgot something and what you think about the above.

ccrisan, 2022-03-20:
> can a project on PyPI be transferred somehow, so that pip install motioneye can be done as before without the need to do keep using your personal PyPI account?

A PyPI token can be added as an (ecrypted) secret to [github's secrets](https://github.com/organizations/motioneye-project/settings/secrets/actions) and can be used from CI jobs in GitHub Actions.

Unfortunately there's no "organization" concept on PyPI (at least not one that I can find) so I guess my user is just as good as anyone else's.

I just created the organization-level `PYPI_TOKEN` secret pointing to a new PyPI token scoped to `motioneye`. As far as I remember, it can be used in the CI like `${{ secrets.PYPI_TOKEN }}`.

MichaIng, 2022-03-20:
Many thanks. Indeed, practically it shouldn't make a difference whether we keep using your PyPI account or a new one, safe with an access token.

The Docker Hub download is anyway always done with account prefix, so for this a new account wouldn't be a problem, after install instructions have been updated.

ultratoto14, 2022-03-24:
Just a small remark, in `dev` branch, the `Dockerfile` has been refactored to use an entrypoint and this `entrypoint.sh` file is not copied in the container, the container cannot start.

MichaIng, 2022-03-24:
Many thanks for the hint. I'm no Docker expert (and didn't find the time to test the container yet, to be true), can you open a PR to fix it?

I guess `ENTRYPOINT ["/entrypoint.sh"]` defines the path within the container, but obviously this doesn't imply copying it from the same dir of the Dockerfile?

ultratoto14, 2022-03-24:
Yep just adding 
```
  COPY . /tmp/motioneye
+COPY extra/entrypoint.sh /entrypoint.sh
```
should do the trick.
I did not test yet the docker container, it may need additional changes, will try to find time and if anything else is needed, I'll open a PR.

MichaIng, 2022-03-24:
PR up: #2386

spupuz, 2022-05-15:
Is version 0.43 usable with docker, does this motioneye will be developed in future?

MichaIng, 2022-05-15:
Here is the Docker container: https://github.com/motioneye-project/motioneye/pkgs/container/motioneye
Future development of motionEye is the whole point of the https://github.com/motioneye-project organisation, the v0.43.0 release etc 😉.

spupuz, 2022-05-15:
Is there a docker compose file o saw that it still refer to ccrisan version

MichaIng, 2022-05-15:
That needs to be updated indeed (the wiki in general), however, you should be able to extract the essence from it.

spupuz, 2022-05-15:
Sorry not sure how to do it

MichaIng, 2022-05-15:
Please test with this change: https://github.com/motioneye-project/motioneye/pull/2482/files

spupuz, 2022-05-15:
if i change with ghcr.io/motioneye-project/motioneye:edge the images from cam are not loades, do i have to setup it from scratch or can i maintain ccrisan settings?

MichaIng, 2022-05-15:
The previous configs should work, but to be sure you could change back (and in case forth) affected camera settings and "Apply".

kleini, 2022-05-16:
@spupuz Have a look into the container logs for any error messages.

spupuz, 2022-05-16:
here are the errors:

`    result = await result
  File "/usr/local/lib/python3.9/dist-packages/motioneye/handlers/picture.py", line 52, in get
    await self.current(camera_id)
  File "/usr/local/lib/python3.9/dist-packages/motioneye/handlers/picture.py", line 110, in current
    picture = mediafiles.get_current_picture(
  File "/usr/local/lib/python3.9/dist-packages/motioneye/mediafiles.py", line 1009, in get_current_picture
    sio = io.StringIO(jpg)
TypeError: initial_value must be str or None, not bytes
   ERROR: 500 GET /picture/3/current/?_=1652690473933&width=0.8&_username=admin&_signature=5f6eed5e9e588eeacf6491b89a180d1a4df2fd08 (172.17.0.1) 2.37ms
   ERROR: initial_value must be str or None, not bytes
Traceback (most recent call last):
  File "/usr/local/lib/python3.9/dist-packages/tornado/web.py", line 1704, in _execute
    result = await result
  File "/usr/local/lib/python3.9/dist-packages/motioneye/handlers/picture.py", line 52, in get
    await self.current(camera_id)
  File "/usr/local/lib/python3.9/dist-packages/motioneye/handlers/picture.py", line 110, in current
    picture = mediafiles.get_current_picture(
  File "/usr/local/lib/python3.9/dist-packages/motioneye/mediafiles.py", line 1009, in get_current_picture
    sio = io.StringIO(jpg)
TypeError: initial_value must be str or None, not bytes
   ERROR: 500 GET /picture/4/current/?_=1652690473933&width=0.8&_username=admin&_signature=0e4eeaa7eb714d41910755800c26416f6b0142d8 (172.17.0.1) 2.67ms
   ERROR: initial_value must be str or None, not bytes
Traceback (most recent call last):
  File "/usr/local/lib/python3.9/dist-packages/tornado/web.py", line 1704, in _execute
    result = await result
  File "/usr/local/lib/python3.9/dist-packages/motioneye/handlers/picture.py", line 52, in get
    await self.current(camera_id)
  File "/usr/local/lib/python3.9/dist-packages/motioneye/handlers/picture.py", line 110, in current
    picture = mediafiles.get_current_picture(
  File "/usr/local/lib/python3.9/dist-packages/motioneye/mediafiles.py", line 1009, in get_current_picture
    sio = io.StringIO(jpg)
TypeError: initial_value must be str or None, not bytes
   ERROR: 500 GET /picture/5/current/?_=1652690473933&width=0.8&_username=admin&_signature=82a9a4882050e5b2b024515c4a21eb213aff4ee0 (172.17.0.1) 0.93ms
   ERROR: mjpg client timed out receiving data for camera 6 on port 8086
   ERROR: mjpg client timed out receiving data for camera 6 on port 8086
   ERROR: mjpg client timed out receiving data for camera 6 on port 8086
   ERROR: mjpg client timed out receiving data for camera 6 on port 8086
   ERROR: mjpg client timed out receiving data for camera 6 on port 8086
   ERROR: mjpg client timed out receiving data for camera 6 on port 8086
   ERROR: initial_value must be str or None, not bytes
Traceback (most recent call last):
  File "/usr/local/lib/python3.9/dist-packages/tornado/web.py", line 1704, in _execute
    result = await result
  File "/usr/local/lib/python3.9/dist-packages/motioneye/handlers/picture.py", line 52, in get
    await self.current(camera_id)
  File "/usr/local/lib/python3.9/dist-packages/motioneye/handlers/picture.py", line 110, in current
    picture = mediafiles.get_current_picture(
  File "/usr/local/lib/python3.9/dist-packages/motioneye/mediafiles.py", line 1009, in get_current_picture
    sio = io.StringIO(jpg)
TypeError: initial_value must be str or None, not bytes
   ERROR: 500 GET /picture/1/current/?_=1652690532931&width=0.8&_username=admin&_signature=4598a07c410838457ec01aed7728ec6f8d0b471d (172.17.0.1) 5.48ms
   ERROR: initial_value must be str or None, not bytes
Traceback (most recent call last):
  File "/usr/local/lib/python3.9/dist-packages/tornado/web.py", line 1704, in _execute
    result = await result
  File "/usr/local/lib/python3.9/dist-packages/motioneye/handlers/picture.py", line 52, in get
    await self.current(camera_id)
  File "/usr/local/lib/python3.9/dist-packages/motioneye/handlers/picture.py", line 110, in current
    picture = mediafiles.get_current_picture(
  File "/usr/local/lib/python3.9/dist-packages/motioneye/mediafiles.py", line 1009, in get_current_picture
    sio = io.StringIO(jpg)
TypeError: initial_value must be str or None, not bytes
   ERROR: 500 GET /picture/3/current/?_=1652690541932&width=0.8&_username=admin&_signature=bde666c0fc09648cc17c3b33b093496766b5982c (172.17.0.1) 4.48ms
   ERROR: initial_value must be str or None, not bytes
Traceback (most recent call last):
  File "/usr/local/lib/python3.9/dist-packages/tornado/web.py", line 1704, in _execute
    result = await result
  File "/usr/local/lib/python3.9/dist-packages/motioneye/handlers/picture.py", line 52, in get
[... 7 lines cut ...]
   ERROR: initial_value must be str or None, not bytes
Traceback (most recent call last):
  File "/usr/local/lib/python3.9/dist-packages/tornado/web.py", line 1704, in _execute
    result = await result
  File "/usr/local/lib/python3.9/dist-packages/motioneye/handlers/picture.py", line 52, in get
    await self.current(camera_id)
  File "/usr/local/lib/python3.9/dist-packages/motioneye/handlers/picture.py", line 110, in current
    picture = mediafiles.get_current_picture(
  File "/usr/local/lib/python3.9/dist-packages/motioneye/mediafiles.py", line 1009, in get_current_picture
    sio = io.StringIO(jpg)
TypeError: initial_value must be str or None, not bytes
[... cut ...]
  File "/usr/local/lib/python3.9/dist-packages/motioneye/handlers/picture.py", line 110, in current
    picture = mediafiles.get_current_picture(
  File "/usr/local/lib/python3.9/dist-packages/motioneye/mediafiles.py", line 1009, in get_current_picture
    sio = io.StringIO(jpg)
TypeError: initial_value must be str or None, not bytes
   ERROR: 500 GET /picture/5/current/?_=1652690541932&width=0.8&_username=admin&_signature=733a86ec02a60b721b082d54d61a8c7bcd7bb576 (172.17.0.1) 3.30ms
   ERROR: mjpg client timed out receiving data for camera 6 on port 8086`

kleini, 2022-05-16:
That looks like issue #2398, as I thought.

spupuz, 2022-05-16:
so what i have to do?

cclauss, 2022-05-16:
#2483

MichaIng, 2022-09-25:
@jmichault 
I'd like to do a (pre-)release of motionEye v0.43.0, including upload to PyPI. The [daily upload workflow](https://github.com/motioneye-project/motioneye/blob/python2/.github/workflows/pypi_daily_build.yml) with `PYPI_TOKEN` fails to authenticate: https://github.com/motioneye-project/motioneye/actions/runs/3094711509/jobs/5008377481
```
ERROR    HTTPError: 403 Forbidden from https://upload.pypi.org/legacy/          
         Invalid or non-existent authentication information. See                
         https://pypi.org/help/#invalid-auth for more information.   
```
Any idea what the reason of this is, as the same workflow works fine on your fork. Are there probably different kind of tokens? @ccrisan 

Also, how to do uploads as pre-releases?

MichaIng, 2022-09-26:
Many thanks. I'll create a (pre-)release + manually triggered workflow tomorrow.

It looks like PyPI detects releases vs pre-releases simply via certain key words like "dev" or "rc" in the version string? At least the previously successful uploads to `test.pypi.org` were treated as pre-releases, and I do not see anything else done than adding a `.dev*` suffix to the version string.

chemelli74, 2022-11-25:
Hi, was 0.43.0 released in the end ?

Thank you in advance,

Simone

coder0107git, 2022-12-21:
Can the dev branch be pre-released on Github?

MichaIng, 2022-12-23:
I'm working on it, i.e. the PyPI upload along with a pre-release tag here on GitHub: #2675

MichaIng, 2022-12-23:
Has anyone time and mood to wrap up some release notes for 0.43.0 (or 0.43.1, possibly technically required)? Aside of Python 3 migration, lots of other features and fixes have been done, l10n, S3 upload, motion 4.4/4.5 support, and lots more.

MarcelCoding, 2022-12-23:
You can generate release notes with GitHub, when creating a release. You might just have to modify it a bit. It is basically a list of all PRs that were merged.

MichaIng, 2022-12-27:
At least for this release this would be really messy, due to quite some back and forth and non-linear merges, also from the previously dedicated l18n/i10n project. But for the future it makes sense indeed. We should then follow some consistent PR/commit naming, which, AFAIK, allows even automatic sorting (bug, enh, feat, frontend, backend or so), isn't it?

MichaIng, 2022-12-28:
Probably someone of you have a good idea or opinion about #2675:
- As `dev` currently has `0.43.0` as version string, `0.43.0b1` or similar would be a downgrade, so it cannot be easily migrated from `dev` branch to PyPI pre-release without defining the version string explicitly.
- Using `0.43.1b1` would do, or is there another clever way?

spupuz, 2023-01-29:
is it the dev image for docker the latest one? why it show always 0.43.0?

MichaIng, 2023-01-29:
We currently do not increase the version string. Will do so once we start shipping a beta and finally stable release + related branch.

The Docker image from [here](https://github.com/motioneye-project/motioneye/pkgs/container/motioneye) is always based on the latest `dev` branch commit.

TiboTake, 2023-03-12:
I found a little Bug.
Wen i use German as lang i cant show the Stream URL. The Popup dosn´t work.

edurenye, 2023-05-30:
Hi, please take a look at https://github.com/motioneye-project/motioneye/issues/2775 Would be great to think about it before releasing multilingual.

don-dolarson, 2023-06-19:
Was intensively using/testing 0.43 "dev" release over a week now on RPi 3B + RPi OS Bullseye 64-bit along with cheap ribbon AZ-Delivery and Innomaker cameras for 10 bucks each (5mp OV5647 both). Works flawless. I've had a problem setting admin and user password from the web interface. Whenever I set 'em, changes wouldn't be saved and needed to manually edit /etc/motion or motioneye.conf (don't remember which of them holds this) to set passwords. Works like a charm. Thanks.

spupuz, 2023-07-03:
hello there with latest update i got errors like these:

```
  File "/usr/local/lib/python3.9/dist-packages/motioneye/mediafiles.py", line 1030, in get_current_picture
    image.thumbnail((width, height), Image.CUBIC)
AttributeError: module 'PIL.Image' has no attribute 'CUBIC'
   ERROR: 500 GET /picture/1/current/?_=1688373818539&width=0.81&_username=admin&_signature=443e749b9bcdec6243f49323d58cee95669091a6 (172.17.0.1) 0.70ms
   ERROR: module 'PIL.Image' has no attribute 'CUBIC'
Traceback (most recent call last):
  File "/usr/local/lib/python3.9/dist-packages/tornado/web.py", line 1786, in _execute
    result = await result
  File "/usr/local/lib/python3.9/dist-packages/motioneye/handlers/picture.py", line 52, in get
    await self.current(camera_id)
  File "/usr/local/lib/python3.9/dist-packages/motioneye/handlers/picture.py", line 110, in current
    picture = mediafiles.get_current_picture(
  File "/usr/local/lib/python3.9/dist-packages/motioneye/mediafiles.py", line 1030, in get_current_picture
    image.thumbnail((width, height), Image.CUBIC)
AttributeError: module 'PIL.Image' has no attribute 'CUBIC'
   ERROR: 500 GET /picture/5/current/?_=1688373818539&width=0.81&_username=admin&_signature=2e3b8a0be521b6dcfd14970ad72f59431a371b38 (172.17.0.1) 0.48ms
   ERROR: module 'PIL.Image' has no attribute 'CUBIC'
Traceback (most recent call last):
  File "/usr/local/lib/python3.9/dist-packages/tornado/web.py", line 1786, in _execute
    result = await result
  File "/usr/local/lib/python3.9/dist-packages/motioneye/handlers/picture.py", line 52, in get
    await self.current(camera_id)
  File "/usr/local/lib/python3.9/dist-packages/motioneye/handlers/picture.py", line 110, in current
    picture = mediafiles.get_current_picture(
  File "/usr/local/lib/python3.9/dist-packages/motioneye/mediafiles.py", line 1030, in get_current_picture
    image.thumbnail((width, height), Image.CUBIC)
AttributeError: module 'PIL.Image' has no attribute 'CUBIC'
   ERROR: 500 GET /picture/6/current/?_=1688373818539&width=0.81&_username=admin&_signature=f292d42212edd27766e2efb114caac9a6735ac0e (172.17.0.1) 0.41ms
   ERROR: module 'PIL.Image' has no attribute 'CUBIC'
Traceback (most recent call last):
  File "/usr/local/lib/python3.9/dist-packages/tornado/web.py", line 1786, in _execute
    result = await result
  File "/usr/local/lib/python3.9/dist-packages/motioneye/handlers/picture.py", line 52, in get
    await self.current(camera_id)
  File "/usr/local/lib/python3.9/dist-packages/motioneye/handlers/picture.py", line 110, in current
    picture = mediafiles.get_current_picture(
  File "/usr/local/lib/python3.9/dist-packages/motioneye/mediafiles.py", line 1030, in get_current_picture
    image.thumbnail((width, height), Image.CUBIC)
AttributeError: module 'PIL.Image' has no attribute 'CUBIC'
   ERROR: 500 GET /picture/3/current/?_=1688373818574&width=0.81&_username=admin&_signature=ce9cd1cd8c1c891ac9d2fa3f0ea9e75606517b0a (172.17.0.1) 0.64ms
   ERROR: module 'PIL.Image' has no attribute 'CUBIC'
Traceback (most recent call last):
  File "/usr/local/lib/python3.9/dist-packages/tornado/web.py", line 1786, in _execute
    result = await result
  File "/usr/local/lib/python3.9/dist-packages/motioneye/handlers/picture.py", line 52, in get
    await self.current(camera_id)
  File "/usr/local/lib/python3.9/dist-packages/motioneye/handlers/picture.py", line 110, in current
    picture = mediafiles.get_current_picture(
  File "/usr/local/lib/python3.9/dist-packages/motioneye/mediafiles.py", line 1030, in get_current_picture
    image.thumbnail((width, height), Image.CUBIC)
AttributeError: module 'PIL.Image' has no attribute 'CUBIC'
   ERROR: 500 GET /picture/7/current/?_=1688373818574&width=0.81&_username=admin&_signature=d95829b4a734f256ff59510a1aea672e0dc5a9a8 (172.17.0.1) 0.52ms
   ERROR: module 'PIL.Image' has no attribute 'CUBIC'
Traceback (most recent call last):
  File "/usr/local/lib/python3.9/dist-packages/tornado/web.py", line 1786, in _execute
    result = await result
  File "/usr/local/lib/python3.9/dist-packages/motioneye/handlers/picture.py", line 52, in get
    await self.current(camera_id)
  File "/usr/local/lib/python3.9/dist-packages/motioneye/handlers/picture.py", line 110, in current
    picture = mediafiles.get_current_picture(
  File "/usr/local/lib/python3.9/dist-packages/motioneye/mediafiles.py", line 1030, in get_current_picture
    image.thumbnail((width, height), Image.CUBIC)
AttributeError: module 'PIL.Image' has no attribute 'CUBIC'
   ERROR: 500 GET /picture/10/current/?_=1688373818574&width=0.81&_username=admin&_signature=defdba7286bf4396042eacd8836b26fd99b2f95e (172.17.0.1) 0.41ms
   ERROR: module 'PIL.Image' has no attribute 'CUBIC'
Traceback (most recent call last):
  File "/usr/local/lib/python3.9/dist-packages/tornado/web.py", line 1786, in _execute
[... 9 lines cut ...]
   ERROR: module 'PIL.Image' has no attribute 'CUBIC'
Traceback (most recent call last):
  File "/usr/local/lib/python3.9/dist-packages/tornado/web.py", line 1786, in _execute
    result = await result
  File "/usr/local/lib/python3.9/dist-packages/motioneye/handlers/picture.py", line 52, in get
    await self.current(camera_id)
  File "/usr/local/lib/python3.9/dist-packages/motioneye/handlers/picture.py", line 110, in current
    picture = mediafiles.get_current_picture(
  File "/usr/local/lib/python3.9/dist-packages/motioneye/mediafiles.py", line 1030, in get_current_picture
    image.thumbnail((width, height), Image.CUBIC)
[... cut ...]
  File "/usr/local/lib/python3.9/dist-packages/motioneye/handlers/picture.py", line 110, in current
    picture = mediafiles.get_current_picture(
  File "/usr/local/lib/python3.9/dist-packages/motioneye/mediafiles.py", line 1030, in get_current_picture
    image.thumbnail((width, height), Image.CUBIC)
AttributeError: module 'PIL.Image' has no attribute 'CUBIC'
   ERROR: 500 GET /picture/9/current/?_=1688373828803&width=0.81&_username=admin&_signature=c830acb9f9cdb00fd524833a7ad8306ab94f9e5b (172.17.0.1) 0.50ms
```

spupuz, 2023-07-03:
what can i do?

spupuz, 2023-07-03:
reconfigured and reimported the conf files for the cam and it worked.

jwhite, 2023-09-05:
Is there currently a way to verify that I am indeed running 'dev' and not the old "0.42.1"

I see above that the version string is not being updated.

zagrim, 2023-09-11:
@jwhite I don't know what you are referring to with "above", but I can confirm that the version show both in UI and in log is 0.43.0 for the current dev version.

jwhite, 2023-09-12:
@zagrim "above" refers to earlier in this thread.  Is it true that 0.43.0 could also signify the last release version that is a non 'dev' version?

zagrim, 2023-09-13:
I don't this it could, and I still do not see where that would be suggested. However, in the python2 branch, which is the based on the old dev branch before migration to Python3, we still have 0.42.1 in https://github.com/motioneye-project/motioneye/blob/python2/motioneye/__init__.py 
Whereas in current dev branch we have 0.43.0 there: https://github.com/motioneye-project/motioneye/blob/dev/motioneye/__init__.py

zagrim, 2023-12-06:
@brjhaverkamp @spupuz Regarding the changes: In the sidebar of this issue there is a linked milestone: https://github.com/motioneye-project/motioneye/milestone/1 which contains still plenty of open issues and some features that have been planned to be included in 0.43 release (libcamera support and Motion 4.4 support perhaps being the biggest ones). In the milestone view one can also see the closed issues (ie. completed changes in current dev branch).

nullr0ute, 2023-12-06:
be included in 0.43 release (libcamera support and Motion 4.4 support perhaps being the biggest ones). In the milestone view one can also see the closed issues (ie. completed changes in current dev branch).

I think some of those big features like libcamera are great but could also be a 0.44 release. With the last release not supporting python3 and py2 being EOL 4 years ago personally I think splitting the release into two would be useful for a lot of users. Perfect is the enemy of good :)

zagrim, 2023-12-07:
Yeah, considering the time it has taken for 0.43 to get to its current state (with also Motion 4.4 support supposedly working but not likely very thoroughly tested), it might be a good idea to split out some of the stuff from that milestone (especially libcamera support seems to be a tough one). I wonder what @MichaIng thinks of that.

Nonetheless, recently there hasn't been much any progress here due to lack of time/energy from anyone (myself included) to move this forward, and getting 0.43 released would require some effort just for the releasing related things.

brjhaverkamp, 2023-12-08:
Hi Zagrim,

Thanks for your thoughts. I see you are also contemplating to release. From my side I would motion (pun intended) to indeed move what is not ready yet as much as possible (libcamera, Motion 4.4) to milestone 0.44 and release 0.43 as soon as possible. 
The move to python3 is a big step that will be appreciated by a lot of users. 

Indeed there hasn't been a lot of progress in the past months. A release might even draw some new people to add to the effort. 
That would be good for the project in general. Atm, I feel a lot of users are currently holding back and sticking to the last stable release ( myself included) 

@Michalng, I am curious to hear your thoughts. 

Bert

MichaIng, 2023-12-21:
Sorry for being mostly inactive lately. I am busy with my main project, the limited spare time I can afford.

motion 4.4 up to 4.6 should work well. All options motionEye sets are adjusted for these versions, and there were no changes after v4.4. But when custom options were added, those need to be updated manually to match the new motion version. Here is a nice overview: https://motion-project.github.io/motion_config.html#Configuration_OptionsAlpha

motionEye 0.43 is not perfect at all, but already much better than the last 0.42 release IMO. Alone Python 3 support, where Python 2 is often not available anymore on modern distros, is a huge benefit, and many bug fixes have been made as well, at least much more than new bugs introduces 😉. Hence, to being things forward, I agree that we should push at least a beta/pre-release to PyPI, to make installing it easier and get more attention. Then at best focusing on fixing the the most urgent bugs, ignoring new features/enhancements for now (unless someone has time and passion to work on a specific topic, of course), and get a "stable" (formally) release out soon as well.

I prepared a PyPI release workflow a while ago: https://github.com/motioneye-project/motioneye/pull/2675
The only question I was waiting for feedback about:
- The `a1` version suffix automatically leads to a release marked as "pre-release" on PyPI, and requires the `pip --pre` flag to be picked (unless the version is defined explicitly). That is perfect as a first step.
- Pre-releases however logically is handled as lower version than the final release without the suffix. So if one has "0.43.0" installed currently, "0.43.0a1" would not be pulled with `pip install -U --pre motioneye`.
- Shall we hence bump the version string to "0.43.1a1" or "0.43.1b1" to make the upgrade easier for those which installed "0.43.0" from the dev branch currently?

In the same turn, I would create a "beta" branch from the "dev" branch and a pre-release tag here on GitHub. The release workflow could then be triggered not only manually, but also automatically once a GitHub release tag is created.

brjhaverkamp, 2023-12-23:
Hi MichaIng

I saw you were very active this year on github in other projects. Too much to do.. Hop you have some time to spare for this. i can't oversee the intricacies of the alpha and beta naming unfortunately. From who do you expect/hope for feedback?

But if this is a more commonly used scheme, I would propose to just move forward with it. #2675 Is exactly a year old today, if I read correctly. So you can argue that if someone objected or had comments, there was ample time.

I am not very versed in github nor programming, but I am a tinkerer and hobbyist using motioneye(os). Is there is anything I can do to help?
Regards,

Bert

zagrim, 2023-12-23:
I don't have any specific experience on which to rely on for this versioning discussion, but I think it's perfectly fine to skip over 0.43.0 in official releases if that makes it easier to move to a useful versioning strategy with alphas and betas etc and still supporting easy upgrades from all the people who've already installed 0.43.0 from the dev branch.

Unless anyone can state any reason why NOT to do so, IMO releasing 0.43.1a1 (or...b1) would be ok as the next step. And as for branching, whatever makes it easy to do pre-releases when needed :smiley:

MichaIng, 2023-12-23:
Okay, and you are right of course, that going forward at some point is better than waiting forever for feedback 😄. Tonight I'll do an `0.43.1a1` to `test.pypi.org` as a final test for the upload/release workflow, then dump to `0.43.1b1`, update the README (install via `pip install --pre motioneye`), branch off a `beta` branch and do an upload/release to `pypi.org`.

MichaIng, 2023-12-24:
Pre-release done on GitHub: https://github.com/motioneye-project/motioneye/releases/tag/0.43.1b1

MichaIng, 2023-12-24:
Can someone help me with the Docker workflow: https://github.com/motioneye-project/motioneye/blob/dev/.github/workflows/docker.yml
- `Warning: 0.43.1b1 is not a valid semver. More info: https://semver.org/`: https://github.com/motioneye-project/motioneye/actions/runs/7315066439
    We would need to add a hyphen before potential `(a|b|rc)[0-9]` suffixes. But where does the `metadata-action` actually get the version string from? Just from the Git branch or tag? Not sure how to solve this when we need a hyphen for Docker but must not have a hyphen for PyPI. But I guess there is a way to override the string. Or we just do not generate versioned Docker containers for pre-releases, but keep only updating the `edge` release and in case add a `beta` one?
- We currently hardcode `type=edge,branch=dev`, which we would want to adjust based on the branch/tag it was pushed to? Or are these `type=` lines pattern which, if they match the branch or tag of the GitHub event, create a respective Docker release tag? Like:
    - `type=edge,branch=dev`: If GitHub branch matches `dev`, apply Docker tag `edge`.
    - `type=semver,pattern={{version}}`: Else, create semantic version tag from the Git tag.
    - `type=semver,pattern={{major}}.{{minor}}`: And one from only major.minor.
    - And we hence could add `type=beta,branch=beta` or something like that?

nullr0ute, 2023-12-31:
>     * `Warning: 0.43.1b1 is not a valid semver. More info: https://semver.org/`: 

Maybe just do it as 0.43.1 and if there's bugs to be fixed just do .2 and .3 etc

MichaIng, 2023-12-31:
Done already 😉. For now we'll just raise `b1` => `b2` etc until we think it is fine to do a stable release. Then as you said, increasing the patch version.

spupuz, 2024-03-09:
I still continue to get the same version each time it update

MichaIng, 2024-03-09:
The latest pre-release is `0.43.1b1`: https://pypi.org/project/motioneye/#history
Assure to use the `--pre` flag with `pip` to pull pre-releases.

spupuz, 2024-03-09:
I'm using docker

MichaIng, 2024-03-09:
This image? https://github.com/motioneye-project/motioneye/pkgs/container/motioneye
Indeed this is edge version, rebuilt on every merged commit to `dev` branch, while we raise the version string only when merging to `beta` or `main` (release) branch.

spupuz, 2024-03-09:
Ok I'm on edge so this is why I see the same version.

ETE-Design, 2024-11-15:
@MichaIng Any chance you can update MotionEyeOS to latest Beta? Also will it work with Raspberry Pi 5 - Even better with the AI Hat+?

brjhaverkamp, 2024-11-24:
Indeed, I am hoping for a new version of Motioneye. Motioneyeos, maybe a second step. Is there any progress? I have a feeling the code is 99% there.

Bert

ETE-Design, 2025-10-31:
@MichaIng Can I ask you to update MotionEYE OS to work for Raspberry PI 5 at the same time?

[2 more replies]