motioneye-project/motioneye #2375: Move i18n/l10n sources out of motioneye/

MichaIng, 2022-03-21:
Currently all sources and build scripts for localisation are within the `motioneye/` sub directory and most of them shipped with the Python package: https://github.com/motioneye-project/motioneye/blob/dev/setup.cfg

I suggest to move everything which is not required for runtime out of the `motioneye/` sub directory. Probably it can be moved into parent `locale/` instead? As far as I understand, only the JSON files from [here](https://github.com/motioneye-project/motioneye/tree/dev/motioneye/static/js) are required, isn't it?

If someone wants to add/update l10n, then a commit/PR needs to be done anyway, which means the whole repo needs to be clone anyway. So having all this in the Python package seems unnecesarry.

@jmichault 
As far as I understand, only the [Makefile](https://github.com/motioneye-project/motioneye/blob/dev/motioneye/Makefile) needs to be adjusted, probably one of the other executed script? Do you have any other objections on this or would it not work for reasons I've overlooked?

jmichault, 2022-03-22:
The mo files in locale/$LANG/LC_MESSAGES/ are also needed. They are used by python code.

MichaIng, 2022-03-22:
> The mo files in locale/$LANG/LC_MESSAGES/ are also needed. They are used by python code.

Ah okay, so the json files for the frontend and mo for the backend.

> I think the only file unnecessary in the Python package is scripts/po2json

What about:
- [Makefile](https://github.com/motioneye-project/motioneye/blob/dev/motioneye/Makefile)
- [traduki_po.sh](https://github.com/motioneye-project/motioneye/blob/dev/motioneye/traduki_po.sh)
- [traduki_python.txt](https://github.com/motioneye-project/motioneye/blob/dev/motioneye/traduki_python.txt)
- [traduko.sh](https://github.com/motioneye-project/motioneye/blob/dev/motioneye/traduko.sh)
- the `locale/*/LC_MESSAGES/*.po` files?

jmichault, 2022-03-22:
>
> 
> * [Makefile](https://github.com/motioneye-project/motioneye/blob/dev/motioneye/Makefile)
> * [babel.cfg](https://github.com/motioneye-project/motioneye/blob/dev/motioneye/babel.cfg)
> * [traduki_po.sh](https://github.com/motioneye-project/motioneye/blob/dev/motioneye/traduki_po.sh)
> * [traduki_python.txt](https://github.com/motioneye-project/motioneye/blob/dev/motioneye/traduki_python.txt)
> * [traduko.sh](https://github.com/motioneye-project/motioneye/blob/dev/motioneye/traduko.sh)
> * the `locale/*/LC_MESSAGES/*.po` files?

They are not include in the package. The setup.cfg file doesn't include them.

MichaIng, 2022-03-22:
Ah I see. Could still make sense to move them out of `motioneye/` to also visually have a clear separation between runtime files and build/development files.

MichaIng, 2022-03-22:
> AFAIK `locale/` files should stay in `motioneye`.

Yes, should be fine, also to not have the `locale/LC_MESSAGES/*/` directory structure doubled once inside and once outside 👍.

jose1711, 2022-03-23:
Why is the template file in Italian and not English? Is this going to be changed at some point?

MichaIng, 2022-03-24:
This it not Italian, but [Esperanto](https://en.wikipedia.org/wiki/Esperanto), a constructed international auxiliary language meant for exactly such task. It is easy and fast to learn. But I agree that for inviting contribution to motionEye it is a blocker, as long as the by far more widespread and de facto standard language for international collaboration is English. I just stumbled above it myself in another PR, where I skipped adding the i18n strings to avoid learning Esperanto first or using translators to turn the English sentence into Esperanto. It also makes it harder to quickly identify HTML elements and messages. Another, IMO, design flaw of Esperanto is that it uses certain accents, which cannot easily be entered on common keyboards.

@jmichault 
I know it would somehow break your initial i18n idea, which is btw awesome (I'm gonna add a language selector to the GUI the next days!), but what do you think about switching to English and making Esperanto an l10n language instead? Not sure whether you use or know a tool which is able to automatically perform this switch between the template language and an l10n language across all locale, frontend and backend files, otherwise we can just do it gradually when changing or adding new strings. This would definitely help contributors and localising some missing strings.

closed 2022-04-05 by MichaIng