motioneye-project/motioneye #2378: Ability to upload videos to google drive teamdrive (shareddrives)

Bartixxx32, 2022-03-22:
Hello, i seen that motioneye is able uploading to google drive, but i tried put my teamdrive path, but it just creatid this patch in /root of my drive. Is any option to upload it to teamdrive?

MichaIng, 2022-03-22:
I do not use Google Drive, but just to assure that it generally works as expected without Google Team Drive:
- The path you enter in `Location` is correctly used for uploads to your Google Drive, or is it always prefixed with `/root`, i.e. you enter `/test` and files land in `/root/test`?
- I see some `parent_id` variable in the code, being set to `root`, used also in the API URLs: https://github.com/motioneye-project/motioneye/blob/16fd474/motioneye/uploadservices.py#L421-L444  
    This was added together with the Google Drive upload option, i.e. was like this right from the start: https://github.com/motioneye-project/motioneye/commit/0964cc3
- Not sure whether it is intended to make it more unlikely that existing files are overwritten. Or is `/root` a hardcoded parent path in Google Drive where all personal files land, while there is a different parent path for Google Team Drive? In this case we could add a selection/slider whether to use the one or the other.

Bartixxx32, 2022-03-24:
Seems to be hardcoded becouse to access teamdrive im pretty sure only way is to use folder id ( maybe that would be better idea than specify path?) Just put directory id to the form and done fast and easy, and always accurate, it wont be messed with foldernames but it will point to id :)

MichaIng, 2022-03-24:
Is `root` probably a folder ID as well, the default one which is always available? And other folder IDs can be freely chosen, or are they generated for every directory on Google Drive? Probably there is even an API to derive them to we can add a selection instead of an input box for it.