motioneye-project/motioneye #2379: notifications but only on first event

bulgin, 2022-03-22:
Hello.  Love motioneye.  However, the notifications settings has no feature to send only x amount of notifications per x amount of time.  I would use it but don't want to receive a multitude of notifications every time there is an event, which can be many depending on the setup.

is there a way to limit the number of notifications?

Thanks

MichaIng, 2022-03-23:
Currently I do not see this being possible. But generally it makes sense. Probably we can add a setting to define a number of seconds (or minutes?) until which no new event action is triggered, which is separately. I'm just not sure jet how to do so for motion start and end events, e.g.:
1. motion starts
2. motion ends after 29 seconds
3. motion start after 5 seconds
4. motion ends after 5 seconds

Let's say we do not want multiple notifications within 30 seconds:
- If this is applied for start and end events as shared timer, we would have both start actions triggered, but no end action.
- So better is it to have separate timers, but in this case, we'd have the first start and end actions triggered, then the second start action, but not the second end action.

Generally we would want to have an end action triggered for every start action which is triggered, just in case a command or web hook is used to set some flag or so. So we could instead set a flag to prevent the end event action if the start even action was skipped, so only a timer for the start even is required. But again this may be bad:
1. motion starts
2. motion ends after 5 seconds
3. motion starts after 5 seconds
4. motion ends after a very long time of significant motion

In the above case, with a start even timer only, we'd get a notification for first start and end events, but none for the second way more significant motion event. So if you get the motion end notification and think "well, this was short, probably a bird flying by", but you do not get any notified about the significant second event.

We could say, well, if the options are well explained, the admin knows the limitations and how to deal with it, but if we implement a new feature, it should be as usable as possible, especially we should avoid introducing the security-relevant risk that significant motion events are completely missed, breaking the whole idea behind it.

A simple solution would be to only allow setting this silence duration if no end event action is configured.

bulgin, 2022-03-23:
> y-To: motioneye-project/motioneye <
> motioneye-project/motioneye ***@***.***>Cc: bulgin <
> [motioneye-project/motioneye] notifications but only on first event
> (Issue #2379)Date: Tue, 22 Mar 2022 17:15:13 -0700

Or maybe a link in the message(s) that if pressed stops the
notifications. . . kind of like a reverse webhook.  But I guess that
would mean another open port?

MichaIng, 2022-03-23:
More importantly it would mean an authentication is required, but otherwise the HTTP API is there can could be used for such. But the question remains for how long notifications should stop and whether this should include end events.

bulgin, 2022-03-23:
What exactly is an "end event"?

(Issue #2379)Date: Tue, 22 Mar 2022 17:54:12 -0700

MichaIng, 2022-03-23:
> What exactly is an "end event"?

When there is no motion anymore. You can define a time in the GUI to define the time after which, if no motion happened, this event is triggered, along with the configured command (and from next release on web hook).

bulgin, 2022-03-23:
So in theory this means that an event occurs, there is a 30 second
pause in motion thus triggering the webhook because the admin put in 30
seconds as the period of silence?  If so that seems like a reasonable
way to send one notification.  Yet there may be and depending on the
area being monitored, many such delayed events.  I think anyone
monitoring an area will generally know if the activity is unusual or
normal. In fact, when you think about it, an alert is designed to get
the owner's attention, not nag them.  Once you have their attention,
they should be able to stop the alerts for a period of time, otherwise
users just won't use the feature because who wants to receive notice
after notice for activity?

(Issue #2379)Date: Tue, 22 Mar 2022 18:01:03 -0700

MichaIng, 2022-03-23:
> Once you have their attention, they should be able to stop the alerts for a period of time,

The 30 seconds was just an example. It makes sense to make this period configurable. And the question remains what to do about the end event, for cases where both are used and users either expect or the configured commands/hooks require an end event to be triggered if a motion start event was triggered before.

MarcelCoding, 2022-04-05:
I have kind of the same problem, I currently getting like 50 emails every day

MarcelCoding, 2022-04-05:
A dbeouce for the start and a min delay from the last event from x minutes. If between the end of the event and the end of the delay another motion is detected the end event notification is canceled.

MichaIng, 2022-04-05:
We basically need a "mute motion" flag:
- One can configure X minutes to skip motion event actions, after an end event happened, which sets "mute motion" to true, or sets it to the expiry time (serving as flag and timestamp the same time).
- While the "mute motion" flag is set, all motion start (and end) event actions are skipped.
- When the configured timeout has been reached, respectively a motion start event sees that the current time is later than the "mute motion" timestamp, it is set back to false.
- Only when "mute motion" is set to false, detected motion triggers the event actions.

It does not eliminate the issue that significant/long motion events can be missed when insignifikant/short motion events happened before, but when this is clearly documented, admins should be able to evaluate the implications for their use case.

MichaIng, 2022-04-06:
> and if a new motion is detected the timer should be reset (start from the beginning)

I'm not sure whether this is a good idea. It means that in theory one might never get any notification if events happen too regularly. IMO it is more intuitive to have the timeout to define an acceptable amount of notifications per time, so after it expired, a new notification can be sent. But of course a valid point of discussion. If there are other opinions, we'll implement it how to majority thinks it's reasonable.

bulgin, 2022-10-11:
Here's an idea: provide a webhook in the email that if pressed places a
mask over the entire camera view. . . and another webhook to remove it.

(Issue #2379)Date: Tue, 22 Mar 2022 18:26:48 -0700