motioneye-project/motioneye #2382: "Stream is closed" error due to browser power savings?

MichaIng, 2022-03-22:
When I have the video stream opened in browser and switch to another tab, after ~30 seconds, motionEye prints the following error:
```
Mar 23 00:45:53 VM-Bookworm meyectl[667]:    ERROR: mjpg client for camera 1 on port 9081 error: Stream is closed
Mar 23 00:45:53 VM-Bookworm meyectl[667]: Traceback (most recent call last):
Mar 23 00:45:53 VM-Bookworm meyectl[667]:   File "/usr/local/lib/python3.9/dist-packages/motioneye/mjpgclient.py", line 282, in _on_before_content_length
Mar 23 00:45:53 VM-Bookworm meyectl[667]:     future.result()
Mar 23 00:45:53 VM-Bookworm meyectl[667]: tornado.iostream.StreamClosedError: Stream is closed
```
I guess it is some power savings feature, i.e. the browser closes the stream after a while when the tab is idle/not shown, which generally seems reasonable. When I switch back to the tab, it takes a second or two for the camera stream to show up again, which is fine.

Functionality is not affected, but the errors spam the log. Probably we can mute them, if it is really the client which closes the stream? Or can you think of any reason where we need a server side error when the client/browser closes the stream for some reason?

zagrim, 2022-04-10:
I didn't remember this, so many issues opened and closed lately. But I seem to get plenty of those also when I start ME backend locally with the browser tab open, it goes on for a 1-2 seconds and then stops (before any camera frame is shown). So it seems to kind of match the way things go with the backgrounded tab.

> any reason where we need a server side error when the client/browser closes the stream for some reason?

If it is the client that closes the stream, likely there's no reason to have that error logged. But I guess it would take some testing with different network related error scenarios to make sure muting that error on that spot doesn't hide anything that would have some value in debugging a setup.

MichaIng, 2022-04-10:
Probably it is possible to derive from the exception type whether the client closed the stream, or probably we can track an event when the client closed the stream to abort the loop in the backend instead of calling `future.result()` again.

tunkio, 2022-04-30:
This stream is closed error occurs when noone is requesting images for camera view(webui). Currently it is configured so that if there is no requests in 10 seconds (MJPG_CLIENT_TIMEOUT) stream is closed. (this causes error). https://github.com/motioneye-project/motioneye/blob/1ae38313678fafdb12b3592f4fdf29df3c4b8d75/motioneye/mjpgclient.py#L424

This error occur also when restarting motioneye. mjpgclient will be called when webui is loaded and it takes while to initialise mjpgclient(stream). During that time there will be lot's of these streaming errors.

We could catch that error, tornado.iostream.StreamClosedError, and log it as debug. This would hide the streaming errors which occurs when you restart motioneye and access webui for the first time.

Someone has also added IOLoop to make lots of retries to get the latest picture. https://github.com/motioneye-project/motioneye/blob/1ae38313678fafdb12b3592f4fdf29df3c4b8d75/motioneye/handlers/picture.py#L117

There is quite nice amount of get_current_picture calls going on. Someone tried to remove that which maybe caused problems because it that is restored in later commit. Do you know nothing about this? Checking for existing clients in mjpgclients get_jpg, https://github.com/motioneye-project/motioneye/blob/ab23ae3675752f3ffa692a6ee449b16d5f2f39ce/motioneye/mjpgclient.py#L339, won't work properly at startup. I think those loops don't share same object, so that list is quite long empty. Every call goes through that initialization. For some reason after the stream is initialized only one instance is returning an image to the requester.

MichaIng, 2022-04-30:
Thanks for starting investigation. The exact error log is done here: https://github.com/motioneye-project/motioneye/blob/1ae38313678fafdb12b3592f4fdf29df3c4b8d75/motioneye/mjpgclient.py#L144-L150

This `_error()` method is called in all the internal `_on_*`/`_check_*`/`_seek_*` functions when a `result()` call fails. So I guess there are either within an internal loop or external global method calls still connection attempts done after the connection was closed. Probably it is possible to avoid the errors in the first place by setting flags, or the `on_close` callback, or doing `closed()` checks for exiting loops.

> There is quite nice amount of get_current_picture calls going on. Someone tried to remove that which maybe caused problems because it that is restored in later commit. Do you know nothing about this?

This was before I joined the project. But this IOLoop does not imply any mjpg client access, it just checks whether/waits until the above single `mediafiles.get_current_picture()` call did already return/assign a value, isn't it? So at least based on my surely limited Python understanding it shouldn't cause this error. Also, at least I do not see the error on any startup or "return to camera browser tab" actions, I only see it when leaving the camera tab and then, what quite fits, after ~10 seconds, when the last access timeout was reached as the non-active browser tab stops sending requests.

Not directly related, but in the _garbage_collector it is first checked for frame timeouts and afterwards for last access timeouts. Would it make sense the other way round, to not possibly print an error about a frame timeout when the connection is about to be closed due to last access timeout anyway?

tunkio, 2022-04-30:
> Thanks for starting investigation. The exact error log is done here: 

Yes, I started to add that StreamError exception catcher with debug logs but then I debugged little further and stopped due we need to check if there is something else.
https://github.com/tunkio/motioneye/blob/dd00f9cc40d3c28f91ef9b91ecc77c35e976bc8a/motioneye/mjpgclient.py#L145-L155
https://github.com/tunkio/motioneye/blob/dd00f9cc40d3c28f91ef9b91ecc77c35e976bc8a/motioneye/mjpgclient.py#L207

>  Also, at least I do not see the error on any startup or "return to camera browser tab" actions, I only see it when leaving the camera tab and then, what quite fits, after ~10 seconds, when the last access timeout was reached as the non-active browser tab stops sending requests.

I get lot's of StreamClosedError after motioneye restart and first time when I go to webui.
```
Apr 30 19:01:20 survnas meyectl[192472]:     data = future.result()
Apr 30 19:01:20 survnas meyectl[192472]: tornado.iostream.StreamClosedError: Stream is closed
Apr 30 19:01:21 survnas meyectl[192472]:    ERROR: mjpg client for camera 1 on port 9081 error: Stream is closed
Apr 30 19:01:21 survnas meyectl[192472]: Traceback (most recent call last):
Apr 30 19:01:21 survnas meyectl[192472]:   File "/usr/local/lib/python3.8/dist-packages/motioneye/mjpgclient.py", line 198, in _on_http
Apr 30 19:01:21 survnas meyectl[192472]:     data = future.result()
Apr 30 19:01:21 survnas meyectl[192472]: tornado.iostream.StreamClosedError: Stream is closed

```

I added some debug and full mjpgclient initializations are done
> This was before I joined the project. But this IOLoop does not imply any mjpg client access, it just checks whether/waits until the above single mediafiles.get_current_picture() call did already return/assign a value, isn't it? 
No. I added following debugs to see if those IOLoops goes directly to the mmjpgclient get_jpg and that initialization:

```
mjpgclient.py
def get_jpg(camera_id, retry=False):
    if camera_id not in MjpgClient.clients:
        with open("/tmp/countfile", "a") as f:
            f.write(f"get_jpg call without existing camera. Camera_id {camera_id} IOLoop retry number {retry}\n")
        # mjpg client not started yet for this camera
```

That count file contains 265-330 lines, it varies between restarts. What I do I restart motioneye, start web browser and keep it active until camera view appears. Added retry count into calls also that I can see them coming into the get_jpg. Below is few lines.

```
get_jpg call without existing camera. Camera_id 1 IOLoop retry number 5
get_jpg call without existing camera. Camera_id 1 IOLoop retry number 0
get_jpg call without existing camera. Camera_id 1 IOLoop retry number 11
get_jpg call without existing camera. Camera_id 1 IOLoop retry number 22
get_jpg call without existing camera. Camera_id 1 IOLoop retry number 17
get_jpg call without existing camera. Camera_id 1 IOLoop retry number 6
get_jpg call without existing camera. Camera_id 1 IOLoop retry number 1
get_jpg call without existing camera. Camera_id 1 IOLoop retry number 12
get_jpg call without existing camera. Camera_id 1 IOLoop retry number 23
```
Like I said there is hundreds of calls. If I remove that IOLoop get_jpg call and initializations drops to 34.

```
# picture is not available usually when the corresponding internal mjpeg client has been closed;
# get_current_picture() will make sure to start a client, but a jpeg frame is not available right away;
# wait at most 5 seconds and retry every 200 ms.
if not picture and retry < 25:
    return None
```
In main.js
https://github.com/tunkio/motioneye/blob/0d2f353eb8e404e9002a6a1465e83ba264efbcc8/motioneye/static/js/main.js#L5385-L5388

We drop fps to 1 when there is no picture available. That I dropped to 0.5 in my planned fix which reduces requests lot too. Specially when you leave that IOLoop intact. 

Like I said that check for existing cameras in mjpgclient don't work.
https://github.com/tunkio/motioneye/blob/0d2f353eb8e404e9002a6a1465e83ba264efbcc8/motioneye/mjpgclient.py#L339

This IOLoop thing is new to me so it takes some time to find out should we do something with this or leave it as it is.

tunkio, 2022-04-30:
That might be normal behaviour if we close stream after 10 sec. Following is in Tornados document:

All outstanding Awaitables will resolve with a a [StreamClosedError](https://www.tornadoweb.org/en/stable/iostream.html#tornado.iostream.StreamClosedError) when the stream is closed

https://github.com/motioneye-project/motioneye/blob/ab23ae3675752f3ffa692a6ee449b16d5f2f39ce/motioneye/mjpgclient.py#L441

MichaIng, 2022-04-30:
So it would be cleanest to add a "closed" flag before calling `close()` and check for that flag to exit and await/request loop, respectively skip the call?

> No. I added following debugs to see if those IOLoops goes directly to the mmjpgclient get_jpg and that initialization:

Ah okay, so not the result/return is stored in the `picture` variable, but the function call itself? Looks I need to learn more about Python 😅. So probably, with above flag, or a method that already exists, we can check whether the connection is actually still/already open before doing the `get_jpg` call here: https://github.com/motioneye-project/motioneye/blob/1ae3831/motioneye/mediafiles.py#L1001

But yeah, while I'm aiming for a clean solution that does not involve suppressing errors but avoiding invalid calls in the first place, that might be a larger task. So suppressing `StreamClosedError` is fine for me for now, to at least remove the spam from logs.

tunkio, 2022-05-01:
Ok, it seems that request storm is happening then when you start motioneye server and there is someone already polling the server. If I wait while before loading webui evertyhing goes smoothly! Well this is good, then we just need to fix that other thing and maybe if I find fix for this also(not so important)

MichaIng, 2022-05-03:
Solved in `dev`.

closed 2022-05-03 by MichaIng