motioneye-project/motioneye #2387: Request to add sound to the video

peterd550, 2022-03-25:
Can we please add sound to the video

MichaIng, 2022-03-25:
Many thanks for your request.

Could you go more into detail, which kind of video do you mean, for locally attached cameras which have a microphone, or remote/IP cameras or both? And shall the audio stream be in the stored media file only, or played on the web interface as well?

I guess with V4L2 utilities is is possible to capture both for a local camera easily (when available) and store it, in case after FFmpeg transcoding/encoding. But it is not so trivial to pass both to the web interface. Currently we use an MJEG picture stream, naturally no audio, to the browser, so that would required a larger rework I guess.

peterd550, 2022-03-27:
Thanks
My cameras all record sound - to the SD card.
What I am looking for is when the camera records with MotionEye to my NAS to have the sound from the camera.
At the moment it is just video with no sound

MichaIng, 2022-03-27:
> My cameras all record sound - to the SD card.

You mean via motionEye as well? I.e. you mean recording via motionEye to a local directory contains sound and doing so with NAS upload contains no sound? Hmm, I'd need to have a look into the code in case/why this is handed differently for those two cases then. I'm new to the project, just if you wonder 😉.

zagrim, 2022-03-28:
I don't think Motion supports audio at all (see https://github.com/Motion-Project/motion/issues/384, https://github.com/Motion-Project/motion/discussions/136). However, there are some ways to get around it by either recording audio separately and then combining audio+video, or using some Motion parameters (see https://github.com/motioneye-project/motioneye/issues/1112, https://www.lavrsen.dk/foswiki/bin/view/Motion/SoundAudioRecording).

peterd550, 2022-03-28:
So can you add that audio script to motion eye?

MichaIng, 2022-03-28:
Generally, `/dev/video0` and alike seem to provide an audio stream as well, when provided by the camera. So if FFmpeg is used, it should be possible have it just included as is But looks like the `motion` maintainer explicitly denies to add audio support, since it is often illegal to record audio without consent and goes beyond the scope of what `motion` aim to do: To detect visual motion, not sound.

But probably I find time to check where/how motionEye catches the media stream and where we would be able to at least leave audio inside, when it is there already. But this has no high priority to me, so may take a while, if no one else is having a look into the code earlier 🙂.

peterd550, 2022-03-28:
Thanks that would be good!
FYI - in Australia it is fine to video with sound as it's for your own use.
If it was brought to court you'll need to get permission to submit it.
Comes under the surveillance act.

Same with the video.
Don't ask me how I know that :(

tv21, 2023-07-06:
The Reolink Cameras all provide sound over the RTSP stream, can this be added to the recordings?

tv21, 2023-07-06:
https://github.com/meirlazar/motion_with_audio_arecord shows a bash script for adding audio to Motion but I do not understand the install instructions.  But assuming it works, could something like this be incorporated into MotionEye?

JamBalaya56562, 2026-11-15:
motion, which records the movies for motionEye, has no audio support, and as linked above its maintainer declined to add it (Motion-Project/motion#384, Motion-Project/motion#136). MotionPlus is the project that records audio. So motionEye can't add it on its side. Closing as not planned.

Marijn0, 2026-11-15:
Actually, MotionPlus has now been merged into the Motion repository, so Motion 5.0 will include the features previously available in MotionPlus, including audio recording. Additional features are also planned, as listed here: https://github.com/Motion-Project/motion/discussions/1847

I'm reopening this issue for now. We'll just have to wait until Motion 5.0 is released, which might take some time.