motioneye-project/motioneye #2391: Completely unable to change passwords

TonyDicken1964, 2022-03-26:
Hi

I have tried over and again to change the admin and surveillance password. I add new passwords, click refresh, clear the cache refresh and the old default admin password remains.

What am I missing?

Thanks

Tony

MichaIng, 2022-03-26:
Did you click "Apply" after entering the passwords? No refresh or cache clearing is required.

If so, can you please paste the service logs:
```sh
journalctl -u motioneye
```

TonyDicken1964, 2022-03-26:
Thanks, that worked fine. Can't understand why that didn't work up to now!

closed 2022-03-26 by TonyDicken1964

MichaIng, 2022-03-26:
One little quirk I also recognised is that the "Apply" button does not appear while you are typing the new password but only when you then select a different field. Would be actually better to show it once you type in something, so it cannot be overseen.

TonyDicken1964, 2022-03-26:
Totally agree.

⁣Get BlueMail for Android ​

***@***.***>