motioneye-project/motioneye #2393: Could not run curl-config: [Errno 2] | error: command 'aarch64-linux-gnu-gcc' failed with exit status 1 twisted

ong-andrew, 2022-03-28:
Platform: Raspberry Pi 4
Distro: 2022-01-28-raspios-bullseye-arm64-lite
motionEye version: 0.43.0
Motion version: 4.3.2
Cameras used: A bunch of very cheap generic network cameras from China.

I encountered errors when installing via: 
sudo python3 -m pip install 'https://github.com/motioneye-project/motioneye/archive/dev.tar.gz'

========

Message:

`__main__.ConfigurationError: Could not run curl-config: [Errno 2] No such file or directory`

Solution: 

`sudo apt install libcurl4-openssl-dev libssl-dev`

========

Message: 

`error: command 'aarch64-linux-gnu-gcc' failed with exit status 1 twisted`

Solution: 

`sudo apt-get install python3.9-dev`

========

I hope someone finds this useful.

PS: I love you ccrisan and new motioneye dev team.

MichaIng, 2022-03-28:
Many thanks for your report.

Those packages are all listed as [build dependencies in our README](https://github.com/motioneye-project/motioneye/tree/dev#installation) for all but ARMv6 and ARMv7 systems. You use an arm64 OS = ARMv8 kernel. Did you follow the README and were the instructions not clear enough?

MichaIng, 2022-04-06:
Marking this as closed, since the solution matches our documentation.

closed 2022-04-06 by MichaIng

jorgeaviles, 2024-02-16:
@MichaIng Does it matter that this OS is actually 32-bit ? `dpkg --print-architecture`= `armhf` which is 32-bit OS while the having a 64-bit ARM architecture `uname -m` = `aarch64`.