motioneye-project/motioneye #2398: No video, error 500 in the logs (python3/dev branch)

jose1711, 2022-03-30:
Using the latest `dev` branch on Arch Linux system with two Logitech webcams connected. For some reason I get no signal from either of the cameras (yet they work fine in e. g. `xawtv`). The log goes like this:
```
mar 31 00:44:15 myhost meyectl[60541]: TypeError: initial_value must be str or None, not bytes
mar 31 00:44:15 myhost meyectl[60541]:    ERROR: 500 GET /picture/1/current/?_=1648680255744&width=0.6&_username=admin&_signature=61975d6068055450a21851ff50a06871d73afd1a (127.0.0.1) 0.60ms
mar 31 00:44:16 myhost meyectl[60541]:    ERROR: initial_value must be str or None, not bytes
mar 31 00:44:16 myhost meyectl[60541]: Traceback (most recent call last):
mar 31 00:44:16 myhost meyectl[60541]:   File "/usr/lib/python3.10/site-packages/tornado/web.py", line 1704, in _execute
mar 31 00:44:16 myhost meyectl[60541]:     result = await result
mar 31 00:44:16 myhost meyectl[60541]:   File "/usr/lib/python3.10/site-packages/motioneye/handlers/picture.py", line 52, in get
mar 31 00:44:16 myhost meyectl[60541]:     await self.current(camera_id)
mar 31 00:44:16 myhost meyectl[60541]:   File "/usr/lib/python3.10/site-packages/motioneye/handlers/picture.py", line 110, in current
mar 31 00:44:16 myhost meyectl[60541]:     picture = mediafiles.get_current_picture(
mar 31 00:44:16 myhost meyectl[60541]:   File "/usr/lib/python3.10/site-packages/motioneye/mediafiles.py", line 1018, in get_current_picture
mar 31 00:44:16 myhost meyectl[60541]:     sio = io.StringIO(jpg)
mar 31 00:44:16 myhost meyectl[60541]: TypeError: initial_value must be str or None, not bytes
mar 31 00:44:16 myhost meyectl[60541]:    ERROR: 500 GET /picture/1/current/?_=1648680256770&width=0.6&_username=admin&_signature=311c4528402b9724d22c7f11c6f1f9880fe09818 (127.0.0.1) 0.60ms
```
Camera activity LED is on. Any idea?

MichaIng, 2022-04-06:
Looks like there is another bytes > string conversion missing. Those are regular V4L2/`/dev/videoX` cameras? I wonder why we never faced this so far.

I'll have a look into the code tomorrow.

Btw, would you mind to help adding Arch/pacman/pkg support to the new [motioneye_init > linux_init](https://github.com/motioneye-project/motioneye/blob/dev/motioneye/extra/linux_init) command for installing motion, v4l2-utils, FFmpeg and the motionEye service + config? Currently only DEB and RPM based distros are supported, but adding Arch (and probably Gentoo, although it uses a different package manager) would be great.

jose1711, 2022-04-07:
> Looks like there is another bytes > string conversion missing. Those are regular V4L2/`/dev/videoX` cameras? I wonder why we never faced this so far.
> 
> I'll have a look into the code tomorrow.

Thank you, yes. These both are standard V4L2 cameras, nothing fancy/special.

> Btw, would you mind to help adding Arch/pacman/pkg support to the new [motioneye_init > linux_init](https://github.com/motioneye-project/motioneye/blob/dev/motioneye/extra/linux_init) command for [...]

On Arch or Arch-based distros it's `pacman` package manager that should resolve and install all the package dependencies as well as to install other auxiliary files, like systemd units or configuration. I've recently become a co-maintainer of `motioneye-git` in AUR and updated the `PKGBUILD` so that it is working with the current `dev` branch. You may see the latest updates here: https://aur.archlinux.org/cgit/aur.git/tree/?h=motioneye-git

Really what Arch users are supposed to do (if using `yay` as AUR helper) is:
`yay -S motioneye-git`.

MichaIng, 2022-04-07:
Ah, I wasn't ware that there is a motionEye package in AUR. Great community I must admin 🙂.

`yay` is always available on Arch, or is it possible via `pacman` as well to install AUR packages? Once v0.43.0 has been released, we can then update the Wiki: https://github.com/motioneye-project/motioneye/wiki/Install-On-Arch

Btw on the [tmpfiles](https://aur.archlinux.org/cgit/aur.git/tree/motioneye.tmpfiles?h=motioneye-git):
- `/var/lib/motioneye` is automatically generated on service start thanks to `StateDirectory=motioneye`. So at least when systemd is used, it is not required to have it in tmpfiles.

And: https://aur.archlinux.org/cgit/aur.git/tree/PKGBUILD?h=motioneye-git#n40
- Does Arch really still use `/var/run`? On all non-legacy distro versions I know, `/var/run` is deprecated and only a symlink to `/run`, even systemd does noisy warnings when `/var/run` is used in any unit, and actively translates it to `/run`. I.e. that `sed` really shouldn't be required.
- While GitHub does redirects, you may want to update `https://github.com/ccrisan/motioneye` to `https://github.com/motioneye-project/motioneye`.

jose1711, 2022-04-07:
> `yay` is always available on Arch, or is it possible via `pacman` as well to install AUR packages? Once v0.43.0 has been released, we can then update the Wiki: https://github.com/motioneye-project/motioneye/wiki/Install-On-Arch

Sure, once there is a release tag I can adopt the `motioneye` `PKGBUILD` and make the updates needed.

AUR has no official support (https://wiki.archlinux.org/title/Arch_User_Repository) and I doubt `pacman` will ever receive an update that would integrate AUR in any way. As for the AUR helpers.. there are couple to choose from, I did use `yaourt` in the past but it is now abandoned so I switched to `yay` which seems to do the job.

> Btw on the [tmpfiles](https://aur.archlinux.org/cgit/aur.git/tree/motioneye.tmpfiles?h=motioneye-git):
> 
>     * `/var/lib/motioneye` is automatically generated on service start thanks to `StateDirectory=motioneye`. So at least when systemd is used, it is not required to have it in tmpfiles.
> 

Thank you, I removed that file.

> And: https://aur.archlinux.org/cgit/aur.git/tree/PKGBUILD?h=motioneye-git#n40
>
>     * Does Arch really still use `/var/run`? On all non-legacy distro versions I know, `/var/run` is deprecated and [...]

Yup, Arch also uses symlink. I think I just left that from the previous commits as I was updating the ticket. But as it's unnecessary I removed this also.

>     * While GitHub does redirects, you may want to update `https://github.com/ccrisan/motioneye` to `https://github.com/motioneye-project/motioneye`.

Fixed. Thank you for your valuable comments.

MichaIng, 2022-04-08:
Great. Okay, so I think we can assume `yay` as the correct modern AUR installer then, to put into our Wiki 🙂.

jose1711, 2022-04-08:
> Great. Okay, so I think we can assume `yay` as the correct modern AUR installer then, to put into our Wiki slightly_smiling_face.

Yes, I think it's a safe bet. Also, Arch users often are aware of what AUR helper they use (and just replace `yay` with whatever they have installed).

> Btw, I wonder what the different, or different aim of the motioneye and motioneye-git packages is. Was the latter only created to make use of the Python 3 (and now `dev`) branch instead of the latest (Python 2) release, i.e. so that `motioneye-git` becomes obsolete with next motionEye release? From our end it would be easier to have one AUR package instead of two, when dealing with user reports, and bundling forces shouldn't be a bad idea in general, as long as I do not miss another difference between both packages which remains a good reason to keep both slightly_smiling_face.

Having `PACKAGE` and `PACKAGE-git` in AUR is a common practice, see [arch wiki](https://wiki.archlinux.org/title/Arch_User_Repository#What_is_the_difference_between_foo_and_foo-git_packages?). `motioneye-git` will stay up even if `motioneye` is updated to the latest stable. That means that users of Arch often have a choice between living on the edge (with `-git`, `-vcs`, `-hg` versions of the packages) and sticking to a stable version with an expectation of better stability. From packaging perspective both `PKGBUILD` should not do things radically different.

MichaIng, 2022-04-09:
Okay, thanks for clarification, makes sense. `motioneye` non-git can point to `main` branch then, once v0.43.0 has been released. So far contributors voted for switching from "master" to "main" as name for the primary branch, which will always match the latest stable release on GitHub and PyPI then.

MichaIng, 2022-04-13:
I tried to understand where this bytes, which shall be a string, are coming from:

https://github.com/motioneye-project/motioneye/blob/16fd474/motioneye/mediafiles.py#L1010-L1018
- `io.StringIO()` takes a string as input to create a file-like object from it. There question may arise why a JPEG is assumed to be a string in the first place.

The `jpg` is returned here: https://github.com/motioneye-project/motioneye/blob/16fd474/motioneye/mjpgclient.py#L338-L374

Calls this method: https://github.com/motioneye-project/motioneye/blob/16fd474/motioneye/mjpgclient.py#L98-L100

`_last_jpg` is assigned here: https://github.com/motioneye-project/motioneye/blob/16fd474/motioneye/mjpgclient.py#L316-L322
So it seems to be what tornado is getting from motion.

The question is why this is assumed to be a string, why this is true in most cases (I never faced this issue) but not in yours, even depending on the browser 🤔. Although it also depends on whether a resizing is done or not, since otherwise the function returns `jpg` before `StringIO` is called. But I didn't check in which case a resizing is done yet.

We could simply run `str()` to convert it to a string, but I'm not sure whether this is actually a good idea if the sources was bytes. However, at least something you can test:
```sh
sed -i 's/StringIO(jpg)/StringIO(str(jpg))/' /usr/local/lib/python3.*/dist-packages/motioneye/mediafiles.py
systemctl restart motioneye
```
We could also use `StringIO` or `BytesIO` after checking which format it is, instead of converting it. I guess PIL is able to handle both.

jose1711, 2022-04-13:
After `sed` substitution:
```
apr 13 08:42:40 olivia meyectl[8750]:   File "/usr/lib/python3.10/site-packages/motioneye/handlers/picture.py", line 52, in get
apr 13 08:42:40 olivia meyectl[8750]:     await self.current(camera_id)
apr 13 08:42:40 olivia meyectl[8750]:   File "/usr/lib/python3.10/site-packages/motioneye/handlers/picture.py", line 110, in current
apr 13 08:42:40 olivia meyectl[8750]:     picture = mediafiles.get_current_picture(
apr 13 08:42:40 olivia meyectl[8750]:   File "/usr/lib/python3.10/site-packages/motioneye/mediafiles.py", line 1019, in get_current_picture
apr 13 08:42:40 olivia meyectl[8750]:     image = Image.open(sio)
apr 13 08:42:40 olivia meyectl[8750]:   File "/usr/lib/python3.10/site-packages/PIL/Image.py", line 3050, in open
apr 13 08:42:40 olivia meyectl[8750]:     raise ValueError(
apr 13 08:42:40 olivia meyectl[8750]: ValueError: StringIO cannot be used to open an image. Binary data must be used instead.
apr 13 08:42:40 olivia meyectl[8750]:    ERROR: 500 GET /picture/1/current/?_=1649832160372&width=0.6 (127.0.0.1) 0.57ms
```

MichaIng, 2022-04-13:
> ValueError: StringIO cannot be used to open an image. Binary data must be used instead.

Okay, to me it looks more and more that this resizing is not done often, so that this quite clear type error has not been detected yet. Probably with Python 2 (and related older module versions) PIL took string file objects, but doesn't do that anymore: https://pillow.readthedocs.io/en/stable/reference/Image.html#PIL.Image.open
Looks like er need to use `BytesIO(jpg.encode())` then, or probably even omit `encode()` it the JPEG stream is assured to be binary already:
```sh
sed -i 's/StringIO(str(jpg))/BytesIO(jpg)/' /usr/local/lib/python3.*/dist-packages/motioneye/mediafiles.py
systemctl restart motioneye
```

jose1711, 2022-04-19:
Substituted and restarted:
```
apr 19 20:33:44 myhost meyectl[16251]: Traceback (most recent call last):
apr 19 20:33:44 myhost meyectl[16251]:   File "/usr/lib/python3.10/site-packages/tornado/web.py", line 1704, in _execute
apr 19 20:33:44 myhost meyectl[16251]:     result = await result
apr 19 20:33:44 myhost meyectl[16251]:   File "/usr/lib/python3.10/site-packages/motioneye/handlers/picture.py", line 52, in get
apr 19 20:33:44 myhost meyectl[16251]:     await self.current(camera_id)
apr 19 20:33:44 myhost meyectl[16251]:   File "/usr/lib/python3.10/site-packages/motioneye/handlers/picture.py", line 110, in current
apr 19 20:33:44 myhost meyectl[16251]:     picture = mediafiles.get_current_picture(
apr 19 20:33:44 myhost meyectl[16251]:   File "/usr/lib/python3.10/site-packages/motioneye/mediafiles.py", line 1042, in get_current_picture
apr 19 20:33:44 myhost meyectl[16251]:     image.save(sio, format='JPEG')
apr 19 20:33:44 myhost meyectl[16251]:   File "/usr/lib/python3.10/site-packages/PIL/Image.py", line 2300, in save
apr 19 20:33:44 myhost meyectl[16251]:     save_handler(self, fp, filename)
apr 19 20:33:44 myhost meyectl[16251]:   File "/usr/lib/python3.10/site-packages/PIL/JpegImagePlugin.py", line 786, in _save
apr 19 20:33:44 myhost meyectl[16251]:     ImageFile._save(im, fp, [("jpeg", (0, 0) + im.size, 0, rawmode)], bufsize)
apr 19 20:33:44 myhost meyectl[16251]:   File "/usr/lib/python3.10/site-packages/PIL/ImageFile.py", line 519, in _save
apr 19 20:33:44 myhost meyectl[16251]:     fp.write(d)
apr 19 20:33:44 myhost meyectl[16251]: TypeError: string argument expected, got 'bytes'
apr 19 20:33:44 myhost meyectl[16251]:    ERROR: 500 GET /picture/1/current/?_=1650393224144&width=0.01&_username=admin&_signature=ab1421c38a87b0419f24194bf512aded4232e6fa (127.0.0.1) 1.08ms
apr 19 20:33:44 myhost meyectl[16251]:  WARNING: 403 GET /login/?_=1650393224669&_username=_&_signature=82c3108ef595378c6ca09d5c0100893865c39c63 (127.0.0.1) 0.27ms
apr 19 20:33:48 myhost meyectl[16251]:    ERROR: string argument expected, got 'bytes'
apr 19 20:33:48 myhost meyectl[16251]: Traceback (most recent call last):
apr 19 20:33:48 myhost meyectl[16251]:   File "/usr/lib/python3.10/site-packages/tornado/web.py", line 1704, in _execute
apr 19 20:33:48 myhost meyectl[16251]:     result = await result
apr 19 20:33:48 myhost meyectl[16251]:   File "/usr/lib/python3.10/site-packages/motioneye/handlers/picture.py", line 52, in get
apr 19 20:33:48 myhost meyectl[16251]:     await self.current(camera_id)
apr 19 20:33:48 myhost meyectl[16251]:   File "/usr/lib/python3.10/site-packages/motioneye/handlers/picture.py", line 110, in current
apr 19 20:33:48 myhost meyectl[16251]:     picture = mediafiles.get_current_picture(
apr 19 20:33:48 myhost meyectl[16251]:   File "/usr/lib/python3.10/site-packages/motioneye/mediafiles.py", line 1042, in get_current_picture
apr 19 20:33:48 myhost meyectl[16251]:     image.save(sio, format='JPEG')
apr 19 20:33:48 myhost meyectl[16251]:   File "/usr/lib/python3.10/site-packages/PIL/Image.py", line 2300, in save
apr 19 20:33:48 myhost meyectl[16251]:     save_handler(self, fp, filename)
apr 19 20:33:48 myhost meyectl[16251]:   File "/usr/lib/python3.10/site-packages/PIL/JpegImagePlugin.py", line 786, in _save
apr 19 20:33:48 myhost meyectl[16251]:     ImageFile._save(im, fp, [("jpeg", (0, 0) + im.size, 0, rawmode)], bufsize)
apr 19 20:33:48 myhost meyectl[16251]:   File "/usr/lib/python3.10/site-packages/PIL/ImageFile.py", line 519, in _save
apr 19 20:33:48 myhost meyectl[16251]:     fp.write(d)
apr 19 20:33:48 myhost meyectl[16251]: TypeError: string argument expected, got 'bytes'
apr 19 20:33:48 myhost meyectl[16251]:    ERROR: 500 GET /picture/1/current/?_=1650393228685&width=0.01&_username=admin&_signature=44250a98e13af489fff55ab2e6f4cdc8264ca64a (127.0.0.1) 1.29ms

```

MichaIng, 2022-04-19:
Let's try to create the target file object as BytesIO as well:
```sh
sed -i 's/sio = io.StringIO()/sio = io.BytesIO()/' /usr/local/lib/python3.*/dist-packages/motioneye/mediafiles.py
systemctl restart motioneye
```

This matches how it is done in `get_media_preview`, where it's about JPEGs as well, although not those from an MJPEG stream: https://github.com/motioneye-project/motioneye/blob/16fd474/motioneye/mediafiles.py#L887-L930

jose1711, 2022-04-19:
```
apr 19 21:35:26 myhost meyectl[37290]:    ERROR: image file is truncated (61 bytes not processed)
apr 19 21:35:26 myhost meyectl[37290]: Traceback (most recent call last):
apr 19 21:35:26 myhost meyectl[37290]:   File "/usr/lib/python3.10/site-packages/tornado/web.py", line 1704, in _execute
apr 19 21:35:26 myhost meyectl[37290]:     result = await result
apr 19 21:35:26 myhost meyectl[37290]:   File "/usr/lib/python3.10/site-packages/motioneye/handlers/picture.py", line 52, in get
apr 19 21:35:26 myhost meyectl[37290]:     await self.current(camera_id)
apr 19 21:35:26 myhost meyectl[37290]:   File "/usr/lib/python3.10/site-packages/motioneye/handlers/picture.py", line 110, in current
apr 19 21:35:26 myhost meyectl[37290]:     picture = mediafiles.get_current_picture(
apr 19 21:35:26 myhost meyectl[37290]:   File "/usr/lib/python3.10/site-packages/motioneye/mediafiles.py", line 1039, in get_current_picture
apr 19 21:35:26 myhost meyectl[37290]:     image.thumbnail((width, height), Image.CUBIC)
apr 19 21:35:26 myhost meyectl[37290]:   File "/usr/lib/python3.10/site-packages/PIL/Image.py", line 2475, in thumbnail
apr 19 21:35:26 myhost meyectl[37290]:     im = self.resize(size, resample, box=box, reducing_gap=reducing_gap)
apr 19 21:35:26 myhost meyectl[37290]:   File "/usr/lib/python3.10/site-packages/PIL/Image.py", line 2043, in resize
apr 19 21:35:26 myhost meyectl[37290]:     self.load()
apr 19 21:35:26 myhost meyectl[37290]:   File "/usr/lib/python3.10/site-packages/PIL/ImageFile.py", line 251, in load
apr 19 21:35:26 myhost meyectl[37290]:     raise OSError(
apr 19 21:35:26 myhost meyectl[37290]: OSError: image file is truncated (61 bytes not processed)
apr 19 21:35:26 myhost meyectl[37290]:    ERROR: 500 GET /picture/1/current/?_=1650396926639&width=0.01&_username=admin&_signature=42af74e8dd86bf8addd43bcfad6023b41f0051d5 (127.0.0.1) 2.03ms
```

jose1711, 2022-04-21:
With the latest replacements  I get a little more image data, this is me sitting in front of the camera with headphones on:
[image]

In the log:
```
apr 21 09:46:28 olivia meyectl[46965]:     future.result()
apr 21 09:46:28 olivia meyectl[46965]: tornado.iostream.StreamClosedError: Stream is closed
apr 21 09:46:58 olivia meyectl[46965]:    ERROR: mjpg client for camera 2 on port 9082 error: Stream is closed
apr 21 09:46:58 olivia meyectl[46965]: Traceback (most recent call last):
apr 21 09:46:58 olivia meyectl[46965]:   File "/usr/lib/python3.10/site-packages/motioneye/mjpgclient.py", line 282, in _on_before_content_length
apr 21 09:46:58 olivia meyectl[46965]:     future.result()
apr 21 09:46:58 olivia meyectl[46965]: tornado.iostream.StreamClosedError: Stream is closed
```
When I get a chance I'll test this on my laptop with integrated camera to see if I get similar behaviour. Thank you for your help and patience.

MichaIng, 2022-04-21:
Okay great. The remaining error btw is known and can be ignored. Of course we should try to solve it as well: #2382

I'll open a PR with these changes and add a bunch of reviewers, let's see whether someone has a cleaner solution in mind. But at least there is one 🙂.

jose1711, 2022-04-24:
FWIW I did a test on a laptop with the latest `master` branch and the symptoms are very similar to those described above. If there a testing branch I will supply results from both machines.

MichaIng, 2022-04-24:
Interesting that this was an issue before already. I though it was introduced with Python 2 => Python 3 conversion. However, no need to test with `master`, but only with `dev`. I didn't find time to open the PR yet. Will do next week.

MichaIng, 2022-04-28:
Probably #2432 solves the issue with the chunked JPEGs and hence the artefacts in your stream. The string/bytes format fixes however still need to be applied.

jose1711, 2022-04-28:
I tested with the latest `dev` branch. On laptop it's much better now - I can get a nice fluent video stream. However the resolution dimmer needs to be set to 100 %, any other value results in no video. On the desktop I was not that successful - here I could at least get a static grayscale image. Which changes should I apply? All `sed` commands listed above? Thanks.

MichaIng, 2022-04-28:
Based on the errors logs, if there are some, the following two should still be required if server side image resizing is done:
```sh
sed -i 's/sio = io.StringIO()/sio = io.BytesIO()/' /usr/local/lib/python3.*/dist-packages/motioneye/mediafiles.py
sed -i 's/StringIO(jpg)/BytesIO(jpg)/' /usr/local/lib/python3.*/dist-packages/motioneye/mediafiles.py
systemctl restart motioneye
```

jose1711, 2022-05-04:
I've got a nice black&white image from both cameras (on my desktop PC) and this message in journal log:
[image]

MichaIng, 2022-05-04:
motionEye does not support motion 4.4 fully yet, please use motion 4.3.2. We're working on it but I need some help to finish the netcam settings mapping: #2462

jose1711, 2022-05-04:
Ha, finally a colorful picture! :tada:  So first I downgraded motion to 4.3.2 and restarted. No luck, then I started digging into motion alone. Stopped motioneye and started:
```
/usr/bin/motion -n -c /etc/motioneye/camera-1.conf -d 5
```
This produced a color stream from the camera. Ok but this is not what motioneye is running so I stopped it and ran
```
/usr/bin/motion -n -c /etc/motioneye/motion.conf -d 5
```
This time it went into b&w mode again. I opened the config file and changed the first from `setup_mode on` to `setup_mode off`. And that actually did it, I restarted motioneye and life is great again. Now I only wonder how the "on" got there because I see
```
data.setdefault('setup_mode', False)
```
in the code :frowning_face:

MichaIng, 2022-05-05:
Hmm, not sure about that either. I didn't know the setup mode until now, to be honest 😅. But great that it works now.

MichaIng, 2022-05-22:
Marking this as closed as it has been solved in `dev` branch.

closed 2022-05-22 by MichaIng