motioneye-project/motioneye #2401: Arch Linux streaming Issue

kmsgli, 2022-04-04:
Hey thanks for the help in advanced.

I was under the assumption my issue was related to my POE switch going bad as all my cameras are static/cut in and out like an old TV with bad reception. After setting up motioneye for a friend with new hardware I realized he had the same issue. When using the RTSP of the cams in VLC they appear fine. Also when testing on a raspberry pi with motioneye on it the streams look fine as well.

The video records well and playback works but using the web interface the stream looks terrible.

This is the message I am getting from systemctl status, not sure if it is related but figured I would toss it in.

```
Apr 04 06:09:30 motion[119565]: [1:ml1:Porch] [CRT] [NET] motion_init: Substream not available.  Image sizes not modulo 16.
Apr 04 06:09:30 motion[119565]: [3:ml3:Driveway] [CRT] [NET] motion_init: Substream not available.  Image sizes not modulo 16.
Apr 04 06:09:31 motion[119565]: [4:ml4:Kitchen] [CRT] [NET] motion_init: Substream not available.  Image sizes not modulo 16.
```

Thanks.

zaphod24, 2022-04-04:
I updated my Odroid N2 from Ubuntu 20.04 to 22.04 and as part of that had to change to the new Python3 dev branch. Now when I bring up my camera feeds they look glitchy:

[image]

Everything detects motion, records and plays back correctly so it seems just the streaming is weird. I can also access the mjpeg stream of each camera on the central server and that also looks ok.

There also doesn't appear to be a view streaming/directly option in the recording list any more either.

kmsgli, 2022-04-05:
That screenshot is exactly the issue I am having. The difference is I used these instructions;
`https://github.com/motioneye-project/motioneye/wiki/Install-On-Arch`

That puts me using python 2 not python 3?

zaphod24, 2022-04-05:
Could just be something new with Ubuntu 22.04, a different jpeg library, etc. on this end

MichaIng, 2022-04-06:
Is `pillow` installed from pre-compiled wheels by PyPI or compiled in place? To rule out libjpeg differences, the pre-compiled wheels could be used, which is the default with `pip` on Debian and Ubuntu at least, not how whether Arch has other default configs. Ah, and which architecture is this? `x86_64` or some ARM?

zaphod24, 2022-04-06:
My arch is arm, specifically Ubuntu 22.04 on an Odroid N2.

For me, Pillow was installed as a part of following the python3 update instructions using python3 pip and appears to be installed in /usr/local/lib/python3.10/dist-packages/Pillow.libs

MichaIng, 2022-04-06:
Okay, then it was the pre-compiled aarch64 build from PyPI, which has libjpeg embedded. I.e. the libjpeg shipped by the distribution is irrelevant.

zaphod24, 2022-04-06:
Any ideas why updating from Ubuntu 20.04 to 22.04 (and being forced to do the new python3 dev branch because no more openssl 1.1) would result in the same hardware/config showing camera feeds like the one I attached above?

zaphod24, 2022-04-06:
I'm not sure about removing a camera, as I don't want to have to redo all of the settings but... If I go to the mjpeg URL for an individual camera, I don't have the issue... It's only on the overview with all 5 of my cameras.

MichaIng, 2022-04-06:
I see. However, since motionEye passes the MJPEG stream though the backend for mask and other overlays, it is not exactly the same. Does it work when you watch a single camera though the motionEye interface in full screen?

I'll do some testing tomorrow with multiple cameras and some public MJPEG/RTSP streams.

Btw @kmsgli it's great to have someone here who is btw using Arch 🙂. Would you mind to help adding Arch/pacman/pkg support to the new [motioneye_init > linux_init](https://github.com/motioneye-project/motioneye/blob/dev/motioneye/extra/linux_init) command for installing motion, v4l2-utils, FFmpeg and the motionEye service + config? Currently only DEB and RPM based distros are supported, but adding Arch (and probably Gentoo, although it uses a different package manager) would be great.

zaphod24, 2022-04-06:
Full screen of 1 camera is also glitchy... I do add a date/time overlay text to the cameras, but that's in the direct backend mjpeg link too which works fine... Very very odd

kmsgli, 2022-04-07:
@MichaIng I would love to help any way I can, I use arch for everything, I am on a `x86_64` cpu.

I have used arch for many years but I have never maintained any packages, so I am not sure how helpful I will be with that.

My current system Is running off this AUR package;
`https://aur.archlinux.org/packages/motioneye`

That no longer builds proper, I built my current system a few years ago so it has update a few times but has been stagnant for some time.

This most recent system I built was for a friend and the AUR package above builds but fails to start motioneye properly so I had to use the instruction in this github which builds correctly and runs but has the streaming issue pictured by @zaphod24. To my understanding this is purely python 2 no python 3 indicating the python 3 upgrade is not the issue here?

My system also has the same issue which I assume has occurred because one of the sub packages in motioneye upgrades to a newer version causing this issue.

I will check and see if I have any old motioneye packages in my pacman cache that i can downgrade to for testing if the issue goes away.

MichaIng, 2022-04-07:
I was just made aware on the other issue of the other AUR motionEye package, which is now based on the Python 3 `dev` branch: https://aur.archlinux.org/packages/motioneye-git

One issue that I now recognise is that both packages pull in motion, which is v4.4.0 on Arch, which is not yet full supported by motionEye. Some camera config keys have changed. Issue tracked here, I'll try to add v4.4.0 support the next days: #2333

Another issue with both AUR packages is that I do not see a `v4l2-utils` dependency, used to get camera information, like supported resolutions, brightness and others. On the other hand, I would have expected many more issues, if it is really missing. `v4l2-ctl` is the command, to check for.

But back to the actual topic:
```sh
Apr 06 19:34:30 archdesktop motion[119565]: [7:nc7:Driveway] [WRN] [NET] netcam_rtsp_connect: norm: capture_rate is less than camera FPS.
Apr 06 19:34:30 archdesktop motion[119565]: [7:nc7:Driveway] [WRN] [NET] netcam_rtsp_connect: norm: Decoding errors will occur.
```
This sounds suspicious in this context. The question is whether the `capture_rate` is lower because of insufficient bandwidth or for a different reason.

kmsgli, 2022-04-07:
I doubt its a bandwidth issue, these are all Ethernet cams and on using VLC to view the RTSP they appear fine.

this aur package has some mention of vl42;
https://aur.archlinux.org/packages/motioneye

That is the package I used to install all my systems.  The git package showed up when python3 got into the mix for motionEye, i tried it a while back but it did not even build so I left it be. In this circumstance the git AUR package is the motionEye using python3 and the non git package is the python 2 version if I am not mistaken?

I would love to upgrade to the motionEye python 3 setup so if you want I can test that and we can forget about trouble shooting the python 2 setup : )

MichaIng, 2022-04-08:
> I doubt its a bandwidth issue, these are all Ethernet cams and on using VLC to view the RTSP they appear fine.

Well, all external streams stack up two times, incoming from the origin and outgoing to the client, so the connection of the cams won't be the bottleneck but only the connection/bandwidth of the motionEye server, or the client, of course.

> this aur package has some mention of vl42;

But only the FFmpeg wrapper for the library, not the CLI utilities, which is what motionEye uses. As said, the command `v4l2-ctl` must be available, else motionEye cannot derive camera details.

> the non git package is the python 2 version if I am not mistaken?

Based of the Git source it uses, yes, but actually it looks like the dependencies like [python-jinja](https://archlinux.org/packages/?q=python-jinja) are the Python 3 modules, not the Python 2 ones, isn't it? This may be the reason why it fails.

zaphod24, 2022-04-08:
I don't think mine is a bandwidth issue as it worked fine before updating from Ubuntu 20.04 to 22.04 (and of course the dev branch of motioneye). Nothing on the hardware side has changed.

kmsgli, 2022-04-08:
> v4l2-ctl

> But only the FFmpeg wrapper for the library, not the CLI utilities, which is what motionEye uses. As said, the command `v4l2-ctl` must be available, else motionEye cannot derive camera details.

Running that command returns the help man for it so it is def installed on my system.

I did not use the AUR on the latest system I installed motionEye on and it has the same issue. I installed that system using the instructions for Arch linux from this github.

Can i test the python-jinja theory in some way?

zagrim, 2022-04-09:
I wonder if the issue with the artefacts in streaming is same as here, and if the same fix would also help: https://github.com/motioneye-project/motioneye/issues/2267#issuecomment-1011138091

zagrim, 2022-04-09:
Anyway, I have similar looking issue in my dev setup (Debian testing), with the same libmicrohttpd12 version as mentioned on that issue (0.9.75-3). Downgrading to the one that is available from current Debian stable (0.9.72-2) removes the colored bands. This was not an issue with the old dev branch (Python2) so I guess there's some incompability between that lib and some Python component that's using it. This comment https://github.com/motioneye-project/motioneye/issues/2267#issuecomment-1011456241 on the same issue mentions something that looks worth investigating, but OTOH there it's about Python2 so that's not a full match with this issue.

Do you guys @zaphod24 @kmsgli happen to have spurious errors in the log like me:
```
    INFO: hello! this is motionEye server 0.43.0
 WARNING: you have motion installed, but no v4l-utils
    INFO: cleanup started
    INFO: wsswitch started
    INFO: tasks started
    INFO: mjpg customer garbage collector has started
    INFO: server started
   ERROR: mjpg client for camera 2 on port 8082 error: Stream is closed
Traceback (most recent call last):
  File "/home/etikka/.local/lib/python3.9/site-packages/motioneye/mjpgclient.py", line 168, in _on_http
    data = future.result()
tornado.iostream.StreamClosedError: Stream is closed
   ERROR: mjpg client for camera 1 on port 8081 error: Stream is closed
Traceback (most recent call last):
  File "/home/etikka/.local/lib/python3.9/site-packages/motioneye/mjpgclient.py", line 168, in _on_http
    data = future.result()
tornado.iostream.StreamClosedError: Stream is closed
Traceback (most recent call last):
  File "/home/etikka/.local/lib/python3.9/site-packages/motioneye/mjpgclient.py", line 168, in _on_http
    data = future.result()
tornado.iostream.StreamClosedError: Stream is closed
```
The errors keep on coming rapidly until I shut ME down.

zaphod24, 2022-04-09:
I believe there is another issue logged about those spurious messages from tornado and they will be eliminated in the future.

About libmicrohttpd12, looks like my system is on 0.9.75-3ubuntu1

MichaIng, 2022-04-09:
Ah right, I totally forgot that it was narrowed down to `libmicrohttpd12` already, hence the library used by `motion` to provide the MPJEG streams. On Debian Bullseye it's v0.9.72-2, on Ubuntu Jammy and Arch Linux it's the latest v0.9.75-3. So some change between these versions seems to have caused the issue. Debian Bookworm provides v0.9.75-3 as well, i.e. I should be able to replicate the issue there.

It could be narrowed down to motion/libmicrohttpd12 taking motionEye out of the game by connecting to a single motion MJPEG stream, i.e. not the one of the remote/network camera directly (if it is not a local camera anyway) but the one provided by motion which is picked up by the motionEye backend. For the first camera this is usually on port 9081, for the second on port 9082 etc.

@zagrim 
The "Steam is closed" error is tracked here: #2382
Definitely not related, it is just unnecessarily logged or not well handled by motionEye when the client closes or halts the camera stream, e.g. because the tab is in background.

zaphod24, 2022-04-09:
Oddly I don't have an issue connecting to the cameras in 9081,2,3,etc

MichaIng, 2022-04-09:
Adding milestone, so I don't forget to try replicating this on Debian Bookworm.

kmsgli, 2022-04-10:
@zagrim I do not have those errors in my log.

@MichaIng  I don not have libmicrohttpd12 installed on my system i have libmicrohttpd and the version pacman indicates is 0.9.75-1.

Do you think downgrading will fix my issue and if so to what version?

MichaIng, 2022-04-10:
Ah yes sorry, `libmicrohttpd12` is the package name on Debian, but it's the same of course. I'd test to downgrade `libmicrohttpd` only at first, probably `motion` as a second step to v4.3.2 (if it is not that one yet), the latest one which is fully supported by motionEye.

kmsgli, 2022-04-11:
Downgrading libmicrohttpd to 9.72 fixed the issue. No need to downgrade motion. I will put it in my ignore package list for the time being. Any idea whats causing the conflict?

zaphod24, 2022-04-11:
That's great news! Any idea how to do that on Ubuntu 22.04?

zaphod24, 2022-04-11:
I managed to install libmicrohttpd12_0.9.72-2_arm64.deb and that did the trick!

I can open another issue, it seems that there is no longer an option to view recordings directly without downloading?

kmsgli, 2022-04-11:
Thats great you got it working, I had no idea how to do that in Ubuntu, Arch has a nice downgrade package that you can install for just this type of situation.

I noticed the view button left a while back, I don't think its an option any more.

zaphod24, 2022-04-12:
Yeah.. It was nice to view it in the browser without having to download. Would be nice for that to come back.

kmsgli, 2022-04-15:
I think I will close this as the problem is solved for now

MichaIng, 2022-04-15:
Let us keep that open. We should further investigate/find a reliable trigger for this issue and then report it upstream, in case. Probably we also need to adjust the way this lib is used in our code due to recent changes, or motion might need to. I hope I find time to debug it after Easter holidays.

kmsgli, 2022-04-15:
No worries, sounds good to me. I you need any testing reach out, I am happy to test.

MichaIng, 2022-04-28:
We worked around the issue with #2432.

Background is that newer libmicrohttp versions use chunked HTTP streams even for non-keep-alive requests, which is indeed the correct/RFC compliant behaviour. The `tornado.IOStream` server we use to pass the MJPEG stream to the motionEye frontend however does not support chunks well, hence the incomplete images and artefacts.

As a workaround we use HTTP/1.0 connections, which do not support chunks, instead of HTTP/1.1. Not the best to use old protocols to disable features, but waiting for chunk support in `tornado.IOStream` or replacing it is more work, now tracked here: #2441

Marking this hence as closed/solved in `dev`.

closed 2022-04-28 by MichaIng

BenjaminRehn, 2022-09-21:
If you using motioneye in Ubuntu 22... and found yourself here while your stream looks like, it's having a great lightshow party ...
try installing libmicrohttpd12_0.9.72-2 like said in the comments...
`wget http://ftp.de.debian.org/debian/pool/main/libm/libmicrohttpd/libmicrohttpd12_0.9.72-2_amd64.deb
sudo dpkg -i libmicrohttpd12_0.9.72-2_amd64.deb`