motioneye-project/motioneye #2406: Motion Notificacion Web Hook URL is reset to default after restarting Home Assistant

sesardelaisla, 2022-04-06:
I have been using MotionEye a few months ago. I have realized that the Web Hook URL field below Motion Notification section is reset to default after restarting Home Assistant. Is it possible to fix this behaviour? Thanks!

MichaIng, 2022-04-06:
With plain motionEye this is not the case, hence I suggest you ask this on the HA addon's repository: https://github.com/hassio-addons/addon-motioneye/issues

sesardelaisla, 2022-04-06:
Thank you for such a quick response! Actually, I just realized that field is not reset if I restart the component only, so it makes sense that the problem is in the HA integration part. I will open an issue in HA addon's repository. Cheers!

closed 2022-04-06 by sesardelaisla