motioneye-project/motioneye #2408: File Upload Error

Ryarg, 2022-04-08:
Error message - Accessing the upload service failed:POST data should be bytes, an iterable of bytes, or a file object. It cannot be of type str.!

I followed the install instructions on this page - https://github.com/motioneye-project/motioneye/tree/dev

I'm using Raspberry Pi OS 64bit Lite on a Pi 4 4GB, the error message appears after click Test Service under File Storage, I've selected Google Drive and added the authorisation key.

Is there anything I can do to fix this?

Thanks.

MichaIng, 2022-04-08:
Many thanks for your report. Can you please post the full error message, which can be found e.g. via:
```sh
journalctl -u motioneye
```

Ryarg, 2022-04-08:
Hi, journalctl -u motioneye produce this:

-- Journal begins at Mon 2022-04-04 15:41:41 BST, ends at Fri 2022-04-08 21:02:09 BST. --
Apr 08 11:23:26 motioneye systemd[1]: Started motionEye Server.
Apr 08 11:23:27 motioneye meyectl[4130]: configure_logging cmd motioneye: False
Apr 08 11:23:27 motioneye meyectl[4130]: configure logging to file: None
Apr 08 11:23:27 motioneye meyectl[4130]:     INFO: hello! this is motionEye server 0.43.0
Apr 08 11:23:28 motioneye meyectl[4130]:     INFO: main config file /etc/motioneye/motion.conf does not exist, using default values
Apr 08 11:23:28 motioneye meyectl[4130]:     INFO: cleanup started
Apr 08 11:23:28 motioneye meyectl[4130]:     INFO: wsswitch started
Apr 08 11:23:28 motioneye meyectl[4130]:     INFO: tasks started
Apr 08 11:23:28 motioneye meyectl[4130]:     INFO: mjpg customer garbage collector has started
Apr 08 11:23:28 motioneye meyectl[4130]:     INFO: server started
Apr 08 11:23:51 motioneye meyectl[4130]:     INFO: interrupt signal received, closing ...
Apr 08 11:23:51 motioneye systemd[1]: Stopping motionEye Server...
Apr 08 11:23:52 motioneye meyectl[4130]:     INFO: server stopped
Apr 08 11:23:52 motioneye meyectl[4130]:     INFO: tasks stopped
Apr 08 11:23:52 motioneye meyectl[4130]:     INFO: goodbye!
Apr 08 11:23:52 motioneye systemd[1]: motioneye.service: Succeeded.
Apr 08 11:23:52 motioneye systemd[1]: Stopped motionEye Server.
Apr 08 11:23:52 motioneye systemd[1]: motioneye.service: Consumed 1.990s CPU time.
Apr 08 11:24:53 motioneye systemd[1]: Started motionEye Server.
Apr 08 11:24:54 motioneye meyectl[4175]: configure_logging cmd motioneye: False
Apr 08 11:24:54 motioneye meyectl[4175]: configure logging to file: None
Apr 08 11:24:54 motioneye meyectl[4175]:     INFO: hello! this is motionEye server 0.43.0
Apr 08 11:24:55 motioneye meyectl[4175]:     INFO: main config file /etc/motioneye/motion.conf does not exist, using default values
Apr 08 11:24:55 motioneye meyectl[4175]:     INFO: cleanup started
Apr 08 11:24:55 motioneye meyectl[4175]:     INFO: wsswitch started
Apr 08 11:24:55 motioneye meyectl[4175]:     INFO: tasks started
Apr 08 11:24:55 motioneye meyectl[4175]:     INFO: mjpg customer garbage collector has started
Apr 08 11:24:55 motioneye meyectl[4175]:     INFO: server started
Apr 08 11:26:10 motioneye meyectl[4175]:     INFO: interrupt signal received, closing ...
Apr 08 11:26:10 motioneye systemd[1]: Stopping motionEye Server...
Apr 08 11:26:11 motioneye meyectl[4175]:     INFO: server stopped
Apr 08 11:26:11 motioneye meyectl[4175]:     INFO: tasks stopped
Apr 08 11:26:11 motioneye meyectl[4175]:     INFO: goodbye!
Apr 08 11:26:11 motioneye systemd[1]: motioneye.service: Succeeded.
Apr 08 11:26:11 motioneye systemd[1]: Stopped motionEye Server.
Apr 08 11:26:11 motioneye systemd[1]: motioneye.service: Consumed 1.964s CPU time.
-- Boot c295a9387d70401fa0d90474ede42cd1 --
Apr 08 11:26:15 motioneye systemd[1]: Started motionEye Server.
Apr 08 11:26:20 motioneye meyectl[507]: configure_logging cmd motioneye: False
Apr 08 11:26:20 motioneye meyectl[507]: configure logging to file: None
Apr 08 11:26:20 motioneye meyectl[507]:     INFO: hello! this is motionEye server 0.43.0
Apr 08 11:26:21 motioneye meyectl[507]:     INFO: main config file /etc/motioneye/motion.conf does not exist, using default values
Apr 08 11:26:21 motioneye meyectl[507]:     INFO: cleanup started
Apr 08 11:26:21 motioneye meyectl[507]:     INFO: wsswitch started
Apr 08 11:26:21 motioneye meyectl[507]:     INFO: tasks started
Apr 08 11:26:21 motioneye meyectl[507]:     INFO: mjpg customer garbage collector has started
Apr 08 11:26:21 motioneye meyectl[507]:     INFO: server started
Apr 08 11:27:20 motioneye meyectl[507]:  WARNING: 403 GET /login/?_=1649413640090&_username=_&_signature=40412cf2edcfa4f6f0f47ed9505c2>
Apr 08 11:27:54 motioneye meyectl[507]:     INFO: adding new v4l2 camera with id 1...
Apr 08 11:27:54 motioneye meyectl[507]:     INFO: main config file /etc/motioneye/motion.conf does not exist, using default values
Apr 08 11:27:55 motioneye motion[790]: [0:motion] [NTC] [ALL] conf_load: Processing thread 0 - config file /etc/motioneye/motion.conf
Apr 08 11:27:55 motioneye motion[790]: [0:motion] [NTC] [ALL] config_camera: Processing camera config file camera-1.conf
Apr 08 11:27:55 motioneye motion[790]: [0:motion] [NTC] [ALL] motion_startup: Logging to syslog

MichaIng, 2022-04-09:
Hmm, I do not see the error there. There do you see it, within the web UI somehow, or in a log file (while it is "Logging to syslog")?

Ryarg, 2022-04-09:
The only error message I see is the that appears after clicking Test service in the web UI "Accessing the upload service failed:POST data should be bytes, an iterable of bytes, or a file object. It cannot be of type str.!"

MichaIng, 2022-04-13:
Here the related code: https://github.com/motioneye-project/motioneye/blob/6a5953b/motioneye/uploadservices.py#L212-L218
Credentials are requested here: https://github.com/motioneye-project/motioneye/blob/dev/motioneye/uploadservices.py#L275-L304

`urllib.request.Request` indeed requires the `data` argument to be bytes. `urlencode` is used above, which returns a string. It's docs even state that this should be encoded to bytes when passed to POST requests: https://docs.python.org/3/library/urllib.parse.html#urllib.parse.urlencode

Easy to fix. I'll do that tomorrow, need to sleep now.

Ryarg, 2022-04-13:
Thanks a lot for your help.

MichaIng, 2022-04-13:
This should fix it: #2416
I'm wondering why we never saw this error with other upload services, since those request data key-value pairs were never bytes-encoded. Probably, urllib is strict only with POST requests and otherwise allows strings, although the `Request` docs state clearly that "The supported object types include bytes, file-like objects, and iterables of bytes-like objects.": https://docs.python.org/3/library/urllib.request.html#urllib.request.Request

MichaIng, 2022-04-20:
@Ryarg 
The fix has been merged. Can you test it if you find time by updating motionEye to latest `dev`: https://github.com/motioneye-project/motioneye/tree/dev#upgrade

Ryarg, 2022-04-23:
Hi, the upgrade hasn't fixed the problem, I still get the same web ui error message.

Accessing the upload service failed:POST data should be bytes, an iterable of bytes, or a file object. It cannot be of type str.!

MichaIng, 2022-04-23:
Did you really upgrade with the given command? Especially the `--force-reinstall` is mandatory here, since the version string did not change. And assure that the service got restarted.

Ryarg, 2022-04-23:
I actually started again from scratch, and got the error message after clicking test service. Then I tried the upgrade, I copied and pasted the commands in to terminal, rebooted, tried test service again and got the same error message.

MichaIng, 2022-04-24:
Ah I see we missed one upload: https://github.com/motioneye-project/motioneye/blob/b9f446d/motioneye/uploadservices.py#L371-L392

The body is somehow manually crafted and not `urlencode`d. Not sure whether this would still make sense. Let's try to only byte-encode it when passing to `Request`:
```sh
sed -i 's/self._request(self.UPLOAD_URL, body, headers)/self._request(self.UPLOAD_URL, body.encode(), headers)/' /usr/local/lib/python3.*/dist-packages/motioneye/uploadservices.py
systemctl restart motioneye
```

There are some more cases where JSON strings are generated via `json.dump()` which are then passed to the request without byte-encoding. AFAIK all of these need to be byte-encoded, since `Request` always requires bytes, at least when using a POST request (what we do here with a JSON data object).

Ryarg, 2022-04-24:
This didn't work, I still get the same error message.

MichaIng, 2022-04-24:
Could you paste the exact log again, to see in which part of the code this happens? I guess it's during actual video upload, not when testing the upload server, is it? And it's about video upload to Google Drive and not image upload, or Google Photos upload?

As said, there are more cases where I do not see proper bytes encoding. I'd find time during next week to do some reason about used methods and go through all cases for consequent encoding.

Ryarg, 2022-04-26:
From Syslog

Apr 26 16:39:26 motion meyectl[606]:   File "/usr/lib/python3.9/urllib/request.py", line 1277, in do_request_
Apr 26 16:39:26 motion meyectl[606]:     raise TypeError(msg)
Apr 26 16:39:26 motion meyectl[606]: TypeError: POST data should be bytes, an iterable of bytes, or a file object. It cannot be of type str.
Apr 26 16:39:26 motion meyectl[606]:    ERROR: gdrive: request failed: POST data should be bytes, an iterable of bytes, or a file object. It cannot be of type str.
Apr 26 16:39:26 motion meyectl[606]:    ERROR: failed to upload file "/var/lib/motioneye/Camera1/2022-04-26/16-38-38.jpg" with service gdrive: POST data should be bytes, an iterable of bytes, or a file object. It cannot be of type str.
Apr 26 16:39:26 motion meyectl[606]: Traceback (most recent call last):
Apr 26 16:39:26 motion meyectl[606]:   File "/usr/local/lib/python3.9/dist-packages/motioneye/uploadservices.py", line 1230, in upload_media_file
Apr 26 16:39:26 motion meyectl[606]:     service.upload_file(target_dir, filename, camera_name)
Apr 26 16:39:26 motion meyectl[606]:   File "/usr/local/lib/python3.9/dist-packages/motioneye/uploadservices.py", line 107, in upload_file
Apr 26 16:39:26 motion meyectl[606]:     self.upload_data(rel_filename, mime_type, data, ctime, camera_name)
Apr 26 16:39:26 motion meyectl[606]:   File "/usr/local/lib/python3.9/dist-packages/motioneye/uploadservices.py", line 369, in upload_data
Apr 26 16:39:26 motion meyectl[606]:     metadata = {'title': filename, 'parents': [{'id': self._get_folder_id(path)}]}
Apr 26 16:39:26 motion meyectl[606]:   File "/usr/local/lib/python3.9/dist-packages/motioneye/uploadservices.py", line 414, in _get_folder_id
Apr 26 16:39:26 motion meyectl[606]:     folder_id = self._get_folder_id_by_path(location)
Apr 26 16:39:26 motion meyectl[606]:   File "/usr/local/lib/python3.9/dist-packages/motioneye/uploadservices.py", line 426, in _get_folder_id_by_path
Apr 26 16:39:26 motion meyectl[606]:     parent_id = self._get_folder_id_by_name(parent_id, name)
Apr 26 16:39:26 motion meyectl[606]:   File "/usr/local/lib/python3.9/dist-packages/motioneye/uploadservices.py", line 466, in _get_folder_id_by_name
Apr 26 16:39:26 motion meyectl[606]:     self._create_folder(parent_id, child_name)
Apr 26 16:39:26 motion meyectl[606]:   File "/usr/local/lib/python3.9/dist-packages/motioneye/uploadservices.py", line 487, in _create_folder
Apr 26 16:39:26 motion meyectl[606]:     self._request(self.CREATE_FOLDER_URL, body, headers)
Apr 26 16:39:26 motion meyectl[606]:   File "/usr/local/lib/python3.9/dist-packages/motioneye/uploadservices.py", line 228, in _request
Apr 26 16:39:26 motion meyectl[606]:     response = utils.urlopen(request)
Apr 26 16:39:26 motion meyectl[606]:   File "/usr/local/lib/python3.9/dist-packages/motioneye/utils/__init__.py", line 421, in urlopen
Apr 26 16:39:26 motion meyectl[606]:     return urllib.request.urlopen(*args, **kwargs)
Apr 26 16:39:26 motion meyectl[606]:   File "/usr/lib/python3.9/urllib/request.py", line 214, in urlopen
Apr 26 16:39:26 motion meyectl[606]:     return opener.open(url, data, timeout)
Apr 26 16:39:26 motion meyectl[606]:   File "/usr/lib/python3.9/urllib/request.py", line 514, in open
Apr 26 16:39:26 motion meyectl[606]:     req = meth(req)
Apr 26 16:39:26 motion meyectl[606]:   File "/usr/lib/python3.9/urllib/request.py", line 1277, in do_request_
Apr 26 16:39:26 motion meyectl[606]:     raise TypeError(msg)
Apr 26 16:39:26 motion meyectl[606]: TypeError: POST data should be bytes, an iterable of bytes, or a file object. It cannot be of type str.
Apr 26 16:39:57 motion dhcpcd[776]: wlan0: hardware address 00:00:00:00:00:00 claims 192.168.1.20
Apr 26 16:39:57 motion systemd[1]: Starting Cleanup of Temporary Directories...

MichaIng, 2022-04-28:
Okay, it's in the upload service internal `_request()` methods where the encoding needs to be additionally done. Please try:
```sh
sed -i '/request = Request(url, data=body, headers=headers)/i\        body = body.encode()' /usr/local/lib/python3.*/dist-packages/motioneye/uploadservices.py
```
But then it must not be done before passing the body to these internal methods, hence reverting the change you did before:
```sh
sed -i 's/self._request(self.UPLOAD_URL, body.encode(), headers)/self._request(self.UPLOAD_URL, body, headers)/' /usr/local/lib/python3.*/dist-packages/motioneye/uploadservices.py
systemctl restart motioneye
```
As far as I checked, the encoding isn't done on any other `_request()` call, so should be safe to do it within this method (two places in the code only) instead of in every parent function before the `_request()` call.

Ryarg, 2022-04-28:
After following the instructions, I now get this error message - 

Accessing the upload service failed:'NoneType' object has no attribute 'encode'!

Syslog
Apr 28 18:01:47 motion meyectl[611]:     metadata = {'title': filename, 'parents': [{'id': self._get_folder_id(path)}]}
Apr 28 18:01:47 motion meyectl[611]:   File "/usr/local/lib/python3.9/dist-packages/motioneye/uploadservices.py", line 415, in _get_folder_id
Apr 28 18:01:47 motion meyectl[611]:     folder_id = self._get_folder_id_by_path(location)
Apr 28 18:01:47 motion meyectl[611]:   File "/usr/local/lib/python3.9/dist-packages/motioneye/uploadservices.py", line 427, in _get_folder_id_by_path
Apr 28 18:01:47 motion meyectl[611]:     parent_id = self._get_folder_id_by_name(parent_id, name)
Apr 28 18:01:47 motion meyectl[611]:   File "/usr/local/lib/python3.9/dist-packages/motioneye/uploadservices.py", line 450, in _get_folder_id_by_name
Apr 28 18:01:47 motion meyectl[611]:     response = self._request(url)
Apr 28 18:01:47 motion meyectl[611]:   File "/usr/local/lib/python3.9/dist-packages/motioneye/uploadservices.py", line 224, in _request
Apr 28 18:01:47 motion meyectl[611]:     body = body.encode()
Apr 28 18:01:47 motion meyectl[611]: AttributeError: 'NoneType' object has no attribute 'encode'
Apr 28 18:02:44 motion meyectl[513]:  WARNING: accessing gdrive failed: 'NoneType' object has no attribute 'encode'
Apr 28 18:02:53 motion meyectl[513]:  WARNING: accessing gdrive failed: 'NoneType' object has no attribute 'encode'
Apr 28 18:03:01 motion meyectl[513]:  WARNING: accessing gdrive failed: 'NoneType' object has no attribute 'encode'
Apr 28 18:03:16 motion meyectl[513]:  WARNING: accessing gdrive failed: 'NoneType' object has no attribute 'encode'
Apr 28 18:04:09 motion meyectl[513]:    ERROR: mjpg client for camera 1 on port 9081 error: Stream is closed
Apr 28 18:04:09 motion meyectl[513]: Traceback (most recent call last):
Apr 28 18:04:09 motion meyectl[513]:   File "/usr/local/lib/python3.9/dist-packages/motioneye/mjpgclient.py", line 282, in _on_before_content_length
Apr 28 18:04:09 motion meyectl[513]:     future.result()
Apr 28 18:04:09 motion meyectl[513]: tornado.iostream.StreamClosedError: Stream is closed
Apr 28 18:04:45 motion meyectl[513]:  WARNING: 403 GET /login/?_=1651165485763 (192.168.1.10) 1.14ms
Apr 28 18:06:56 motion motion: [0:motion] [NTC] [ALL] conf_load: Processing thread 0 - config file /etc/motioneye/motion.conf
Apr 28 18:06:56 motion motion: [0:motion] [NTC] [ALL] config_camera: Processing camera config file camera-1.conf
Apr 28 18:06:56 motion motion: [0:motion] [NTC] [ALL] motion_startup: Logging to syslog
Apr 28 18:06:56 motion motion: [0:motion] [NTC] [ALL] motion_startup: Motion 4.3.2 Started
Apr 28 18:06:56 motion motion: [0:motion] [NTC] [ALL] motion_startup: Using default log type (ALL)
Apr 28 18:06:56 motion motion: [0:motion] [NTC] [ALL] motion_startup: Using log type (ALL) log level (WRN)
Apr 28 18:06:58 motion meyectl[513]:    ERROR: mjpg client for camera 1 on port 9081 error: Stream is closed
Apr 28 18:06:58 motion meyectl[513]: Traceback (most recent call last):
Apr 28 18:06:58 motion meyectl[513]:   File "/usr/local/lib/python3.9/dist-packages/motioneye/mjpgclient.py", line 282, in _on_before_content_length
Apr 28 18:06:58 motion meyectl[513]:     future.result()
Apr 28 18:06:58 motion meyectl[513]: tornado.iostream.StreamClosedError: Stream is closed
Apr 28 18:07:03 motion meyectl[513]:  WARNING: accessing gdrive failed: 'NoneType' object has no attribute 'encode'
Apr 28 18:07:47 motion systemd[1]: Started Session 3 of user garry.
Apr 28 18:09:35 motion systemd[1]: Starting Cleanup of Temporary Directories...
Apr 28 18:09:35 motion systemd[1]: systemd-tmpfiles-clean.service: Succeeded.
Apr 28 18:09:35 motion systemd[1]: Finished Cleanup of Temporary Directories.
Apr 28 18:09:57 motion dhcpcd[2040]: wlan0: hardware address 00:00:00:00:00:00 claims 192.168.1.26
Apr 28 18:09:58 motion dhcpcd[2040]: wlan0: hardware address 00:00:00:00:00:00 claims 192.168.1.26
Apr 28 18:09:58 motion dhcpcd[2040]: wlan0: 10 second defence failed for 192.168.1.26
Apr 28 18:09:58 motion avahi-daemon[408]: Withdrawing address record for 192.168.1.26 on wlan0.
Apr 28 18:09:58 motion avahi-daemon[408]: Leaving mDNS multicast group on interface wlan0.IPv4 with address 192.168.1.26.
Apr 28 18:09:58 motion avahi-daemon[408]: Interface wlan0.IPv4 no longer relevant for mDNS.
Apr 28 18:09:58 motion dhcpcd[2040]: wlan0: deleting route to 192.168.1.0/24
Apr 28 18:09:58 motion dhcpcd[2040]: wlan0: deleting default route via 192.168.1.1
Apr 28 18:09:58 motion dhcpcd[2040]: wlan0: rebinding lease of 192.168.1.26
Apr 28 18:09:59 motion dhcpcd[2040]: wlan0: probing address 192.168.1.26/24
Apr 28 18:10:04 motion dhcpcd[2040]: wlan0: leased 192.168.1.26 for 86400 seconds
Apr 28 18:10:04 motion avahi-daemon[408]: Joining mDNS multicast group on interface wlan0.IPv4 with address 192.168.1.26.
Apr 28 18:10:04 motion avahi-daemon[408]: New relevant interface wlan0.IPv4 for mDNS.
Apr 28 18:10:04 motion avahi-daemon[408]: Registering new address record for 192.168.1.26 on wlan0.IPv4.
Apr 28 18:10:04 motion dhcpcd[2040]: wlan0: adding route to 192.168.1.0/24
Apr 28 18:10:04 motion dhcpcd[2040]: wlan0: adding default route via 192.168.1.1
Apr 28 18:10:04 motion systemd[1]: Stopping Network Time Synchronization...
Apr 28 18:10:04 motion systemd[1]: systemd-timesyncd.service: Succeeded.
[... cut ...]
Apr 28 18:10:04 motion systemd-timesyncd[4576]: Initial synchronization to time server 80.82.244.120:123 (0.debian.pool.ntp.org).
Apr 28 18:14:19 motion meyectl[513]:    ERROR: mjpg client for camera 1 on port 9081 error: Stream is closed
Apr 28 18:14:19 motion meyectl[513]: Traceback (most recent call last):
Apr 28 18:14:19 motion meyectl[513]:   File "/usr/local/lib/python3.9/dist-packages/motioneye/mjpgclient.py", line 282, in _on_before_content_length
Apr 28 18:14:19 motion meyectl[513]:     future.result()
Apr 28 18:14:19 motion meyectl[513]: tornado.iostream.StreamClosedError: Stream is closed

MichaIng, 2022-04-28:
Okay, looks like sometimes there is no `body` passed. So we need to check whether it's a string (or not `null`) before trying the encoding. Not so easy to implement with a `sed`, so I'll open a PR instead.

digi-care, 2022-05-03:
Encode value of body.

- GoogleDrive::upload_data
- GoogleDrive::_create_folder

```
 motioneye/uploadservices.py | 6 +++---
 1 file changed, 3 insertions(+), 3 deletions(-)

diff --git a/motioneye/uploadservices.py b/motioneye/uploadservices.py
index 4998ac2..f45074c 100644
--- a/motioneye/uploadservices.py
+++ b/motioneye/uploadservices.py
@@ -381,9 +381,9 @@ class GoogleDrive(UploadService, GoogleBase):
             '',
         ]
 
-        body = '\r\n'.join(body)
+        body = '\r\n'.join(body).encode()
         body += data
-        body += f'\r\n--{self.BOUNDARY}--'
+        body += f'\r\n--{self.BOUNDARY}--'.encode()
 
         headers = {
             'Content-Type': f'multipart/related; boundary="{self.BOUNDARY}"',
@@ -481,7 +481,7 @@ class GoogleDrive(UploadService, GoogleBase):
             'mimeType': 'application/vnd.google-apps.folder',
         }
 
-        body = json.dumps(metadata)
+        body = json.dumps(metadata).encode()
 
         headers = {'Content-Type': 'application/json; charset=UTF-8'}
 
```

MichaIng, 2022-05-03:
What do your think about doing the encoding consistently once when passing it to the request? So once doesn't need to take care of this when changing the code which crafts the body. In your case, is `data` bytes already? I wonder how it was even possible before to concatenate it to a string and another string afterwards. However, if possible, better to encode the whole body in one turn instead doing this for every part separately.

Not sure what to do with None bodies. Those cannot return any reasonable response anyway (?), so returning an empty "response" in the first place, skipping the request? Otherwise:
```py
if body is None:
  body = ''
```
consistently before trying to encode it.

jmichault, 2022-05-13:
Tested with PR #2456 : 
* "test service" returns OK
* upload files with subdirectories works OK

closed 2022-05-13 by jmichault