motioneye-project/motioneye #2409: Docker Image - fdisk package is missing

Thomaskru, 2022-04-08:
Hi!

Building Docker Image from dev branch. It seems that fdisk package is missing in Dockerfile. Running the Image shows following error:

`   DEBUG: listing cameras
   ERROR: failed to list disks using "fdisk -l": [Errno 2] No such file or directory: 'fdisk'
Traceback (most recent call last):
  File "/usr/local/lib/python3.9/dist-packages/motioneye/controls/diskctl.py", line 159, in _list_disks_fdisk
    output = utils.call_subprocess(['fdisk', '-l'], stderr=utils.DEV_NULL)
  File "/usr/local/lib/python3.9/dist-packages/motioneye/utils/__init__.py", line 657, in call_subprocess
    return subprocess.run(
  File "/usr/lib/python3.9/subprocess.py", line 505, in run
    with Popen(*popenargs, **kwargs) as process:
  File "/usr/lib/python3.9/subprocess.py", line 951, in __init__
    self._execute_child(args, executable, preexec_fn, close_fds,
  File "/usr/lib/python3.9/subprocess.py", line 1823, in _execute_child
    raise child_exception_type(errno_num, err_msg, err_filename)
FileNotFoundError: [Errno 2] No such file or directory: 'fdisk'
   DEBUG: listing mounts...
   DEBUG: found mount "overlay" at "/"
`

Please add fdisk Package in "motioneye/docker/Dockerfile":

`RUN case "$(dpkg --print-architecture)" in \
      'armhf') PACKAGES='gcc libjpeg-dev python3-distutils python3-dev'; printf '%b' '[global]\nextra-index-url=https://www.piwheels.org/simple/\n' > /etc/pip.conf;; \
      *) PACKAGES='gcc libcurl4-openssl-dev libssl-dev python3-dev';; \
    esac && \
    apt-get -q update && \
    DEBIAN_FRONTEND="noninteractive" apt-get -qq --option Dpkg::Options::="--force-confnew" --no-install-recommends install \
      ca-certificates curl python3 `**fdisk**` $PACKAGES && \`

Errors are gone. 

P.S.: Sorry for the formatting.

Thx.

Thomas

MichaIng, 2022-04-09:
Many thanks, makes sense. For completeness, `fdisk -l` is used here in the code: https://github.com/motioneye-project/motioneye/blob/16fd474/motioneye/controls/diskctl.py#L157-L164

It could be discussed whether it makes sense at all to list disks from within a container, which has no direct disk access anyway. But until we have a proper solution for this, we should mute the error by installing `fdisk`.

Check out the edit I done to your post to enhance multi line code/log format 🙂. Code fences:
````
```
line 1
line 2
```
````

MichaIng, 2022-04-12:
Solved in `dev`.