motioneye-project/motioneye #2414: Feature request

giff1000, 2022-04-11:
Hi there, sorry I'm not sure how to label this properly , it should be a feature request.

Would it be at all possible to add the ability to save video files as webm files?

Thank you very much

Kind regards 

Paul

MichaIng, 2022-04-12:
Many thanks for your request.

Generally it is possible to convert to webm via FFmpeg, but it requires additional libraries, which AFAIK are not compile into most distribution's FFmpeg packages: https://stackoverflow.com/questions/6954845/how-to-create-a-webm-video-file
... although, on Debian: https://packages.debian.org/bullseye/libavcodec58
`libvorbis0a` and `libvpx6` are dependencies of the FFmpeg codec library.

Probably we can ask FFmpeg about supported video codecs and offer those. I've actually not have a look at if/how video formats can be selected right now. There is a setting in motionEye camera configs with a file extension > format mapping, in current dev code since #2110.

JamBalaya56562, 2026-11-13:
Closing like #2017: the movies are encoded by Motion, and Motion 4.x has no WebM container (it is in Motion's development branch for version 5). Once a Motion release supports it, motionEye can offer it as a movie format.

closed 2026-11-13 by JamBalaya56562