motioneye-project/motioneye #2417: Too many false positives for motion detection after setting editable mask

fun4stuff, 2022-04-14:
motionEye Version	0.42
Motion Version	4.2.2
OS Version	Ubuntu 18.10

Motion detection works great before enabling the mask. The problem is I want mask out the road and sidewalk so passing cars and people don't set off motion detection. 

After I mask that section off, i get many more motion alerts from seemingly nothing. 

Is there any documentation for this feature?

Thanks!

MichaIng, 2022-04-14:
Can you please try it with the new Python 3 version? Many changes have been applied and I want to avoid spending too much time on v0.42.0 issues which may be not present on v0.43.0 anymore: https://github.com/motioneye-project/motioneye/tree/dev#installation

zagrim, 2022-04-15:
Just adding a mask should never be able to increase detected motion (since it just effectively masks off areas in the camera frame from the detection algorithm in Motion). So, did you change any other motion detection settings? If you just disable the mask, do you still get high amount of motion events, or do those go down to the same level that you had prior to adding the mask?

You might also want to turn on "Show Frame Changes" (unless you already have), and maybe also enable "Create Debug Media Files" to be able to see where in the captured frames motion was detected (note: this will create large number of pictures with "m" at the end of the file name, so don't forget to turn that off to avoid filling up your media storage!). There's a example pair of images in Motion docs which I think show pretty well how the debug image compares to the actual one showing the parts that have movement (with the blue one being the highest amount of movement).

Anyway, I think you just need to adjust the detection settings to accommodate the effect of masking, and depending on the setting you have there might be some other parts in the frame that might have non-interesting movement, too (like tree/bush branches or their shadows moving with the wind etc).

MichaIng, 2022-04-15:
Yeah, a mask of course should never increase but decrease motion. Although I never had a look into to code part yet. Since it is a privacy mask in the first place, it may be expected that it does not prevent motion detection below the mask but only mask those from the final camera stream to hide private parts/information from that stream.

Nice to know about the debug mode, which should help troubleshooting the case.

zagrim, 2022-04-16:
@MichaIng to avoid further confusion: There are two masks: privacy mask (set under "Video Device", corresponds to `motion_file` in camera config) and the one affecting motion detection (set under "Motion Detection"). The latter will exclude the masked area from motion detection, while the former will (according to [Motion docs](https://motion-project.github.io/motion_config.html#mask_privacy), I actually thought it would just mask the output) mask the area completely (so no motion is detected there, and no pixels from that area are ever present in the video stream).

MichaIng, 2022-04-17:
Ah, thanks for clarification, I didn't know the motion detection mask yet.

JamBalaya56562, 2026-11-10:
Closing, as this got no reply to zagrim's explanation: the mask under Video Device is a privacy mask, which only hides the area in the picture, while the one under Motion Detection is what keeps the road and sidewalk out of detection. Please open a new issue if the Motion Detection mask still adds false alarms with a current release.