motioneye-project/motioneye #2418: Telegram notification python error

jcroy, 2022-04-14:
Out of the box telegram integration fails. I followed the install instructions closely, so I don't think I'm missing something. 

```
ERROR: Exception in callback make_message.<locals>.on_media_files(<Future finished result=[]>) at /usr/local/lib/python3.9/dist-packages/motioneye/sendtelegram.py:74
handle: <Handle make_message.<locals>.on_media_files(<Future finished result=[]>) at /usr/local/lib/python3.9/dist-packages/motioneye/sendtelegram.py:74>
Traceback (most recent call last):
  File "/usr/lib/python3.9/asyncio/events.py", line 80, in _run
    self._context.run(self._callback, *self._args)
  File "/usr/local/lib/python3.9/dist-packages/motioneye/sendtelegram.py", line 78, in on_media_files
    timestamp = time.mktime(moment.timetuple())
AttributeError: 'str' object has no attribute 'timetuple'
```

MichaIng, 2022-04-14:
Many thanks for your report.

Here is the related code line: https://github.com/motioneye-project/motioneye/blob/16fd474/motioneye/sendtelegram.py#L78
This function is called when the script is loaded, `moment` is parsed from options: https://github.com/motioneye-project/motioneye/blob/16fd474/motioneye/sendtelegram.py#L198

`timetuple` is a method of `datetime`'s `time` object: https://docs.python.org/3/library/datetime.html#datetime.date.timetuple
Looks like `moment` is not assured to be a `time` object. It'l take some time for me to have a deeper look into this, where `moment` is initially created, stored, parsed and manipulated.

@sirdel
Since you fixed storing the Telegram settings, does this work for you?

sirdel, 2022-04-16:
I've fixed a few other bugs with the sendtelegram script and also added a check to the argument parser to ensure "moment" is in isoformat

MichaIng, 2022-04-20:
Fixed for next release 👍.

PR #2420 (2022-04-20): Send telegram bug fix
Fix a few more issues with the telegram script and tided it up, removing old "email" refrences. Should also fix #2418
closed 2022-04-20 by MichaIng