motioneye-project/motioneye #2421: No recording "view now" option in browser with 0.43

zaphod24, 2022-04-17:
I recently updated to 0.43 / Python3 and there is no longer an option to watch a recording directly in the browser, there is only an option to download the recording. Is that intentional?

zagrim, 2022-04-18:
I don't think that is intentional, and I seem to have the same issue in my development setup :thinking:

zagrim, 2022-04-18:
Seems like something has ben broken, as when the dialog should open there's an error in browser console (my case was opening a saved still image):

```
jquery.min.js?v=0.43.0:2 Uncaught TypeError: e.indexOf is not a function
    at S.fn.init.S.fn.load (jquery.min.js?v=0.43.0:2:84932)
    at runPictureDialog (main.js?v=0.43.0:3779:9)
    at HTMLDivElement.<anonymous> (main.js?v=0.43.0:4401:25)
    at HTMLDivElement.dispatch (jquery.min.js?v=0.43.0:2:43064)
    at HTMLDivElement.v.handle (jquery.min.js?v=0.43.0:2:41048)
```
The code at the [line 3779 of main.js](https://github.com/motioneye-project/motioneye/blob/a93729c533d5878354001b7cba48a795ee3627c4/motioneye/static/js/main.js#L3779) is `img.load(updateModalDialogPosition);`

I have other things planned for now so I will not dig any deeper at this time, but perhaps someone else can continue from here. What puzzles me is how the UI can have broken when it has now really been that much changed in dev...

zaphod24, 2022-04-19:
Would be great for this to work again as I watch after I get alerts on my phone

tunkio, 2022-04-19:
You can fix it by changing img.load(updateModalDialogPosition); to img.on('load', updateModalDialogPosition);

The load method is removed in jQuery version 3. jQuery is updated at Mar 19, 2022, so it might cause some other odd UI related things.

MichaIng, 2022-04-19:
Many thanks for providing the solution. The jQuery upgrade caused this by deprecating the old event syntax. I thought I went through all web UI pages/elements to watch out for browser console errors after the upgrade, but seems I missed that one.

zaphod24, 2022-04-23:
Made the change above, definitely working better. Is it intentional that it plays at 5x speed and that it plays all recordings back to back for that day?

MichaIng, 2022-04-23:
Many thanks, I added further jQuery 3 related changes across all JavaScript files.

Not sure about whether playback speed and playing all recordings is intended. I'd need to look into the code, but not before next week.

tunkio, 2022-04-23:
Speed and auto continue are hard coded.

```
mPlayer.playbackRate = 5;
video_container.on('ended', function() {
    if( pos > 0 ) {
        nextArrow.click();
        playButton.click();
       mPlayer.playbackRate = 5;
    }
});
```

tunkio, 2022-04-23:
We could add control for speed and continuous play but for now you have possibility to reduce speed by using browser own features. In firefox you need to right click video control bar and adjust speed from menu(0.5, 1, 1.25, 1.5, 2). In Chrome and Edge you need to click three dots in right side of control bar and adjust playback speed from there.

zagrim, 2022-04-24:
As for the 5x speed, I wonder if it's this #2431? It just created that when noticed the issue while testing on the changes.

MichaIng, 2022-04-25:
Merged into `dev`, I'll mark this as closed.

PR #2426 (2022-04-25): Fixes for video and image view now (Issue #2421)
Fix #2421 and broken image view now