motioneye-project/motioneye #2423: Wrong mask setting file read for privacy mask ui values

tunkio, 2022-04-22:
If you enable both motion and privacy mask at the same time, json response for UI will contain motion mask values for both masks(motion and private). Now if you save camera settings both masks will contain same mask(motion area).

Invalid filename forming in **motioneye/utils/__init__.py** function **parse_editable_mask_file**:

```
    file_name = os.path.join(settings.CONF_PATH, 'mask_%s.pgm' % camera_id)
```

Fix:

```
    postfix = "_privacy" if mask_class == "privacy" else ""
    file_name = os.path.join(settings.CONF_PATH, f'mask_{camera_id}{postfix}.pgm')
```

MichaIng, 2022-04-22:
Many thanks, looks right 👍. Could you open a PR with this fix?

zagrim, 2022-04-24:
I can verify that the issue does exist, ran into it the other day but didn't have the time to look into it then and just thought it was some mixup in my dev setup (production running a release without built-in privacy mask support in ME).

tunkio, 2022-04-24:
No problemo! I need to be more precise when reading source code. Missed that filename forming function totally, maybe due there was so many lines on it :) But good catch!

tunkio, 2022-04-30:
This is already fixed. Closing.

PR #2424 (2022-04-24): Fix for invalid motion and privacy ui json values (Issue #2423)
Fix #2423
closed 2022-04-30 by tunkio