motioneye-project/motioneye #2429: Live stream can stall if netcam is offline for a while

sgstair, 2022-04-23:
I saw this issue also described in #1970

Firstly, my system is motionEye 0.42.1 and Motion 4.4.0 on Debian 11. It's connecting to several rtsp network cameras.

While I was setting up the system, I had to disconnect one of the cameras for a while, a few hours or more. When I reconnected the camera, it was clear that motion had reconnected to it - I saw the corresponding rise in CPU usage, and it did record new videos when motion was detected, and was recording pictures (I have an interval snapshot rule enabled) - but it was still displaying the grey "Unable to open video device" message rather than a live view.

Modifying one of the camera settings (triggering a motion restart) did return the live view to working condition.

I did notice that my daemon.log was accumulating logs every 10 seconds:
```
Apr 20 16:36:08 NVR1 meyectl[492]:    ERROR: mjpg client timed out receiving data for camera 2 on port 8082
Apr 20 16:36:18 NVR1 meyectl[492]:    ERROR: mjpg client timed out receiving data for camera 2 on port 8082
Apr 20 16:36:28 NVR1 meyectl[492]:    ERROR: mjpg client timed out receiving data for camera 2 on port 8082
Apr 20 16:36:38 NVR1 meyectl[492]:    ERROR: mjpg client timed out receiving data for camera 2 on port 8082
Apr 20 16:36:48 NVR1 meyectl[492]:    ERROR: mjpg client timed out receiving data for camera 2 on port 8082
```
And motion was also periodically logging also, though unfortunately there are not timestamps. I noticed it changing a few times at the same times the interval snapshots should be taken.
```
[2:ml2:Front] [WRN] [ALL] motion_init: Could not fetch initial image from camera
[2:ml2:Front] [WRN] [ALL] motion_init: Motion continues using width and height from config file(s)
[2:ml2:Front] [WRN] [ALL] mlp_retry: Retrying until successful connection with camera
[2:ml2:Front] [ERR] [NET] netcam_rtsp_open_context: norm: Unable to open camera(Front): No route to host
[2:ml2:Front] [ERR] [VID] vid_start: Netcam RTSP failed to open
[2:ml2:Front] [ERR] [NET] netcam_rtsp_open_context: norm: Unable to open camera(Front): No route to host
[2:ml2:Front] [ERR] [VID] vid_start: Netcam RTSP failed to open
[2:ml2:Front] [WRN] [ALL] motion_init: Could not fetch initial image from camera
[2:ml2:Front] [WRN] [ALL] motion_init: Motion continues using width and height from config file(s)
[2:ml2:Front] [WRN] [ALL] mlp_retry: Retrying until successful connection with camera
[2:ml2:Front] [ERR] [NET] netcam_rtsp_open_context: norm: Unable to open camera(Front): No route to host
[2:ml2:Front] [ERR] [VID] vid_start: Netcam RTSP failed to open
```

Unfortunately I lost the motion log from when the situation started; my guess is that the error progressing from "no response from host" to "no route to host" might have interrupted the mjpg stream in a way that motioneye does not account for, and so it hung open without trying to reconnect.
I looked over the code at motioneye/mjpgclient.py - and see that it doesn't seem to have a mechanism to reconnect a hung stream that isn't closed (besides restarting motion). Technically it may be a motion bug though, I'm not sure yet.

I quickly tried to reproduce the error on my setup, but my conclusion is that more than a 15 minute outage is required to produce this behavior. I observed the camera feed recovering correctly in all of my shorter tests. 

I experienced this problem twice though in setting up this system though, I think both cases had a netcam outage of more than an hour. I'll probably try to reproduce this to get some more information about what's going wrong.

MichaIng, 2022-04-24:
Many thanks for your report. We need to have a closer look into the code. Probably motionEye gives up after a certain amount of connection retries, or it increases the retry interval so that you just need to wait longer when the camera is available again.

Generally I think it's fine to keep retrying with an interval that doesn't imply too heavy resource and network overhead. After a few shorter retires, it could be even muted (degraded to debug) in logs, with an initial log like "failed to connect to camera X times, silently retrying every X seconds from now on" or so, so that logs are not spammed throughout a longer period?

sgstair, 2022-04-24:
Hey @MichaIng :)
from what I understand, that log line `ERROR: mjpg client timed out receiving data...` can only occur when motioneye is connected, but the server is just not providing frames for some reason. So the connection is never closed in this case, I think.

I'm wondering if maybe it would be smart to force close a client that doesn't receive frames for a substantial time so it would retry, but this feels more like a workaround.

MichaIng, 2022-04-24:
I'm not sure whether it currently is really unable to receive data as fast as the network camera is reachable again, of whether in some cases it would require to close the stream and start freshly. In the latter case, since `motion` already has the issue, I guess the respective `motion` process would need to be restarted. E.g. after a certain amount of timeouts, this could be done automatically, but probably not over and over throughout a whole night.

sgstair, 2022-05-03:
I did attempt to repro the other day, left a camera unplugged for about 21 hours. System still recovered correctly afterwards; there were a few notable differences though that may suggest the way to cause this problem.
I only got this log line from motion:
`May  2 08:00:43 NVR1 motion: [9:nc9:Back West] [ERR] [NET] netcam_rtsp_handler_reconnect: norm: Reconnecting with camera....`
rather than the `Could not fetch initial image from camera` message before
Also, in the broken case, motion was not saving the scheduled interval snapshots, but in this case it was.

Probably the repro is to disconnect the camera, and then restart motioneye (or change configuration to restart motion); I'll try again later today probably.

sgstair, 2022-05-03:
I have some further data, my repro was as follows:
* Disconnect a network camera
* Restart motioneye
* After some time, reconnect network camera
I observe that motion reconnects to the camera successfully, and was able to take an interval snapshot; but the preview in motioneye UI is still the grey `Unable to open video device`
Motioneye was logging the mjpg timeout message
`May  3 11:03:52 NVR1 meyectl[377399]:    ERROR: mjpg client timed out receiving data for camera 3 on port 8083`
However! When I opened a browser to this port to test it, it worked correctly, and that also seems to have caused the stream to recover for motioneye.

This suggests to me that there's something getting dropped in `motion` when it starts without being able to connect to the rtsp stream and has a client connect to the mjpg stream before it's received any frames. And it probably should be reconnecting something also when the camera stream starts, and not only when a request is made for the mjpg/etc stream. 
I'll take a closer look when I get a chance

sgstair, 2022-05-05:
I noticed something else important; in the repro scenario above, the mjpg stream shown by motioneye seems to only update when I also have the browser open to that port to access the motion stream directly. It pauses/freezes on the last provided frame when I close that browser. 
And I'm getting periodic logs in motioneye
`NVR1 meyectl[377399]:  WARNING: mjpg client connection for camera 3 on port 8083 is closed`
So, motioneye is reopening the stream in this case, and that's not helping. Will try to dig into the motion code soon.

MichaIng, 2022-05-09:
Like the MJPEG stream goes into sure sleeping mode and motionEye does not prevent that, but the browser (with a fresh connection) wakes it up 🤔.

I really think a reconnection attempt loop when gaining the next frame fails for some time is a good idea. But probably there is another smarter way to keep the stream/connection up in the first place.

sgstair, 2022-05-19:
I found a possible desynchronization in `motion`, but it doesn't really explain the behavior I'm seeing. 
Just leaving some notes here for future reference. Short on time at the moment but I'll dig more through the history and related code in a bit.

The stream `cnct_count` property is potentially unbalanced, `webu_mhd_deinit` may decrease the connection count in some rare cases where the count was not incremented; The count should be incremented in `webu_answer_strm_type` (because after this point, a deallocation of the stream will result in the count being decremented) rather than `webu_stream_cnct_count`, or some other method of balancing it.

Some of what I'm seeing would be explained by an unbalance of `cnct_count` (it would prevent new frames from being sent unless a second consumer was started) but I don't currently believe the failures I expect are happening would lead to them, given the current code. Maybe so somehow, I just don't see it yet.

sgstair, 2022-06-16:
I think I was able to clearly identify the problem in motion after running with more verbose logs, see the motion issue for more details

MichaIng, 2022-06-19:
Great, a fix has been committed already: https://github.com/Motion-Project/motion/commit/a511def

closed 2022-06-19 by MichaIng

sgstair, 2022-06-19:
Yup, there are still cases where motion might freeze the stream like this, but they should be exceptionally rare and mostly when people are intentionally reconfiguring the cameras, so much less of an issue.
Thanks for helping & looking into this!