motioneye-project/motioneye #2430: Timelapse video fails on creation

zagrim, 2022-04-24:
I noticed that there seems to be similar "string vs. bytes" issue with timelapse video creation (from saved pics) that have been fixed for other parts of the code, too. When clicking on Timelapse button and then OK on the dialog, there's this error:

```
 WARNING: 403 GET /login/?_=1650788601212&_username=_&_signature=e557c31d147d2a8a67fb92d078ba6842b45279e1 (127.0.0.1) 0.67ms
   ERROR: Exception in callback functools.partial(<function make_timelapse_movie.<locals>.poll_media_list_process at 0x7f710ca551f0>)
Traceback (most recent call last):
  File "/home/etikka/.local/lib/python3.9/site-packages/tornado/ioloop.py", line 741, in _run_callback
    ret = callback()
  File "/home/etikka/.local/lib/python3.9/site-packages/motioneye/mediafiles.py", line 731, in poll_media_list_process
    make_movie(pictures)
  File "/home/etikka/.local/lib/python3.9/site-packages/motioneye/mediafiles.py", line 791, in make_movie
    poll_movie_process(pictures)
  File "/home/etikka/.local/lib/python3.9/site-packages/motioneye/mediafiles.py", line 814, in poll_movie_process
    frame_index = re.findall(r'frame=\s*(\d+)', output)
  File "/usr/lib/python3.9/re.py", line 241, in findall
    return _compile(pattern, flags).findall(string)
TypeError: expected string or bytes-like object
   ERROR: string argument without an encoding
Traceback (most recent call last):
  File "/home/etikka/.local/lib/python3.9/site-packages/tornado/web.py", line 1704, in _execute
    result = await result
  File "/home/etikka/.local/lib/python3.9/site-packages/motioneye/handlers/picture.py", line 70, in get
    await self.timelapse(camera_id, group)
  File "/home/etikka/.local/lib/python3.9/site-packages/motioneye/handlers/picture.py", line 538, in timelapse
    key = mediafiles.set_prepared_cache(status['data'])
  File "/home/etikka/.local/lib/python3.9/site-packages/motioneye/mediafiles.py", line 1052, in set_prepared_cache
    key = hashlib.sha1(bytes(str(time.time()))).hexdigest()
TypeError: string argument without an encoding
   ERROR: 500 GET /picture/1/timelapse/2022-04-18/?check=true&_=1650788623412&interval=60&framerate=20&_username=admin&_signature=36ace2fc0df749c289b6e47d275017a0809b9ecc (127.0.0.1) 3.31ms
```

MichaIng, 2022-04-24:
It should work when replacing `bytes(str(time.time()))` with `str(time.time()).encode()`, being the simpler method for string => bytes encoding which does not require any argument, but uses UTF-8 as format argument by default (fine here, in case of a simple float in string format).

There is no direct float => bytes conversion method, AFAIK.

MichaIng, 2022-04-25:
Indeed, this simple change fixes it. I'll open a PR.

MichaIng, 2022-05-22:
Marking this as closed as it has been solved in `dev` branch.

closed 2022-05-22 by MichaIng