motioneye-project/motioneye #2431: Buttons for playing a saved video and playing it on higher speed (timelapse) on top of each other

zagrim, 2022-04-24:
There seem to be two icons overlapping in the center of video preview (play and timelapse "play * 5 and chain" - what a weird tooltip). 

The icon for timelapse is probably changed inadvertantly, since now it uses the same icon as single-camera view. I wonder if it had the "fast-forward" icon I seem to have removed in https://github.com/motioneye-project/motioneye/pull/1968/files#diff-a773ec03496a82b602d5f994a6401f5d62ecc3420d788a646c7b5da78f059d6c since I didn't realise that the icon was in use even though it wasn't present in the camera button bar.

cryptage21, 2022-04-24:
I confirm, buttons are overlapped. 

https://i.ibb.co/z80MkQF/Buttons-Overlapped.png

MichaIng, 2022-04-29:
Fixed in `dev` 🙂.

PR #2445 (2022-04-29): Fix for overlapping play and timelapse buttons (Issue #2431)
Added missing timelapse button and fixed overlapping play and timelapse buttons. Fix #2431
commit 1ae38313 (2022-04-29): Fix #2431

Added missing timelapse button and fixed overlapping play and timelapse buttons.
closed 2022-04-29 by MichaIng