motioneye-project/motioneye #2434: UI doesn't show camera when editable mask is enabled

af7567, 2022-04-24:
Using a fresh install with no cameras set up, I added my camera which worked fine.
I then went to turn on the editable mask and edit and save the mask file.
Now when visiting the motioneye page I just get the message "An error occurred. Refreshing is recommended."

It seems like the calculation for the mask lines isn't working right and it trying to access pixels from the mask file which don't exist. I have checked the mask file which was created and it has the right dimensions. In this case my camera is 1536 x 2048

motioneye.log
```
2022-04-24 20:22:40: [motioneye]    ERROR: list index out of range
Traceback (most recent call last):
  File "/usr/lib64/python3.9/site-packages/tornado/web.py", line 1704, in _execute
    result = await result
  File "/usr/lib64/python3.9/site-packages/motioneye/handlers/config.py", line 49, in get
    await self.list()
  File "/usr/lib64/python3.9/site-packages/motioneye/handlers/config.py", line 476, in list
    ui_config = config.motion_camera_dict_to_ui(local_config)
  File "/usr/lib64/python3.9/site-packages/motioneye/config.py", line 1599, in motion_camera_dict_to_ui
    ui['motion_mask_lines'] = utils.parse_editable_mask_file(
  File "/usr/lib64/python3.9/site-packages/motioneye/utils/__init__.py", line 620, in parse_editable_mask_file
    pixel = pixels[py * width + px]
IndexError: list index out of range
```

motion still runs OK and the mask is working, it is just the motioneye UI which is inaccessible.

MichaIng, 2022-04-24:
Many thanks for your report.

When did you install motionEye, does it contain this change already? https://github.com/motioneye-project/motioneye/pull/2424/files

Also did you use the privacy mask or the motion mask or both?

I think it's not related, but since your failure [is in the remainder handling](https://github.com/motioneye-project/motioneye/blob/8976ce3/motioneye/utils/__init__.py#L615-L620), there is another issue I faced with it: https://github.com/motioneye-project/motioneye/pull/2403#issuecomment-1089056122
If the camera width cannot be divided by 32, the remainder is not handled correctly, loading to this too large last left and bottom rectangles with small cameras.

The reason why there is a remainder at all in your case is this change: https://github.com/motioneye-project/motioneye/pull/2403/files#diff-4e98d8ae5ebf19f5f90280ddf3865c4e51d4f39332a15a1477f19e0149307683R580
This allowed us to remove the `numpy` dependency. The remainder itself however shouldn't cause an issue of course, and my thoughts about a simpler mask grid processing should harden the whole function much.

af7567, 2022-04-25:
I upgraded motioneye yesterday to test it before making the bug report. I just checked on the server and it does have the change you mentioned already.

I don't have the privacy mask enabled, only the motion mask. The mask options in my camera-1.conf are:

```
mask_privacy
smart_mask_speed 0
mask_file "/etc/motioneye/mask_1.pgm"
```

I just tried changing the 6 lines with range() to use numpy.arange() again in `build_editable_mask_file` and `parse_editable_mask_file` to test but still get the index out of range error.

tunkio, 2022-04-25:
I have same files and my setup works. Is it possible if you could zip that mask file and upload it here?

af7567, 2022-04-25:
Yes, the camera is 1536 x 2048, and no rotation done by motion
```
width 1536
height 2048
rotate 0
```
I have uploaded the mask file created with motioneye, and also a blank mask file automatically created by motioneye when I removed the file before.
[masks.zip](https://github.com/motioneye-project/motioneye/files/8556276/masks.zip)
With both these mask files I get the out of range error.

tunkio, 2022-04-25:
Yeah, orientation is causing this problem. Mask file doesn't affect and seems to work properly but if I swap my existing camera height and width I get exactly same error.

MichaIng, 2022-04-25:
Good catch. Let me open a PR with the suggested simpler mask grid/remainder handling, which probably solves the issue already. It should reduce a lot of arithmetic as well, making it easier to understand how values are processed.

af7567, 2022-04-25:
I just tested on my server swapping the width and height, and editing + saving the mask again. All seems to work OK with width 2048 and height 1536 for me too (apart from the image looking squashed ofc :) )

jmichault, 2022-05-13:
@af7567 , can you try the branch "2434-ui-doesnt-show-camera-when-editable-mask-is-enabled", and test with both orientation ?

af7567, 2022-05-13:
I have just tested that branch with both 1536x2048 and 2048x1536 and it seems to work OK :)
I created some mask files with the UI too for each orientation and checked them in an image viewer and they appear as expected.

jmichault, 2022-05-14:
Thanks, so i commit the pull request and close this issue.

closed 2022-05-14 by jmichault