motioneye-project/motioneye #2436: Installation problem

nmg196, 2022-04-25:
I'm not hugely familiar with Linux and I'm struggling to run MotionEye (I used to use motionEye OS but my new camera needs custom drivers). 

I think I've followed all the instructions OK for Raspberry Pi 11 Lite (Bullseye) but I get errors logged in the startup (see below).

I'm confused if it should still be running on Python 2.7  or 3.9 but I think I have both installed and 2.7 seems to be the default if I just type 'python' in bash.  However in the error logs below it references python 3.9 and I'm not sure why.

What am I doing wrong? Any help greatfully appreciated!

`-- Boot 75bc2c65d307492294a837d96c48f4f8 --
Apr 23 18:22:13 motioneye2 systemd[1]: Started motionEye Server.
Apr 23 18:22:13 motioneye2 meyectl[382]: Traceback (most recent call last):
Apr 23 18:22:13 motioneye2 meyectl[382]:   File "/usr/local/bin/meyectl", line 5, in <module>
Apr 23 18:22:13 motioneye2 meyectl[382]:     from motioneye.meyectl import main
Apr 23 18:22:13 motioneye2 meyectl[382]:   File "/usr/local/lib/python3.9/dist-packages/motioneye/meyectl.py", line 28, in <module>
Apr 23 18:22:13 motioneye2 meyectl[382]:     import settings
Apr 23 18:22:13 motioneye2 meyectl[382]: ModuleNotFoundError: No module named 'settings'
Apr 23 18:22:13 motioneye2 systemd[1]: motioneye.service: Main process exited, code=exited, status=1/FAILURE
Apr 23 18:22:13 motioneye2 systemd[1]: motioneye.service: Failed with result 'exit-code'.
Apr 23 18:23:24 motioneye2 systemd[1]: Started motionEye Server.
Apr 23 18:23:24 motioneye2 meyectl[812]: Traceback (most recent call last):
Apr 23 18:23:24 motioneye2 meyectl[812]:   File "/usr/local/bin/meyectl", line 5, in <module>
Apr 23 18:23:24 motioneye2 meyectl[812]:     from motioneye.meyectl import main
Apr 23 18:23:24 motioneye2 meyectl[812]:   File "/usr/local/lib/python3.9/dist-packages/motioneye/meyectl.py", line 28, in <module>
Apr 23 18:23:24 motioneye2 meyectl[812]:     import settings
Apr 23 18:23:24 motioneye2 meyectl[812]: ModuleNotFoundError: No module named 'settings'
Apr 23 18:23:24 motioneye2 systemd[1]: motioneye.service: Main process exited, code=exited, status=1/FAILURE
Apr 23 18:23:24 motioneye2 systemd[1]: motioneye.service: Failed with result 'exit-code'.
Apr 23 18:28:04 motioneye2 systemd[1]: Started motionEye Server.
Apr 23 18:28:04 motioneye2 meyectl[868]: Traceback (most recent call last):
Apr 23 18:28:04 motioneye2 meyectl[868]:   File "/usr/local/bin/meyectl", line 5, in <module>
Apr 23 18:28:04 motioneye2 meyectl[868]:     from motioneye.meyectl import main
Apr 23 18:28:04 motioneye2 meyectl[868]:   File "/usr/local/lib/python3.9/dist-packages/motioneye/meyectl.py", line 28, in <module>
Apr 23 18:28:04 motioneye2 meyectl[868]:     import settings
Apr 23 18:28:04 motioneye2 meyectl[868]: ModuleNotFoundError: No module named 'settings'
Apr 23 18:28:04 motioneye2 systemd[1]: motioneye.service: Main process exited, code=exited, status=1/FAILURE`

MichaIng, 2022-04-26:
How did you install motionEye? The code matches the [python2](https://github.com/motioneye-project/motioneye/blob/python2/motioneye/meyectl.py#L28) branch, but you're obviously installed motionEye for Python 3.

It is btw common that `python` is the Python 2 command while `python3` is the Python 3 command.

Please follow [these](https://github.com/motioneye-project/motioneye/tree/dev#installation) instructions to install the Python 3 compatible development version of motionEye.

If you want to use it with Python 2, assure to install it correctly with `pip2` for Python 2. But since Python 2 has become EOL since beginning 2020, I would strongly recommend to use Python 3.

nmg196, 2022-04-26:
Thanks - I didn't find those instructions so maybe I found some old ones for python 2.

I've followed the linked instructions but when I run `sudo motioneye_init`  I get the following error:

```
Message from syslogd@motioneye2 at Apr 26 09:04:36 ...
 motion[1896]: [0:motion] [EMG] [ALL] motion_startup: Exit motion, cannot create log file /var/log/motion/motion.log: Permission denied
```

closed 2022-04-26 by nmg196