motioneye-project/motioneye #2438: Camera compatibility with DFRobot low light camera

nmg196, 2022-04-26:
Does anyone know if MotionEye should be compatible with this camera:
https://wiki.dfrobot.com/Night%20Camera%20Module%20for%20Raspberry%20Pi%20SKU:%20SEN0338

The camera works with it's own software (veye) but in MotionEye, I don't see any cameras recognised under MMAL or V4L2 :(

Thanks!
Nick

nmg196, 2022-04-26:
Ahh OK - I got it working by installing the V4L2 driver using these instructions: http://wiki.veye.cc/index.php/V4L2_mode_for_Raspberry_Pi

The frame rate doesn't seem that great though. Looks like it's about 5 FPS even when set to 25 in MotionEye.

MichaIng, 2022-04-26:
Is the original camera stream as well at 5 FPS? You should see one value for the original stream and one served by motionEye. Of course it can be lower due to network limitations e.g.

nmg196, 2022-04-26:
Hmm I'm not sure where I'd see those figures?  I doubt it's network limitations as the Pi is plugged into Gigabit ethernet.

MichaIng, 2022-04-28:
The figures are only visible when the stream is there, of course, i.e. you need to set the resolution to a supported one. The supported resolutions for V4L2 cameras are btw obtained via:
```sh
v4l2-ctl -d /dev/video0 --list-formats-ext | grep -vi stepwise | grep -oE '[0-9]+x[0-9]+'
```
if the camera is provided via `/dev/video0`. If for some reason this doesn't output anything, a common resolution list is shown, which doesn't say whether any of them is really supported or not.

Another question, do you use the current stable Python 2 version of motionEye, or the new Python 3 compatible version (not released to PyPI yet), installable via [these instructions](https://github.com/motioneye-project/motioneye/tree/dev#installation)?

nmg196, 2022-04-29:
Hi,
It seems my camera only supports 1920x1080

[image]

I'm using the Python 3.7 version from those instructions. 

I'm also not getting any recordings for some reason with these settings:

[image]

Thanks

MichaIng, 2022-04-29:
Can you please share the service logs:
```sh
journalctl -u motioneye
```

nmg196, 2022-04-29:
Yeah here's a screenshot of today's log. The red part repeats for hours:

[image]

MichaIng, 2022-04-30:
You need to select a different "Movie format" in the settings. AFAIK H.264/OMX one is only supported with the Raspberry Pi FFmpeg build. Would actually make sense to check for this once at motionEye start and show a the regular H.264 codec if it's not an RPi, respectively if `ffmpeg` doesn't list the OMX as supported format.

nmg196, 2022-05-03:
Thanks - that fixed it :)

nmg196, 2022-05-03:
The frame rate is still really slow for some reason. I will post a sample video so you can see what I mean. The stream is reported at being close to 29FPS, but in reality it's nothing close to that - maybe 4-5 frames per second.  I'm wondering if the exposure duration is actually 1/5th of a second or something and that's causing the 5FPS effect, but that wouldn't explain the 29FPS reported by motionEye.

MichaIng, 2022-05-03:
There should be two frame rates shown in motionEye, one is the intended frame rate and one is what finally appears at the frontend. Are both reported at close to 29 FPS?

nmg196, 2022-05-03:
Yes, both say 30ish: 
[image]

MichaIng, 2022-05-03:
Okay, then I'm not sure whether it's the camera itself providing a new picture only at 5 FPS while it is possible to grab (then repeating) pictures at 30 FPS. Clarification could bring watching the `motion` stream directly instead of the motionEye stream, and testing with another camera software, like mjpg-streamer, whether it similarly appears to be as well 5 FPS in real.

JamBalaya56562, 2026-11-10:
Closing, as the camera works with veye's V4L2 driver, as you found. For the 5 fps you saw while motionEye reports about 30, watching Motion's own stream or another tool such as mjpg-streamer tells whether the camera itself delivers new frames that slowly.

closed 2026-11-10 by JamBalaya56562