motioneye-project/motioneye #2439: Buster pi 4b doesn't start python errors

bulgin, 2022-04-26:
Hello.  Installed buster on pi 4b.  Lite (Buster Legacy).  Will not start.

See errors:
systemctl status motioneye
● motioneye.service - motionEye Server
   Loaded: loaded (/etc/systemd/system/motioneye.service; enabled; vendor preset: enabled)
   Active: failed (Result: exit-code) since Tue 2022-04-26 13:56:39 EDT; 12s ago
  Process: 625 ExecStart=/usr/local/bin/meyectl startserver -c /etc/motioneye/motioneye.conf (code=exited, status=1/FAILURE)
 Main PID: 625 (code=exited, status=1/FAILURE)

Apr 26 13:56:38 575 meyectl[625]:     if get_camera(camera_id):
Apr 26 13:56:38 575 meyectl[625]:   File "/usr/local/lib/python2.7/dist-packages/motioneye/config.py", line 409, in get_camera
Apr 26 13:56:38 575 meyectl[625]:     _set_default_motion_camera(camera_id, camera_config)
Apr 26 13:56:38 575 meyectl[625]:   File "/usr/local/lib/python2.7/dist-packages/motioneye/config.py", line 1872, in _set_default_motion_camera
Apr 26 13:56:38 575 meyectl[625]:     data.setdefault('target_dir', os.path.join(settings.MEDIA_PATH, data['camera_name']))
Apr 26 13:56:38 575 meyectl[625]:   File "/usr/lib/python2.7/posixpath.py", line 68, in join
Apr 26 13:56:38 575 meyectl[625]:     if b.startswith('/'):
Apr 26 13:56:38 575 meyectl[625]: AttributeError: 'int' object has no attribute 'startswith'
Apr 26 13:56:39 575 systemd[1]: motioneye.service: Main process exited, code=exited, status=1/FAILURE
Apr 26 13:56:39 575 systemd[1]: motioneye.service: Failed with result 'exit-code'.

Could use some help.  Thanks.

MichaIng, 2022-04-26:
Can you please try the new Python 3 version of motionEye, following the instructions given here: https://github.com/motioneye-project/motioneye/tree/dev#installation

As of limited development time, I'm not keen to debug the Python 2 version, but instead aiming to get a stable Python 3 release up ASAP 😉.

bulgin, 2022-04-27:
That's great! It works! Thanks!

MichaIng, 2022-04-28:
Great. I'll mark this issue as closed then. Feel free to reopen if required.

closed 2022-04-28 by MichaIng