motioneye-project/motioneye #2444: apt missed

bad-friday, 2022-04-29:
it seems that the apt-packet is missing

MichaIng, 2022-04-30:
Is this question addressing motionEye or the distribution you are using? `apt` is the package manager for Debian based distribution, like Debian, Raspberry Pi OS, Ubuntu, Kali Linux and such. Other distributions use other package managers.

bad-friday, 2022-04-30:
I'd try to state it's on a Raspery Pi 4.
I didn't realize that motioneye isn't based on Debian.
Is there a packet manager for motioneye?
Also irritating for me the filesystem seems to be read only

MichaIng, 2022-04-30:
Ah so you so not man motionEye: the application with can be installed in any (Linux) OS, but motionEyeOS, an actual OS based om Buildroot, with mainly motionEye installed.

Bear with me that I cannot say much about motionEyeOS, but AFAIK it is more to be used like a container which comes with everything required for motionEye to run pre-installed. So the root filesystem is read-only and only sone specific data filesystems are read-write mounted.

I'll move your question over to the motionEyeOS repository where hopefully others can further clarify.