motioneye-project/motioneye #2447: How can I help translating?

Xitee1, 2022-04-29:
I would like to help translating into german.
How can I do that? Is there already some translating website available like crowdin or something?

MichaIng, 2022-04-30:
Many thanks for offering help 👍.

There is no website yet, but that would be actually nice. Currently it's cloning the repo, manually editing the `.po` files like https://github.com/motioneye-project/motioneye/blob/dev/motioneye/locale/en/LC_MESSAGES/motioneye.po, then compiling them via [make](https://github.com/motioneye-project/motioneye/blob/dev/Makefile) and committing via PR changes back to this repository. What I am not sure about is whether there is a nice GUI application for doing these translation updates instead of editing the files manually? A website would of course be awesome.

jmichault, 2022-05-01:
Hello,
Actually, we don't have a translating website.
To help translating, i suggest to follow these steps : 
1) make your own fork : 
  click on button «fork» on top of page, and follow instructions.
2) clone your fork on your linux machine , and change to the dev branch : 
```
git clone https://github.com/Xitee1/motioneye.git
cd motioneye
git checkout dev
```
3) edit the two po files `motioneye/locale/de/LC_MESSAGES/motioneye.js.po` and `motioneye/locale/de/LC_MESSAGES/motioneye.po`
4) commit and push your changes : 
```
git add motioneye/locale/de/LC_MESSAGES/motioneye.js.po
git add motioneye/locale/de/LC_MESSAGES/motioneye.po
git commit -m 'german translation'
git push
```
5) propose your changes as a pull request : 
On github, being on your fork, click on «Pull Requests», then on «New Pull Request». Select the dev branch on both sides, and click on «Create Pull Request», and follow instructions.

Notes for step 3 : 
* it's possible to edit files manually, but it's easier to use a poeditor like [poedit](https://poedit.net/) or [virtaal](https://virtaal.translatehouse.org/)
* original text is in esperanto. All texts are pre-translated with google translate to help you, so you usually have only to correct google errors, but if you have any doubts, you can : 
  * see the english translation in the corresponding po file in motioneye/locale/en/LC_MESSAGES/
  * use the multilingual esperanto online dictionary <https://www.reta-vortaro.de/>
  * consult the [esperanto grammar](https://www.akademio-de-esperanto.org/fundamento/) \(only 16 rules\)
  * and why not learn the basics of esperanto with a 10-hour training, like <https://esperanto12.net/>
 
If you want to test the result of your work, you can update your local motioneye with this commands : 
```
make
sudo systemctl stop motioneye
sudo python3 -m pip install --upgrade --force-reinstall --no-deps .
sudo systemctl start motioneye
```

Xitee1, 2022-05-01:
Thanks. I will try that.

jmichault, 2022-05-03:
I have created a translation project on weblate : <https://hosted.weblate.org/projects/motioneye-project/>.
Probably easier for you to use.

MichaIng, 2022-05-03:
Wow, awesome, looks very good. I'll start updating German translations tomorrow 🚀.

NicoWeij, 2022-09-22:
After changing the language in the browser and saving it, the screen is refreshed and only the preferences are visible.
Where can I change the language back to English via SSH.

MichaIng, 2022-09-22:
Could you please open a new issue about this? Let's keep individual things separated 😉. I'll mark this as closed as the original question has been answered.

closed 2022-09-22 by MichaIng