motioneye-project/motioneye #2448: Webcam detected by Linux but not by Motioneye

MacAre, 2022-04-30:
Hello,
My Logitech QuickCam E2500 webcam which worked with Buster 32 bit on Raspberry PI3 is not detected by Motioneye on Bullseye.
Do you know where the problem could come from? thanks for your advices.

New installation done yesterday.
motionEye 0.42.1
Motion 4.3.2
Raspian 11 (5.15.32-v7+ #1538 SMP Thu Mar 31 19:38:48 BST 2022 armv7l GNU/Linux) 32bit with no desktop environment.

**ls /dev**
```
autofs         disk       gpiomem  loop3         media2     ptmx   ram15  random   tty    tty16  tty24  tty32  tty40  tty49  tty57  tty8       v4l     vcs5   vcsm-cma  video10  video21
block          dma_heap   hwrng    loop4         mem        pts    ram2   rfkill   tty0   tty17  tty25  tty33  tty41  tty5   tty58  tty9       vchiq   vcs6   vcsu      video11  video22
btrfs-control  dri        initctl  loop5         mmcblk0    ram0   ram3   serial   tty1   tty18  tty26  tty34  tty42  tty50  tty59  ttyACM0    vcio    vcsa   vcsu1     video12  video23
bus            fd         input    loop6         mmcblk0p1  ram1   ram4   serial1  tty10  tty19  tty27  tty35  tty43  tty51  tty6   ttyACM1    vc-mem  vcsa1  vcsu2     video13  video31
cachefiles     full       kmsg     loop7         mmcblk0p2  ram10  ram5   shm      tty11  tty2   tty28  tty36  tty44  tty52  tty60  ttyAMA0    vcs     vcsa2  vcsu3     video14  watchdog
cec0           fuse       log      loop-control  mqueue     ram11  ram6   snd      tty12  tty20  tty29  tty37  tty45  tty53  tty61  ttyprintk  vcs1    vcsa3  vcsu4     video15  watchdog0
char           gpiochip0  loop0    mapper        net        ram12  ram7   stderr   tty13  tty21  tty3   tty38  tty46  tty54  tty62  uhid       vcs2    vcsa4  vcsu5     video16  zero
console        gpiochip1  loop1    media0        null       ram13  ram8   stdin    tty14  tty22  tty30  tty39  tty47  tty55  tty63  uinput     vcs3    vcsa5  vcsu6     video18
cuse           gpiochip2  loop2    media1        ppp        ram14  ram9   stdout   tty15  tty23  tty31  tty4   tty48  tty56  tty7   urandom    vcs4    vcsa6  vhci      video20
```

**sudo lsusb -v | more**
```
Bus 001 Device 005: ID 046d:089d **Logitech, Inc. QuickCam E2500 series**
Device Descriptor:
  bLength                18
  bDescriptorType         1
  bcdUSB               1.10
  bDeviceClass            0
  bDeviceSubClass         0
  bDeviceProtocol         0
  bMaxPacketSize0         8
  idVendor           0x046d Logitech, Inc.
  idProduct          0x089d QuickCam E2500 series
  bcdDevice            1.00
  iManufacturer           0
  iProduct                0
  iSerial                 0
  bNumConfigurations      1
  Configuration Descriptor:
    bLength                 9
    bDescriptorType         2
    wTotalLength       0x0150
    bNumInterfaces          3
    bConfigurationValue     1
    iConfiguration          0
    bmAttributes         0xa0
      (Bus Powered)
      Remote Wakeup
    MaxPower              100mA
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        0
      bAlternateSetting       0
      bNumEndpoints           2
      bInterfaceClass       255 Vendor Specific Class
      bInterfaceSubClass    255 Vendor Specific Subclass
      bInterfaceProtocol    255 Vendor Specific Protocol
      iInterface              0
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x81  EP 1 IN
        bmAttributes            1
          Transfer Type            Isochronous
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0000  1x 0 bytes
[... 353 lines cut ...]
        tSamFreq[ 0]        16000
      Endpoint Descriptor:
        bLength                 9
        bDescriptorType         5
        bEndpointAddress     0x83  EP 3 IN
        bmAttributes            1
          Transfer Type            Isochronous
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0020  1x 32 bytes
        bInterval               1
        bRefresh                0
        bSynchAddress           0
        AudioStreaming Endpoint Descriptor:
          bLength                 7
          bDescriptorType        37
          bDescriptorSubtype      1 (EP_GENERAL)
          bmAttributes         0x00
          bLockDelayUnits         0 Undefined
          wLockDelay         0x0000
Device Status:     0x0000
  (Bus Powered)
``
![2022-04-30 20_53_30-Add camera](https://user-images.githubusercontent.com/10264829/166119645-a24ee394-edcd-4831-89f9-ae5a175f5ca2.png)
`

MichaIng, 2022-04-30:
This is a USB camera, right?

That such a large number of (non-functional) /dev/video* and /dev/media* are shown on RPi has been reported here already: #2438
But this was attached via RPi camera slot and not a regular USB camera, so the solution to switch to legacy camera module stack doesn't work in your case. I'm hence still, or specially here puzzled how one is supposed to access a camera that is spread across 16 kernel devices for some reason... Looks like I need to check whether my USB cameras as well add 16 devices on my RPi 2 and RPi Zero W, probably it is a kernel/firmware bug. Like that we cannot manage it via `v4l2-ctl`.

MacAre, 2022-04-30:
Yes, it's a USB webcam
I just tested with Raspbian Lite Bullseye 64 bit, it's the same. Linux detects the USB webcam but not MotionEye.

[image]
5.15.32-v8+ #1538 SMP PREEMPT Thu Mar 31 19:40:39 BST 2022 aarch64 GNU/Linux

I will go back to Buster for my PI3.

Several days ago, I installed Bullseye Lite 32 bit on PI0W with another USB webcam model and it works.

MichaIng, 2022-04-30:
Our of interest, if you detach the camera, are all the other devices show still up in the list?

MacAre, 2022-04-30:
I went back to **Buster** so the requested test may not interest you. But here is the result if I unplug the USB webcam.
[image]

Note: there are also 2 arduino (RFLink) connected by USB on the PI3.

MichaIng, 2022-05-01:
These bcm2835 ones sound like provided by RPi camera module driver. Is it enabled, e.g. via `start_x=1` or `camera_auto_detect=1` in `/boot/config.txt`? Or are the USB cameras all shown like this with the same name?

I already thought that we should add the device path as additional identifier, although that such a large number of devices all with the same name even exists doesn't look right to me (issue with RPi firmware/kernel).

MacAre, 2022-05-01:
I don't have these commands in /boot/config.txt
(Warning, I went back to Buster).
```
# For more options and information see
# http://rpf.io/configtxt
# Some settings may impact device functionality. See link above for details

# uncomment if you get no picture on HDMI for a default "safe" mode
#hdmi_safe=1

# uncomment this if your display has a black border of unused pixels visible
# and your display can output without overscan
#disable_overscan=1

# uncomment the following to adjust overscan. Use positive numbers if console
# goes off screen, and negative if there is too much border
#overscan_left=16
#overscan_right=16
#overscan_top=16
#overscan_bottom=16

# uncomment to force a console size. By default it will be display's size minus
# overscan.
#framebuffer_width=1280
#framebuffer_height=720

# uncomment if hdmi display is not detected and composite is being output
#hdmi_force_hotplug=1

# uncomment to force a specific HDMI mode (this will force VGA)
#hdmi_group=1
#hdmi_mode=1

# uncomment to force a HDMI mode rather than DVI. This can make audio work in
# DMT (computer monitor) modes
#hdmi_drive=2

# uncomment to increase signal to HDMI, if you have interference, blanking, or
# no display
#config_hdmi_boost=4

# uncomment for composite PAL
#sdtv_mode=2

#uncomment to overclock the arm. 700 MHz is the default.
#arm_freq=800

# Uncomment some or all of these to enable the optional hardware interfaces
#dtparam=i2c_arm=on
#dtparam=i2s=on
#dtparam=spi=on

# Uncomment this to enable infrared communication.
#dtoverlay=gpio-ir,gpio_pin=17
#dtoverlay=gpio-ir-tx,gpio_pin=18

# Additional overlays and parameters are documented /boot/overlays/README

# Enable audio (loads snd_bcm2835)
dtparam=audio=on

[pi4]
# Enable DRM VC4 V3D driver on top of the dispmanx display stack
dtoverlay=vc4-fkms-v3d
max_framebuffers=2

[all]
#dtoverlay=vc4-fkms-v3d
```
My USB cameras are not displayed in bcm2835 but in USB Camera.

JamBalaya56562, 2026-11-10:
Closing, as you went back to Buster and this was with motionEye 0.42.1. Please open a new issue if a USB webcam that Linux lists under `/dev/video*` is still missing from Add Camera with a current release.

closed 2026-11-10 by JamBalaya56562