motioneye-project/motioneye #2449: Not possible to add more than 3 cameras

sf-206, 2022-05-01:
I have tried to add a fourth camera 
Is this one won't show up. Is this a known bug?
It is working fine with the current release 0.42.1 but not on the current dev. version.

MichaIng, 2022-05-01:
What do you mean by "this one won't show up"? Is the frame added but grey or so? Or is it not added at all? Also are there any related error messages, either in browser console, as red text on the website, or in server logs?
```sh
journalctl -u motioneye
```

sf-206, 2022-05-01:
It is not added at all. How ca I provide logs? Can you please give me a short instruction.

MichaIng, 2022-05-01:
Just to clarify where it fails:
- You select "add camera" from the menu.
- You see the camera in the list and hit "Ok".
- The page reloads and the camera is not there?

Since by default only three cameras are in a row, you cannot scroll down to move it into view? Does it work when you switch "Layout Columns" to show 4 cameras in a row?

And for the logs, run the command I provided above on your servers console, copy&paste the whole output here. At best via SSH client, e.g. in PuTTY you can select the output via mouse, which copies it to clipboard automatically.

sf-206, 2022-05-02:
I am able to "add camera" from the menu.
Camera Configuration shows up on left side. But it won't never show up on the right side. Not even with any symbol or in gray. I have tried scrolling but it is just not there. Tried different Browser (Chrome / Firefox) without any help. 
After a refresh of the browser the UI does not show up completely anymore. Please see screenshot below. I can not select any camera nor does the view change after a while. The progress bar in the middle keeps moving. 

[image]

[0:motion] [NTC] [ALL] conf_load: Processing thread 0 - config file /etc/motioneye/motion.conf
[0:motion] [NTC] [ALL] config_camera: Processing camera config file camera-1.conf
[0:motion] [NTC] [ALL] config_camera: Processing camera config file camera-2.conf
[0:motion] [NTC] [ALL] config_camera: Processing camera config file camera-4.conf
[0:motion] [NTC] [ALL] motion_startup: Logging to syslog
[0:motion] [NTC] [ALL] motion_startup: Motion 4.3.2 Started
[0:motion] [NTC] [ALL] motion_startup: Using default log type (ALL)
[0:motion] [NTC] [ALL] motion_startup: Using log type (ALL) log level (WRN)
[2:ml2:H-Tuere] [WRN] [ALL] motion_init: Pass-through processing disabled.
[3:ml3:Camera4] [WRN] [NET] netcam_next: called with no data in buffer
[3:nc4:Camera4] [WRN] [NET] netcam_read_next_header: Error reading image header, streaming mode (1). Null header.
[3:nc4:Camera4] [WRN] [NET] netcam_read_first_header: Error reading first header ()
[3:nc4:Camera4] [ERR] [NET] netcam_handler_loop: Unrecognized image header (1)
[3:nc4:Camera4] [WRN] [NET] netcam_read_next_header: Error reading image header, streaming mode (1). Null header.
[3:nc4:Camera4] [ERR] [NET] netcam_connect: connect() failed (106): Transport endpoint is already connected
[3:nc4:Camera4] [ERR] [NET] netcam_handler_loop: re-opening camera (streaming)
[2:ml2:H-Tuere] [WRN] [NET] netcam_next: called with no data in buffer
[1:ml1:Carport] [CRT] [NET] motion_init: Substream not available.  Image sizes not modulo 16.
[3:ml3:Camera4] [WRN] [NET] netcam_next: called with no data in buffer
[2:ml2:H-Tuere] [WRN] [NET] Above message repeats 1 times
[2:ml2:H-Tuere] [WRN] [NET] netcam_init_jpeg: no new pic, no signal rcvd
[3:ml3:Camera4] [WRN] [NET] netcam_next: called with no data in buffer
[3:nc4:Camera4] [ERR] [NET] netcam_handler_loop: camera re-connected
[2:ml2:H-Tuere] [WRN] [NET] netcam_init_jpeg: no new pic, no signal rcvd
[3:nc4:Camera4] [WRN] [NET] netcam_read_next_header: Error reading image header, streaming mode (1). Null header.
[3:nc4:Camera4] [WRN] [NET] netcam_read_first_header: Error reading first header ()
[3:nc4:Camera4] [ERR] [NET] netcam_handler_loop: Unrecognized image header (1)
[3:nc4:Camera4] [WRN] [NET] netcam_read_next_header: Error reading image header, streaming mode (1). Null header.
[3:nc4:Camera4] [ERR] [NET] netcam_connect: connect() failed (106): Transport endpoint is already connected
[3:nc4:Camera4] [ERR] [NET] netcam_handler_loop: re-opening camera (streaming)
[2:ml2:H-Tuere] [WRN] [NET] netcam_init_jpeg: no new pic, no signal rcvd
[3:ml3:Camera4] [ERR] [ALL] motion_init: Error capturing first image
[2:ml2:H-Tuere] [WRN] [NET] netcam_init_jpeg: no new pic, no signal rcvd
[3:ml3:Camera4] [WRN] [NET] netcam_init_jpeg: no new pic, no signal rcvd
[2:ml2:H-Tuere] [WRN] [NET] netcam_init_jpeg: no new pic, no signal rcvd
[3:ml3:Camera4] [WRN] [NET] netcam_init_jpeg: no new pic, no signal rcvd
[2:ml2:H-Tuere] [WRN] [NET] Above message repeats 1 times
[2:ml2:H-Tuere] [WRN] [NET] netcam_init_jpeg: no new pic, no signal rcvd
[3:ml3:Camera4] [WRN] [NET] netcam_init_jpeg: no new pic, no signal rcvd
[2:ml2:H-Tuere] [WRN] [NET] netcam_init_jpeg: no new pic, no signal rcvd
[3:ml3:Camera4] [WRN] [NET] netcam_init_jpeg: no new pic, no signal rcvd
[3:nc4:Camera4] [ERR] [NET] Above message repeats 1 times
[3:nc4:Camera4] [ERR] [NET] netcam_handler_loop: camera re-connected
[3:ml3:Camera4] [WRN] [NET] netcam_init_jpeg: no new pic, no signal rcvd
[2:ml2:H-Tuere] [WRN] [NET] netcam_init_jpeg: no new pic, no signal rcvd
[3:ml3:Camera4] [WRN] [NET] netcam_init_jpeg: no new pic, no signal rcvd
[3:nc4:Camera4] [WRN] [NET] netcam_read_next_header: Error reading image header, streaming mode (1). Null header.
[3:nc4:Camera4] [WRN] [NET] netcam_read_first_header: Error reading first header ()
[3:nc4:Camera4] [ERR] [NET] netcam_handler_loop: Unrecognized image header (1)
[3:nc4:Camera4] [WRN] [NET] netcam_read_next_header: Error reading image header, streaming mode (1). Null header.
[3:nc4:Camera4] [ERR] [NET] netcam_connect: connect() failed (106): Transport endpoint is already connected
[3:nc4:Camera4] [ERR] [NET] netcam_handler_loop: re-opening camera (streaming)
[3:ml3:Camera4] [WRN] [NET] netcam_init_jpeg: no new pic, no signal rcvd
[2:ml2:H-Tuere] [WRN] [NET] Above message repeats 2 times
[2:ml2:H-Tuere] [WRN] [NET] netcam_init_jpeg: no new pic, no signal rcvd
[3:ml3:Camera4] [WRN] [NET] netcam_init_jpeg: no new pic, no signal rcvd
[2:ml2:H-Tuere] [WRN] [NET] netcam_init_jpeg: no new pic, no signal rcvd
[3:ml3:Camera4] [WRN] [NET] netcam_init_jpeg: no new pic, no signal rcvd
[3:nc4:Camera4] [ERR] [NET] Above message repeats 3 times
[3:nc4:Camera4] [ERR] [NET] netcam_handler_loop: camera re-connected
[3:ml3:Camera4] [WRN] [NET] netcam_init_jpeg: no new pic, no signal rcvd
[2:ml2:H-Tuere] [WRN] [NET] netcam_init_jpeg: no new pic, no signal rcvd
[3:ml3:Camera4] [WRN] [NET] netcam_init_jpeg: no new pic, no signal rcvd
[... cut ...]
[3:ml3:Camera4] [WRN] [NET] netcam_init_jpeg: no new pic, no signal rcvd
[2:ml2:H-Tuere] [WRN] [NET] netcam_init_jpeg: no new pic, no signal rcvd
[3:ml3:Camera4] [WRN] [NET] netcam_init_jpeg: no new pic, no signal rcvd
[3:nc4:Camera4] [ERR] [NET] Above message repeats 2 times
[3:nc4:Camera4] [ERR] [NET] netcam_handler_loop: camera re-connected
[3:ml3:Camera4] [WRN] [NET] netcam_init_jpeg: no new pic, no signal rcvd
[2:ml2:H-Tuere] [WRN] [NET] netcam_init_jpeg: no new pic, no signal rcvd
[3:ml3:Camera4] [WRN] [NET] netcam_init_jpeg: no new pic, no signal rcvd

MichaIng, 2022-05-02:
There are a lot of packet and connection errors by `motion`. Can you show the motionEye service logs are suggested above:
```sh
journalctl -u motioneye
```

sf-206, 2022-05-04:
I tried a fresh new installation.
1. Raspberry Pi OS instalaed (11 bullseye)
2. Installed MotionEye from installation Guide https://github.com/motioneye-project/motioneye/tree/dev
3. Added Network Camera 1 -> fine
4. Added Network Camera 2 -> fine
5. Added remote MotionEye camera (H-Tuere) -> fine
6. Added Network Camera 4 -> no picture shown as described above.

Please find the logs below. I hope this helps.
pi@raspberrypi:~ $ journalctl -u motioneye
-- Journal begins at Mon 2022-04-04 13:05:58 BST, ends at Wed 2022-05-04 09:33:59 BST. --
May 04 09:27:17 raspberrypi systemd[1]: Started motionEye Server.
May 04 09:27:19 raspberrypi meyectl[3725]: configure_logging cmd motioneye: False
May 04 09:27:19 raspberrypi meyectl[3725]: configure logging to file: None
May 04 09:27:19 raspberrypi meyectl[3725]:     INFO: hello! this is motionEye server 0.43.0
May 04 09:27:20 raspberrypi meyectl[3725]:     INFO: main config file /etc/motioneye/motion.conf does not exist, using default values
May 04 09:27:20 raspberrypi meyectl[3725]:     INFO: cleanup started
May 04 09:27:20 raspberrypi meyectl[3725]:     INFO: wsswitch started
May 04 09:27:20 raspberrypi meyectl[3725]:     INFO: tasks started
May 04 09:27:20 raspberrypi meyectl[3725]:     INFO: mjpg customer garbage collector has started
May 04 09:27:20 raspberrypi meyectl[3725]:     INFO: server started
May 04 09:27:35 raspberrypi meyectl[3725]:  WARNING: 403 GET /login/?_=1651652848120&_username=admin&_signature=268b8d9e832483f9d617990ddbe83b0f2cc3fa5d (109.250.4>
May 04 09:27:41 raspberrypi meyectl[3725]:     INFO: An interrupt signal received, closing …
May 04 09:27:41 raspberrypi systemd[1]: Stopping motionEye Server...
May 04 09:27:43 raspberrypi meyectl[3725]:     INFO: server stopped
May 04 09:27:43 raspberrypi meyectl[3725]:     INFO: tasks stopped
May 04 09:27:43 raspberrypi meyectl[3725]:     INFO: goodbye!
May 04 09:27:43 raspberrypi systemd[1]: motioneye.service: Succeeded.
May 04 09:27:43 raspberrypi systemd[1]: Stopped motionEye Server.
May 04 09:27:43 raspberrypi systemd[1]: motioneye.service: Consumed 4.862s CPU time.
-- Boot bb9f767f908c403fb2aecd49085062a8 --
May 04 09:27:48 raspberrypi systemd[1]: Started motionEye Server.
May 04 09:27:52 raspberrypi meyectl[454]: configure_logging cmd motioneye: False
May 04 09:27:52 raspberrypi meyectl[454]: configure logging to file: None
May 04 09:27:52 raspberrypi meyectl[454]:     INFO: hello! this is motionEye server 0.43.0
May 04 09:27:54 raspberrypi meyectl[454]:     INFO: main config file /etc/motioneye/motion.conf does not exist, using default values
May 04 09:27:54 raspberrypi meyectl[454]:     INFO: cleanup started
May 04 09:27:54 raspberrypi meyectl[454]:     INFO: wsswitch started
May 04 09:27:54 raspberrypi meyectl[454]:     INFO: tasks started
May 04 09:27:54 raspberrypi meyectl[454]:     INFO: mjpg customer garbage collector has started
May 04 09:27:54 raspberrypi meyectl[454]:     INFO: server started
May 04 09:28:08 raspberrypi meyectl[454]:  WARNING: 403 GET /login/?_=1651652883390&_username=admin&_signature=7ce85e43f881d54b2650fe1bf7758704a1941356 (109.250.42>
May 04 09:30:59 raspberrypi meyectl[454]:     INFO: adding new netcam camera with id 1...
May 04 09:30:59 raspberrypi meyectl[454]:     INFO: main config file /etc/motioneye/motion.conf does not exist, using default values
May 04 09:30:59 raspberrypi motion[771]: [0:motion] [NTC] [ALL] conf_load: Processing thread 0 - config file /etc/motioneye/motion.conf
May 04 09:30:59 raspberrypi motion[771]: [0:motion] [NTC] [ALL] config_camera: Processing camera config file camera-1.conf
May 04 09:30:59 raspberrypi motion[771]: [0:motion] [NTC] [ALL] motion_startup: Logging to syslog
May 04 09:30:59 raspberrypi motion[771]: [0:motion] [NTC] [ALL] motion_startup: Motion 4.3.2 Started
May 04 09:30:59 raspberrypi motion[771]: [0:motion] [NTC] [ALL] motion_startup: Using default log type (ALL)
May 04 09:30:59 raspberrypi motion[771]: [0:motion] [NTC] [ALL] motion_startup: Using log type (ALL) log level (WRN)
May 04 09:31:47 raspberrypi meyectl[454]:     INFO: adding new netcam camera with id 2...
May 04 09:31:50 raspberrypi motion[870]: [0:motion] [NTC] [ALL] conf_load: Processing thread 0 - config file /etc/motioneye/motion.conf
May 04 09:31:50 raspberrypi motion[870]: [0:motion] [NTC] [ALL] config_camera: Processing camera config file camera-1.conf
May 04 09:31:50 raspberrypi motion[870]: [0:motion] [NTC] [ALL] config_camera: Processing camera config file camera-2.conf
May 04 09:31:50 raspberrypi motion[870]: [0:motion] [NTC] [ALL] motion_startup: Logging to syslog
May 04 09:31:50 raspberrypi motion[870]: [0:motion] [NTC] [ALL] motion_startup: Motion 4.3.2 Started
May 04 09:31:50 raspberrypi motion[870]: [0:motion] [NTC] [ALL] motion_startup: Using default log type (ALL)
May 04 09:31:50 raspberrypi motion[870]: [0:motion] [NTC] [ALL] motion_startup: Using log type (ALL) log level (WRN)
May 04 09:31:51 raspberrypi motion[870]: [2:ml2:Camera2] [WRN] [NET] netcam_next: called with no data in buffer
May 04 09:32:31 raspberrypi meyectl[454]:    ERROR: HTTP 403: Forbidden
May 04 09:32:31 raspberrypi meyectl[454]: Traceback (most recent call last):
May 04 09:32:31 raspberrypi meyectl[454]:   File "/usr/local/lib/python3.9/dist-packages/tornado/web.py", line 1704, in _execute
May 04 09:32:31 raspberrypi meyectl[454]:     result = await result
[... cut ...]
[3:nc4:Camera4] [WRN] [NET] netcam_read_next_header: Error reading image header, streaming mode (1). Null header.
[3:nc4:Camera4] [ERR] [NET] netcam_handler_loop: Unrecognized image header (1)
[3:ml3:Camera4] [WRN] [NET] netcam_init_jpeg: no new pic, no signal rcvd
[3:nc4:Camera4] [WRN] [NET] Above message repeats 17 times
[3:nc4:Camera4] [WRN] [NET] netcam_read_next_header: Error reading image header, streaming mode (1). Null header.
[3:ml3:Camera4] [WRN] [NET] netcam_init_jpeg: no new pic, no signal rcvd

tunkio, 2022-05-04:
Have you always configured cameras in same order?

Meaning have you always configured following as third?
Added remote MotionEye camera (H-Tuere) -> fine

If so, can you put that last or first and what will happen then? I suppose this is a different type than the others.

sf-206, 2022-05-04:
Tried different order. It always fails on the 4th camera.
Even tried to configure 4 normal network cameras -> same problem.

sf-206, 2022-05-04:
Thank you for the hint. Tried to configure 4 network cameras which worked fine.
As soon as I am configuring a remote MotionEye camera the issue starts -> no further cameras can be added. 
By the way. The other MotionEye client is on the same SW level.

sf-206, 2022-05-04:
Even adding a second reomote MotionEye camera results in the same issue.
So it seems to be only related to remote MotionEye camera configuration.
I hope this helps.

tunkio, 2022-05-04:
Thanks for testing. That narrows search area a bit.

sf-206, 2022-05-04:
If I can test something else please let me know.

tunkio, 2022-05-04:
Okei, I think I found the problem. I'll try to fix it, well soon :), and make a pull request.

sf-206, 2022-05-04:
Thank you for the very fast fix. I have just added two remote MotionEye cameras which is working fine now.

But I am not able to change the configuration for the remote MotionEye camera. Changing any setting and Apply results in "An error occurred. Refreshing is recommended"

tunkio, 2022-05-04:
Ok need to check that then. This might take while, other things todo.

tunkio, 2022-05-04:
Ok, couple new files updated. handlers/base.py and remote.py. Update both installations if you want to test.

Error handling should be fixed now and authentication should work too. Huh it was pretty nasty to find that problem 😄

sf-206, 2022-05-04:
Unfortunately now motion eye is requesting user name and pw on any change. No matter if remote motion eye or network camera. The user name + pw is never accepted. I am sure it is correct.

tunkio, 2022-05-04:
Ok. Then i forget to take something into pr. I have two machine setup my self and it worked there. I'll check that tomorrow.

tunkio, 2022-05-05:
Ok took back couple things. If you can test it once more. Replace remote.py and handlers/config.py and if you didn't restore handlers/base.py yet, do that also.

sf-206, 2022-05-05:
GREAT it is working. Thank you so much!

PR #2464 (2022-05-12): Fixed remote motioneye camera config bug. (#2449)
Originally remote motioneye camera config generator caused camera loop break which resulted empty config list. Fixes #2449
closed 2022-05-12 by jmichault