motioneye-project/motioneye #2453: Exposure/Brightness with Raspberry Pi Camera Module V2.1

johncliffton, 2022-05-02:
Hi,
I am having an issue with motioneye and an Raspberry Pi Camera Module V2.1.
Exposure/shutter speed seems to be adjusted once motioneye (...or the camera module...) starts ONLY.
Meaning: When starting the camera in the morning with sunset, images will be completely blurred at noon. Or when starting the camera at noon, it will be completly dark at sunset.
So, shutterspeed/exposure will not be adjusted while Raspbian and motioneye are running. This would not matter if I had a static lightning environment - but since I am using it to observe the garden... sun and weather definately have an impact.

I tried using the automatic brightness option, which somehow will e.g. adjust down the brightness when blurred for a moment and then reset. However, I assume shutterspeed should rather be adjusted.
Hopefully this explanation makes sense. I am not an expert. Is that a known issue or am I missing a setting here?
Changing the camera would be difficult due to the ZeroView I am using for mounting it.

Setup:
motionEye Version | 0.42.1
Motion Version | 4.3.2
OS Version | Raspbian 11
Raspberry Pi Camera Module V2.1

Thank you in advance.

Kind regards

MichaIng, 2022-05-02:
motionEye automatically adjusts the brightness, but AFAIK only when the stream is watched on web UI, so it may be expected that this doesn't happen before. Not sure whether the camera has an automatic brightness adjustment hardware or driver wise?

However, we'd need to have a look into the code, where brightness adjustment is done by motionEye backend, and see whether we can do this as well for recordings without the web UI being watched.

It would be btw great if you could test the Python 3 `dev` version of motionEye: https://github.com/motioneye-project/motioneye/tree/dev#installation
There were quite a lot of changes, so it would be good to assure that this issue has not been solved already, intentionally or indirectly by other changes.

johncliffton, 2022-05-13:
> It would be btw great if you could test the Python 3 `dev` version of motionEye: https://github.com/motioneye-project/motioneye/tree/dev#installation There were quite a lot of changes, so it would [...]

I have tried that and it seems a lot better. Thanks for the hint.

JamBalaya56562, 2026-11-10:
Closing, as this was with motionEye 0.42.1 on the legacy camera stack, where Motion kept the exposure it started with, while `raspistill` adjusted it. Current Raspberry Pi OS runs the camera through libcamera instead; please open a new issue if exposure still does not follow the light with a current release.

closed 2026-11-10 by JamBalaya56562