motioneye-project/motioneye #2455: Help configuring MotionEye for bird watching needed

bad-friday, 2022-05-02:
I have a Raspberry Pi 4 with a Raspberry Pi camera V2.
I would appreciate help configuring MotionEye to watch birds.
So far I have not been able to get MotionEye to record anything except leaves moving.
Just now a blackbird passed in front of the camera at a distance of 2m but nothing was recorded.
I think screenshots of a working application might help.

zagrim, 2022-05-03:
Sounds like you need to adjust the parameters under "Motion Detection" to suit your needs. But if your purpose is to catch birds flying by and not to record moving leaves, that might be difficult to achieve. Unfortunately there are no such parameters or settings in Motion that would make the detection be able to ignore movement in the background while still picking up movement in the foreground (since it has no idea of "background" or "foreground"). The way Motion detects motion (very roughly):

1. It calculates the difference in colour values of the same pixel in two adjanced frames in the video, for all the pixels in the frame (in other words, the pixel difference of the frames), then 
2. applies some filters (Noise Level, Despeckle Filter) to do some tidying up of small scale movement and/or unifying clusters of small moving speckles into a larger one, and then 
3. looks if the total of pixel differences in the whole frame exceeds the set Frame Change Threshold (and if the difference is below Maximum Change Threshold if that is set). 

If the checks at step 3 pass, then it checks yet another group of parameters (Motion Gap, Minimum Motion Frames) for whether it should record that frame (and possibly also those that match Captured Before and Captured After).

So, if your scene is such that the birds you want to catch on video are flying past the moving leaves, and the leaves are shown large enough so that Noise Level setting can't clean that up, I think you'd need something that can make the difference between "a bird-like moving object" and some other movement - which is sadly over the capabilities of Motion, and also would require far more computing resources than a RPi can offer.

However, if there is at least some part of the scenery such that it is free from leaves (or other insignificant moving objects), and the birds tend to fly over that part of the frame, you might have some success by setting motion detection mask over those parts with a lot of moving leaves, keeping Minimum Motion Frames low, and setting Captured Before and Captured After so that even though motion was detected only in one area of the scene, the recording would span from the bird entering the view to bird existing the view (those are set in "frames" so you might start with values matching the FPS reading of your camera to get one second prior to detection and one second after end of detected motion).

Enabling Show Frame Changes and maybe also Create Debug Media Files should help you by visualising where motion gets detected.

zagrim, 2022-05-03:
Myself I've been struggling with moving leaves and branches in a bit different setting (I can mask out the trees and bushes, but not the shadows they cast) while trying to record human movement, and even that is hard to do without either having a large amount of non-interesting videos of dancing shadows on a windy and sunny day, or a significant amount of undetected human movement. As my priority is to have most the human movement without filling up storage, I've been playing with Smart Mask which effectively builds a dynamic motion detection mask based on detected movement. It works quite ok for avoiding videos of dancing shadows, but has the side-effect that if somebody walks on the scene on those areas where the shadows move, also that goes undetected.

Beside full-blown object detection and/or labelling of shapes (i.e. "a bird", "a person", "a tree") which is compute-intensive, there might be some ways to detect the difference between continuous back-and-forth motion motion (like the leaves) and more directional motion (like a bird flying by, or a person walking), but that would require more of temporal dimension than what Motion uses. It currently uses only the intra-frame information, and I think it would require at least a sequence of those spanning a second in time or so to be able to make the difference. I have no idea if the Motion developers have any future plans on implementing such enhanced algorithms, but since that would also increase the resource requirements it might be considered out of scope for that project. The very reason we can run Motion on RPis is that it has always doing the detection in a simple and efficient way. Unfortunately that way works well only in controlled (indoor) environments.

bad-friday, 2022-05-03:
It seems that finding the right settings is a little more complicated than I thought.

closed 2022-05-07 by bad-friday