motioneye-project/motioneye #2461: Timelapse creating problem

ThomasTom86, 2022-05-03:
Hello!  I get the following error message when creating a timelapse video!

best regards 
[image]
[image]
[image]

MichaIng, 2022-05-03:
When did you install motionEye? We merged a fix for creating timelapse videos from pictures just recently: #2437

ThomasTom86, 2022-05-04:
about a week ago.

MichaIng, 2022-05-04:
The commit was merged 6 days ago. Please try to update motionEye via linked instructions.

ThomasTom86, 2022-05-15:
i have updated to the last release 0.43.0 but creating the timelapse still doesn't work.

MichaIng, 2022-05-15:
You followed the upgrade instructions, especially `--force-reinstall`? Also did you `systemctl restart motioneye` afterwards? If so, please post the service logs again:
```sh
journalctl -u motioneye
```

ThomasTom86, 2022-05-16:
yes I did the update the same way

this is the log…
[image]

ThomasTom86, 2022-05-16:
[image]
[image]

ThomasTom86, 2022-05-16:
yes I did the update the same way

this is the log ….
[image]
[image]
[image]

JamBalaya56562, 2026-11-10:
Closing, as this got no answer to the distribution question. The timelapse fix of #2437 is in every release since 0.43.1, and the later "rtsp not supported in libcurl" log came from a libcurl built without RTSP. Please open a new issue with `journalctl -u motioneye` if a timelapse still fails with a current release.

closed 2026-11-10 by JamBalaya56562