motioneye-project/motioneye #2465: Can't install Motioneye on Alpine linux

klewf, 2022-05-04:
Can't install on Alpine.
Step by step instructions required.

MichaIng, 2022-05-04:
I guess the major question is how to install Python 3 on Alpine Linux. The installing pip and motionEye should be straight forward: https://github.com/motioneye-project/motioneye/tree/dev#installation

cclauss, 2022-05-06:
I believe it is time to make a new [release](https://github.com/motioneye-project/motioneye/releases) so that our default branch no longer points to a Python that no one uses anymore.

MichaIng, 2022-05-07:
As fast as the most urgent issues are addressed: https://github.com/motioneye-project/motioneye/milestone/1
The one or the other could be also skipped for this release, but there are still some syntax/type errors due to Python 2 > Python 3 migration, missing motion 4.4 support, and one or two older but obvious issues in the Python code, which we should address first. Then I suggest to do a beta, including upload to PyPI and DockerHub as pre-release, and based on feedback a stable release ASAP. But to be discussed here: #2373

JamBalaya56562, 2026-11-06:
These steps install motionEye on Alpine `edge`, checked with 0.44.0, Python 3.14 and motion 4.7.1 in an x86_64 container and in an aarch64 chroot on a Raspberry Pi 5: the server starts and `motion` runs.

```sh
apk add python3 python3-dev py3-pip gcc musl-dev curl curl-dev ffmpeg v4l-utils
apk add --repository https://dl-cdn.alpinelinux.org/alpine/edge/testing motion
python3 -m venv /opt/motioneye
/opt/motioneye/bin/pip install motioneye
```

On Alpine 3.24 motionEye installs and starts as well, but `motion` cannot be installed from these repositories, so no camera can be added. `motioneye_init` assumes APT or RPM, so the service has to be set up by hand, see [`motioneye/extra`](https://github.com/motioneye-project/motioneye/tree/dev/motioneye/extra). With 0.44.0 on Python 3.14 the media cleanup fails with `failed to list config dir`, which is fixed for the next release.

A picture from a real camera was not confirmed: in the chroot, Alpine's libcamera 0.7.2 did not find a Camera Module 3 under the Raspberry Pi OS kernel, and network streams from that camera gave no picture. The service was not checked either. Closing as completed, as this was about the install; please reopen if these steps fail.

closed 2026-11-06 by JamBalaya56562