motioneye-project/motioneye #2466: FTP Upload not working

Spankythemusical, 2022-05-04:
I deleted the python2 branch installed and installed the dev branch.  The videos are showing (minor log errors, but nothing that is a show stopper), streaming works, and the "Test Service" for ftp uploads is successful.  The videos, however, are never uploaded.  According to `journalctl -u motioneye`, this is the error.

`May 04 19:46:05 ubuntuvm meyectl[100734]:    ERROR: failed to upload file "/var/lib/motioneye/Camera2/2022-05-04/19-35-53.mp4" with service ftp: initial_value must be str or None, not bytes
May 04 19:46:05 ubuntuvm meyectl[100734]: Traceback (most recent call last):
May 04 19:46:05 ubuntuvm meyectl[100734]:   File "/usr/local/lib/python3.10/dist-packages/motioneye/uploadservices.py", line 1309, in upload_media_file
May 04 19:46:05 ubuntuvm meyectl[100734]:     service.upload_file(target_dir, filename, camera_name)
May 04 19:46:05 ubuntuvm meyectl[100734]:   File "/usr/local/lib/python3.10/dist-packages/motioneye/uploadservices.py", line 108, in upload_file
May 04 19:46:05 ubuntuvm meyectl[100734]:     self.upload_data(rel_filename, mime_type, data, ctime, camera_name)
May 04 19:46:05 ubuntuvm meyectl[100734]:   File "/usr/local/lib/python3.10/dist-packages/motioneye/uploadservices.py", line 1007, in upload_data
May 04 19:46:05 ubuntuvm meyectl[100734]:     conn.storbinary(f'STOR {filename}', io.StringIO(data))
May 04 19:46:05 ubuntuvm meyectl[100734]: TypeError: initial_value must be str or None, not bytes
May 04 19:46:10 ubuntuvm meyectl[100734]:    ERROR: failed to upload file "/var/lib/motioneye/Camera1/2022-05-04/19-35-55.mp4" with service ftp: '>' not supported between instances of 'NoneType' and 'int'
May 04 19:46:10 ubuntuvm meyectl[100734]: Traceback (most recent call last):
May 04 19:46:10 ubuntuvm meyectl[100734]:   File "/usr/local/lib/python3.10/dist-packages/motioneye/uploadservices.py", line 1309, in upload_media_file
May 04 19:46:10 ubuntuvm meyectl[100734]:     service.upload_file(target_dir, filename, camera_name)
May 04 19:46:10 ubuntuvm meyectl[100734]:   File "/usr/local/lib/python3.10/dist-packages/motioneye/uploadservices.py", line 108, in upload_file
May 04 19:46:10 ubuntuvm meyectl[100734]:     self.upload_data(rel_filename, mime_type, data, ctime, camera_name)
May 04 19:46:10 ubuntuvm meyectl[100734]:   File "/usr/local/lib/python3.10/dist-packages/motioneye/uploadservices.py", line 1002, in upload_data
May 04 19:46:10 ubuntuvm meyectl[100734]:     conn = self._get_conn()
May 04 19:46:10 ubuntuvm meyectl[100734]:   File "/usr/local/lib/python3.10/dist-packages/motioneye/uploadservices.py", line 1042, in _get_conn
May 04 19:46:10 ubuntuvm meyectl[100734]:     self._conn.connect(self._server, port=self._port)
May 04 19:46:10 ubuntuvm meyectl[100734]:   File "/usr/lib/python3.10/ftplib.py", line 149, in connect
May 04 19:46:10 ubuntuvm meyectl[100734]:     if port > 0:
May 04 19:46:10 ubuntuvm meyectl[100734]: TypeError: '>' not supported between instances of 'NoneType' and 'int'`

The ftp upload did work with the previous install.

MichaIng, 2022-05-04:
Indeed, there are missing str => bytes conversions, and when doing so, a missing check whether the body is actually `None`. See the issue here: #2408

There is another partly fix: #2456
But we need to do this in a consistent and efficient way. I hope I find time tomorrow to do this.

Spankythemusical, 2022-05-04:
Thank you for the response.  I was going through other listed issues last night, as I had an error that was different to the current but still having to do with the str/bytes issue.  I found two changes to the mediafiles.py file, which I incorporated.  The port > 0 has occurred since.  I'll keep checking for updates.  Thank you for your hard work!

closed 2022-05-04 by Spankythemusical