motioneye-project/motioneye #2473: [FEATURE REQUEST] Support Proxy Authentication

hobbyquaker, 2022-05-08:
I'd like to have a configuration option that allows me to set a HTTP Header that can used for doing external proxy authentication with e.g. Authentik. 

As an example see PAPERLESS_ENABLE_HTTP_REMOTE_USER and PAPERLESS_HTTP_REMOTE_USER_HEADER_NAME in Paperless docs: https://paperless-ngx.readthedocs.io/en/latest/configuration.html#hosting-security

Motioneye would just have to accept the presence of this header as a valid session with the supplied user.

MichaIng, 2022-05-08:
Many thanks for your request.

I see the idea behind it, but how far I can think about it, it seems to be quite complicate to implement since motionEye supports an admin and a surveillance user (multiple ones are requested already) where different GUI elements are shown. And how should motionEye know which header is for authentication, i.e. when to skip the internal authentication, which header is for username, whether these usernames match the ones configured in motionEye, to know which GUI elements to show, and whether the header is just set by an attacker without any actual authentication being done: So motionEye would disable it's authentication for everyone who knows that just this header needs to be set.

At least it must be a dedicated option to explicitly disable motionEye's authentication, not done based on any header. But still unsure how to deal with user names and user-dependant GUI elements: It must be assured that a person behind the surveillance user cannot gain admin GUI access. Basically I see no chance besides a motionEye setting which allow to set respective username headers explicitly as trusted header, with the risk on user end, like `Satisfy Any` in Apache.

hobbyquaker, 2022-05-08:
> seems to be quite complicate to implement since motionEye supports an admin and a surveillance user (multiple ones are requested already) where different GUI elements are shown.

Authentik (and I guess other authentication tools too) can also supply group-memberships via additional headers. 

> So motionEye would disable it's authentication for everyone who knows that just this header needs to be set.

Yes, I think it would be sufficient just to point out that the Header-auth should only be enabled if a reverse proxy handles access. From Paperless docs as example:

```
This will allow authentication by simply adding a Remote-User: <username> header to a request. Use with care! You especially must ensure that any such header is not passed from your proxy server to paperless.

If you’re exposing paperless to the internet directly, do not use this.
```

> At least it must be a dedicated option to explicitly disable motionEye's authentication, not done based on any header

Sure, Motioneye should not just accept this header per se - only if it's enabled by a config option.

MichaIng, 2022-05-09:
Ah so we could have a look into how OctoPrint implemented it to not reinvent it from scratch, that is good.

hobbyquaker, 2022-05-10:
Yes, see https://github.com/OctoPrint/OctoPrint/blob/master/src/octoprint/server/util/__init__.py#L231