motioneye-project/motioneye #2475: FTP upload in Python 3 dev not working

johncliffton, 2022-05-09:
Hi,
I am trying the Python 3 dev version because of [this ](https://github.com/motioneye-project/motioneye/issues/2453)issue that I am having.
This seems to be having issues with FTP uploads and will keep all files on local storage.

Error:
May  9 22:04:49 raspberrypi meyectl[661]:    ERROR: failed to upload file "/var/lib/motioneye/Camera1/2022-05-09/22-04-42.jpg" with service ftp: initial_value must be str or None, not bytes
May  9 22:04:49 raspberrypi meyectl[661]: Traceback (most recent call last):
May  9 22:04:49 raspberrypi meyectl[661]:   File "/usr/local/lib/python3.9/dist-packages/motioneye/uploadservices.py", line 1309, in upload_media_file
May  9 22:04:49 raspberrypi meyectl[661]:     service.upload_file(target_dir, filename, camera_name)
May  9 22:04:49 raspberrypi meyectl[661]:   File "/usr/local/lib/python3.9/dist-packages/motioneye/uploadservices.py", line 108, in upload_file
May  9 22:04:49 raspberrypi meyectl[661]:     self.upload_data(rel_filename, mime_type, data, ctime, camera_name)
May  9 22:04:49 raspberrypi meyectl[661]:   File "/usr/local/lib/python3.9/dist-packages/motioneye/uploadservices.py", line 1007, in upload_data
May  9 22:04:49 raspberrypi meyectl[661]:     conn.storbinary(f'STOR {filename}', io.StringIO(data))
May  9 22:04:49 raspberrypi meyectl[661]: TypeError: initial_value must be str or None, not bytes
May  9 22:05:06 raspberrypi meyectl[661]:    ERROR: failed to upload file "/var/lib/motioneye/Camera1/2022-05-09/22-05-00.jpg" with service ftp: initial_value must be str or None, not bytes
May  9 22:05:06 raspberrypi meyectl[661]: Traceback (most recent call last):
May  9 22:05:06 raspberrypi meyectl[661]:   File "/usr/local/lib/python3.9/dist-packages/motioneye/uploadservices.py", line 1309, in upload_media_file
May  9 22:05:06 raspberrypi meyectl[661]:     service.upload_file(target_dir, filename, camera_name)
May  9 22:05:06 raspberrypi meyectl[661]:   File "/usr/local/lib/python3.9/dist-packages/motioneye/uploadservices.py", line 108, in upload_file
May  9 22:05:06 raspberrypi meyectl[661]:     self.upload_data(rel_filename, mime_type, data, ctime, camera_name)
May  9 22:05:06 raspberrypi meyectl[661]:   File "/usr/local/lib/python3.9/dist-packages/motioneye/uploadservices.py", line 1007, in upload_data
May  9 22:05:06 raspberrypi meyectl[661]:     conn.storbinary(f'STOR {filename}', io.StringIO(data))
May  9 22:05:06 raspberrypi meyectl[661]: TypeError: initial_value must be str or None, not bytes

MichaIng, 2022-05-10:
Yeah there are still some str-bytes conversions missing. I should find time until end of the week to fix this. There it's an open PR which starts with fixes: https://github.com/motioneye-project/motioneye/pull/2456

jmichault, 2022-05-13:
@johncliffton , can you try the 2475-ftp-upload-in-python-3-dev-not-working ?

johncliffton, 2022-05-13:
@jmichault , happy to do that but a complete noob in what I have to do.
I tried
`sudo systemctl stop motioneye`
`sudo python3 -m pip install --upgrade --force-reinstall --no-deps 'https://github.com/motioneye-project/motioneye/archive/dev.tar.gz' `
`sudo systemctl start motioneye`
but the problem persists. Mostly sure I did something wrong. Please kindly point me into the right direction.

jmichault, 2022-05-13:
Just replace dev with 2475-ftp-upload-in-python-3-dev-not-working : 

```
sudo systemctl stop motioneye
sudo python3 -m pip install --upgrade --force-reinstall --no-deps 'https://github.com/motioneye-project/motioneye/archive/2475-ftp-upload-in-python-3-dev-not-working.tar.gz'
sudo systemctl start motioneye
```

johncliffton, 2022-05-14:
@jmichault sorry, was not aware that it's that easy.
Tried 2475-ftp-upload-in-python-3-dev-not-working and can confirm it is working. Thank you.
I also re-tried [#2453](https://github.com/motioneye-project/motioneye/issues/2453) and sadly this persists.
I assume this cannot be tied to this particular devpart?

MichaIng, 2022-05-14:
Great, I'll mark this as closed then.

closed 2022-05-14 by MichaIng