motioneye-project/motioneye #2476: Unable to Open Video Device after Resolution Change

fictionalname, 2022-05-11:
Hello folks,

I hope you can help with this strange issue.

Hardware : Pi Zero W with a wide angle IR camera with 2 IR lights bolted to the side
Software : DietPi, then MotionEye using their automated process.

Issue : 
- Logged into the MotionEye web interface
- Added the MMAL camera
- Camera worked perfectly first time
- Upped the resolution to 1024x768
- The interface hung & I was unable to Putty onto the Pi Zero W
- Rebooted the Pi Zero W
- Logged back into the MotionEye web interface 
- Video window was showing a grey screen saying “Unable to open video device”
- I tried reducing the resolution back to 640x48 (and even options below) but it still shows the grey screen
- Tried rebooting again – still the same

Please tell me how I can revive the connection to the camera.

If I’ve missed out any important info, please let me know.

Thanks muchly.

MichaIng, 2022-05-11:
Does the service log show anything related?
```sh
journalctl -u motioneye
```

fictionalname, 2022-05-11:
Hi Michalng,

Thanks for taking the time to respond!

This is so confusing for me.  I had the camera working very breifly again, but now it's not even opening the device, it's just the camera with a line through it.  I'm guessing this is cause Motion is't running any more.  I ran the command you suggested and I'm getting this : 

-- Journal begins at Wed 2022-05-11 21:11:59 BST, ends at Wed 2022-05-11 21:14:31 BST. --
May 11 21:12:18 DietPi systemd[1]: Started motionEye Server.
May 11 21:12:42 DietPi meyectl[396]: configure_logging cmd motioneye: False
May 11 21:12:42 DietPi meyectl[396]: configure logging to file: None
May 11 21:12:42 DietPi meyectl[396]:     INFO: hello! this is motionEye server 0.43.0
May 11 21:13:31 DietPi meyectl[396]:     INFO: cleanup started
May 11 21:13:31 DietPi meyectl[396]:     INFO: wsswitch started
May 11 21:13:32 DietPi meyectl[396]:     INFO: tasks started
May 11 21:13:32 DietPi meyectl[396]:     INFO: mjpg customer garbage collector has started
May 11 21:13:33 DietPi meyectl[396]:     INFO: server started
May 11 21:13:40 DietPi motion[703]: [0:motion] [NTC] [ALL] conf_load: Processing thread 0 - config file /etc/motioneye/motion.conf
May 11 21:13:40 DietPi motion[703]: [0:motion] [NTC] [ALL] config_camera: Processing camera config file camera-1.conf
May 11 21:13:40 DietPi motion[703]: [0:motion] [NTC] [ALL] motion_startup: Logging to syslog
May 11 21:13:40 DietPi motion[703]: [0:motion] [NTC] [ALL] motion_startup: Motion 4.3.2 Started
May 11 21:13:40 DietPi motion[703]: [0:motion] [NTC] [ALL] motion_startup: Using default log type (ALL)
May 11 21:13:40 DietPi motion[703]: [0:motion] [NTC] [ALL] motion_startup: Using log type (ALL) log level (WRN)
May 11 21:14:11 DietPi motion[703]: [0:motion] [ERR] [ALL] motion_watchdog: Thread 1 - Watchdog timeout. Trying to do a graceful restart
May 11 21:14:21 DietPi motion[703]: [0:motion] [ERR] [ALL] motion_watchdog: Thread 1 - Watchdog timeout did NOT restart, killing it!
May 11 21:14:24 DietPi meyectl[396]:    ERROR: motion not running, starting it
May 11 21:14:31 DietPi motion[925]: [0:motion] [NTC] [ALL] conf_load: Processing thread 0 - config file /etc/motioneye/motion.conf
May 11 21:14:31 DietPi motion[925]: [0:motion] [NTC] [ALL] config_camera: Processing camera config file camera-1.conf
May 11 21:14:31 DietPi motion[925]: [0:motion] [NTC] [ALL] motion_startup: Logging to syslog
May 11 21:14:31 DietPi motion[925]: [0:motion] [NTC] [ALL] motion_startup: Motion 4.3.2 Started
May 11 21:14:31 DietPi motion[925]: [0:motion] [NTC] [ALL] motion_startup: Using default log type (ALL)
May 11 21:14:31 DietPi motion[925]: [0:motion] [NTC] [ALL] motion_startup: Using log type (ALL) log level (WRN)

Clearly it has been running but it's failing to launch now properly now.  How can I find out why it is crashing?

I did notice that the GPU/RAM setting had gone back to 96 so I set it back to 128 again, but that hasn't helped.

I'm not sure if it is connected as I'm just grasping at straws here, but I changed the I2C state and SPI state and it seemed to work breifly but I don't really know what they do, and it may have just been the subsequent reboot which kicked it into life.

I have read that there may be some issues with Bullseye and the Pi cameras.  Could this be causing the issue?  If so, why does it seem to work very briefly then totally stop.

Any suggestions would be greatly appreciated as I've already flattened the SD and reinstalled DietPi to get to this stage, trying to get this to work.

Thanks!

MichaIng, 2022-05-14:
I2C and SPI are not required for running cameras on the RPi camera connector, AFAIK also 128 MiB RAM are probably reasonable but not strictly required.

Just to assure the camera with legacy stack is enabled correctly, and let's check for kernel errors as well:
```sh
vcgencmd get_camera
dmesg -l emerg,alert,crit,err
```

fictionalname, 2022-05-15:
Morning,

Thanks for your next suggestion.  Here is what I got from the 2 commands:
root@DietPi:~# vcgencmd get_camera
supported=1 detected=1, libcamera interfaces=0
root@DietPi:~# dmesg -l emerg,alert,crit,err
root@DietPi:~#

Supported & detected for the camera and not response at all from the dmseg command which I guess means no error messages?

Thanks.

fictionalname, 2022-05-15:
Further info:

I tried uninstalling MotionEye, reboot the reinstall and it is still failing to show the camera.

I noted as part of the install there was a permissions error writing to the /etc/motion folder so as a test I added the root user to the montion group (no change).   It was in the install script so I didn't manage to read any other info around that error.

I also noted that the /etc/motioneye folder was the only one in the etc folder not owned by root (it was motion).  Again as a test, I changed the owner to root.  This had not effect, even after a reboot.

I also read on-line there may be issues with the motion detection so I switched it off in the GUI.  I'm not getting additional erros in the log files:

```
May 15 22:08:55 DietPi motion[1938]: [0:motion] [NTC] [ALL] conf_load: Processing thread 0 - config file /etc/motioneye/motion.conf
May 15 22:08:55 DietPi motion[1938]: [0:motion] [NTC] [ALL] config_camera: Processing camera config file camera-1.conf
May 15 22:08:55 DietPi motion[1938]: [0:motion] [NTC] [ALL] motion_startup: Logging to syslog
May 15 22:08:55 DietPi motion[1938]: [0:motion] [NTC] [ALL] motion_startup: Motion 4.3.2 Started
May 15 22:08:55 DietPi motion[1938]: [0:motion] [NTC] [ALL] motion_startup: Using default log type (ALL)
May 15 22:08:55 DietPi motion[1938]: [0:motion] [NTC] [ALL] motion_startup: Using log type (ALL) log level (WRN)
May 15 22:09:25 DietPi motion[1938]: [0:motion] [ERR] [ALL] motion_watchdog: Thread 1 - Watchdog timeout. Trying to do a graceful restart
May 15 22:09:35 DietPi motion[1938]: [0:motion] [ERR] [ALL] motion_watchdog: Thread 1 - Watchdog timeout did NOT restart, killing it!
May 15 22:09:44 DietPi meyectl[396]:    ERROR: motion not running, starting it
May 15 22:09:46 DietPi meyectl[396]:    ERROR: Exception in callback functools.partial(<bound method IOLoop._discard_future_result of <tornado.platform.asyncio.AsyncIOMainLoop object at 0xb3e01178>>, <Task finished name='Task-1268' coro=<_disable_initial_motion_detection() done, defined at /usr/local/lib/python3.9/dist-packages/motioneye/motionctl.py:528> exception=HTTP 599: Failed to connect to 127.0.0.1 port 7999: Connection refused>)
May 15 22:09:46 DietPi meyectl[396]: Traceback (most recent call last):
May 15 22:09:46 DietPi meyectl[396]:   File "/usr/local/lib/python3.9/dist-packages/tornado/ioloop.py", line 741, in _run_callback
May 15 22:09:46 DietPi meyectl[396]:     ret = callback()
May 15 22:09:46 DietPi meyectl[396]:   File "/usr/local/lib/python3.9/dist-packages/tornado/ioloop.py", line 765, in _discard_future_result
May 15 22:09:46 DietPi meyectl[396]:     future.result()
May 15 22:09:46 DietPi meyectl[396]:   File "/usr/local/lib/python3.9/dist-packages/motioneye/motionctl.py", line 540, in _disable_initial_motion_detection
May 15 22:09:46 DietPi meyectl[396]:     await set_motion_detection(camera_id, False)
May 15 22:09:46 DietPi meyectl[396]:   File "/usr/local/lib/python3.9/dist-packages/motioneye/motionctl.py", line 272, in set_motion_detection
May 15 22:09:46 DietPi meyectl[396]:     resp = await AsyncHTTPClient().fetch(request)
May 15 22:09:46 DietPi meyectl[396]: tornado.curl_httpclient.CurlError: HTTP 599: Failed to connect to 127.0.0.1 port 7999: Connection refused
```

Please let me know if you need any more information.

Thanks muchly.

MichaIng, 2022-05-19:
The ` /etc/motion` permissions error is part of the `motion` package install step, hence outside of our control, but also not relevant for motionEye: We use own motion configs in `/etc/motioneye` instead, where permissions are granted correctly.

Is btw the `motion` service running? AFAIK it could conflct:
```sh
systemctl disable --now motion
ssystemctl restart motioneye
```

fictionalname, 2022-05-22:
Hi again,

Thanks for the further suggestion.

I tried running those commands to disable motion but it had no effect.

I have just erased the SD card did the following:
- Clean install of the newest version of DietPi
- Went throught the automated startup wizard
- Installed MotionEye from the menu system
- No other software or config changes

The hardware is a Pi Zero with no additional hardware, other than the IR camera with 2 external IR lights.

After I did this, the MMAL camera was detected for a small amount of time and showed the picture in MotionEye as I would expect.

I made no further changes, and the image from the camera froze and would not update.

The motioneye log looks like this:

'''
-- Journal begins at Sun 2022-05-22 18:07:08 BST, ends at Sun 2022-05-22 18:21:37 BST. --
May 22 18:08:06 DietPi systemd[1]: Started motionEye Server.
May 22 18:08:19 DietPi meyectl[383]: configure_logging cmd motioneye: False
May 22 18:08:19 DietPi meyectl[383]: configure logging to file: None
May 22 18:08:19 DietPi meyectl[383]:     INFO: hello! this is motionEye server 0.43.0
May 22 18:16:51 DietPi meyectl[383]:     INFO: cleanup started
May 22 18:16:51 DietPi meyectl[383]:     INFO: wsswitch started
May 22 18:16:51 DietPi meyectl[383]:     INFO: tasks started
May 22 18:16:51 DietPi meyectl[383]:     INFO: mjpg customer garbage collector has started
May 22 18:16:52 DietPi meyectl[383]:     INFO: server started
May 22 18:17:47 DietPi meyectl[383]:     INFO: adding new mmal camera with id 1...
May 22 18:17:47 DietPi meyectl[383]:     INFO: main config file /etc/motioneye/motion.conf does not exist, using default values
May 22 18:17:55 DietPi motion[629]: [0:motion] [NTC] [ALL] conf_load: Processing thread 0 - config file /etc/motioneye/motion.conf
May 22 18:17:55 DietPi motion[629]: [0:motion] [NTC] [ALL] config_camera: Processing camera config file camera-1.conf
May 22 18:17:55 DietPi motion[629]: [0:motion] [NTC] [ALL] motion_startup: Logging to syslog
May 22 18:17:55 DietPi motion[629]: [0:motion] [NTC] [ALL] motion_startup: Motion 4.3.2 Started
May 22 18:17:55 DietPi motion[629]: [0:motion] [NTC] [ALL] motion_startup: Using default log type (ALL)
May 22 18:17:55 DietPi motion[629]: [0:motion] [NTC] [ALL] motion_startup: Using log type (ALL) log level (WRN)
May 22 18:18:45 DietPi motion[629]: [0:motion] [ERR] [ALL] motion_watchdog: Thread 1 - Watchdog timeout. Trying to do a graceful restart
May 22 18:18:55 DietPi motion[629]: [0:motion] [ERR] [ALL] motion_watchdog: Thread 1 - Watchdog timeout did NOT restart, killing it!
May 22 18:19:07 DietPi meyectl[383]:    ERROR: mjpg client timed out receiving data for camera 1 on port 9081
May 22 18:19:17 DietPi meyectl[383]:    ERROR: mjpg client timed out receiving data for camera 1 on port 9081
May 22 18:19:27 DietPi meyectl[383]:    ERROR: mjpg client timed out receiving data for camera 1 on port 9081
May 22 18:19:37 DietPi meyectl[383]:    ERROR: mjpg client timed out receiving data for camera 1 on port 9081
May 22 18:19:47 DietPi meyectl[383]:    ERROR: mjpg client timed out receiving data for camera 1 on port 9081
'''

The window for the camera image is showing a distorted icon of a camera with a line through it.

Is this just my setup or are other people having this issue?

Thanks for your help!

MichaIng, 2022-05-22:
It always triggers the watchdog timeout, then fails to terminate, then is SIGKILLed, then shows `VIDIOC_S_FMT: : Device or resource busy`. Is the old motion process probably still running, blocking the video device? Check in e.g. `htop`.

fictionalname, 2022-05-22:
It seems that the only version of motion running is the one associated with MotionEye:

[image]

MichaIng, 2022-05-22:
Hmm, no `motion` process is running at all 🤔.

There other error is `v4l2_pixfmt_select: Palette selection failed for format UYVY`, probably a motion setting is invalid (in combination with this camera). Is there a color/pixel/palette format option which you can change?

fictionalname, 2022-05-22:
Hmmm, how strange.  I can't figure this out at all!  It's just a base build of DietPi with MotionEye on top.

It's just the standard Pi IR camera like this : https://shop.pimoroni.com/products/night-vision-camera-module-for-raspberry-pi?variant=12516582817875

Where would I change these settings?

MichaIng, 2022-05-22:
I mean in the motionEye camera settings.

don-dolarson, 2023-06-11:
I'd like to share with my findings as I was fighting 3 days to get things cooperate and finally got it to work.
My problem was a grey preview window with "unable to open device" text, no matter what I did, until now.

Raspberry Pi Model B Rev 2, Raspberry OS Buster 2023-05-03, AZ-Delivery 5mp OV5647 ribbon camera.
https://github.com/motioneye-project/motioneye/tree/dev#installation

motionEye Version | 0.43.0
Motion Version | 4.5.1
OS Version | Raspbian 10

Whenever system starts, running services are:
[image]

It results in:
```
dondolarson@MotionEyeOS:~ $ journalctl -u motioneye -f
Jun 10 19:14:02 MotionEyeOS meyectl[1652]:     INFO: adding new mmal camera with id 1...
Jun 10 19:14:08 MotionEyeOS motion[2398]: [0:motion] [NTC] [ALL] conf_load: Processing thread 0 - config file /etc/motioneye/motion.conf
Jun 10 19:14:08 MotionEyeOS motion[2398]: [0:motion] [NTC] [ALL] config_camera: Processing camera config file camera-1.conf
Jun 10 19:14:08 MotionEyeOS motion[2398]: [0:motion] [NTC] [ALL] motion_startup: Logging to syslog
Jun 10 19:14:08 MotionEyeOS motion[2398]: [0:motion] [NTC] [ALL] motion_startup: Motion 4.5.1 Started
Jun 10 19:14:08 MotionEyeOS motion[2398]: [0:motion] [NTC] [ALL] motion_startup: Using default log type (ALL)
Jun 10 19:14:08 MotionEyeOS motion[2398]: [0:motion] [NTC] [ALL] motion_startup: Using log type (ALL) log level (WRN)
Jun 10 19:14:08 MotionEyeOS motion[2398]: [1:ml1:Camera1] [ERR] [VID] v4l2_pixfmt_set: Error setting pixel format.: Device or resource busy
Jun 10 19:14:08 MotionEyeOS motion[2398]: [1:ml1:Camera1] [ERR] [VID] Above message repeats 1 times
Jun 10 19:14:08 MotionEyeOS motion[2398]: [1:ml1:Camera1] [ERR] [VID] v4l2_pixfmt_select: Palette selection failed for format UYVY
Jun 10 19:14:08 MotionEyeOS motion[2398]: [1:ml1:Camera1] [ERR] [VID] v4l2_pixfmt_select: Unable to find a compatible palette format.
Jun 10 19:14:08 MotionEyeOS motion[2398]: [1:ml1:Camera1] [ERR] [VID] vid_start: V4L2 device failed to open
Jun 10 19:14:08 MotionEyeOS motion[2398]: [1:ml1:Camera1] [WRN] [ALL] motion_init: Could not fetch initial image from camera
Jun 10 19:14:08 MotionEyeOS motion[2398]: [1:ml1:Camera1] [WRN] [ALL] motion_init: Motion continues using width and height from config file(s)
Jun 10 19:14:10 MotionEyeOS motion[2398]: [1:ml1:Camera1] [WRN] [ALL] mlp_retry: Retrying until successful connection with camera
Jun 10 19:14:10 MotionEyeOS motion[2398]: [1:ml1:Camera1] [ERR] [VID] v4l2_pixfmt_set: Error setting pixel format.: Device or resource busy
Jun 10 19:14:10 MotionEyeOS motion[2398]: [1:ml1:Camera1] [ERR] [VID] Above message repeats 1 times
Jun 10 19:14:10 MotionEyeOS motion[2398]: [1:ml1:Camera1] [ERR] [VID] v4l2_pixfmt_select: Palette selection failed for format UYVY
Jun 10 19:14:10 MotionEyeOS motion[2398]: [1:ml1:Camera1] [ERR] [VID] v4l2_pixfmt_select: Unable to find a compatible palette format.
Jun 10 19:14:10 MotionEyeOS motion[2398]: [1:ml1:Camera1] [ERR] [VID] vid_start: V4L2 device failed to open
```

After running:
```
sudo service motioneye stop
sudo service motion stop
sudo service motioneye start
```
Running services are:

[image]

Resulting in:
```
dondolarson@MotionEyeOS:~ $ journalctl -u motioneye -f
-- Logs begin at Sat 2023-06-10 18:20:17 CEST. --
Jun 10 19:29:48 MotionEyeOS motion[2636]: [0:motion] [NTC] [ALL] motion_startup: Logging to syslog
Jun 10 19:29:48 MotionEyeOS motion[2636]: [0:motion] [NTC] [ALL] motion_startup: Motion 4.5.1 Started
Jun 10 19:29:48 MotionEyeOS motion[2636]: [0:motion] [NTC] [ALL] motion_startup: Using default log type (ALL)
Jun 10 19:29:48 MotionEyeOS motion[2636]: [0:motion] [NTC] [ALL] motion_startup: Using log type (ALL) log level (WRN)
Jun 10 19:30:47 MotionEyeOS motion[2669]: [0:motion] [NTC] [ALL] conf_load: Processing thread 0 - config file /etc/motioneye/motion.conf
Jun 10 19:30:47 MotionEyeOS motion[2669]: [0:motion] [NTC] [ALL] config_camera: Processing camera config file camera-1.conf
Jun 10 19:30:47 MotionEyeOS motion[2669]: [0:motion] [NTC] [ALL] motion_startup: Logging to syslog
Jun 10 19:30:47 MotionEyeOS motion[2669]: [0:motion] [NTC] [ALL] motion_startup: Motion 4.5.1 Started
Jun 10 19:30:47 MotionEyeOS motion[2669]: [0:motion] [NTC] [ALL] motion_startup: Using default log type (ALL)
Jun 10 19:30:47 MotionEyeOS motion[2669]: [0:motion] [NTC] [ALL] motion_startup: Using log type (ALL) log level (WRN)
```

```
dondolarson@MotionEyeOS:~ $ sudo service motion status
● motion.service - Motion - Security camera monitoring software.
   Loaded: loaded (/lib/systemd/system/motion.service; enabled; vendor preset: enabled)
   Active: inactive (dead) since Sat 2023-06-10 19:39:35 CEST; 59min ago
     Docs: man:motion(1)
[... 10 lines cut ...]
   Active: active (running) since Sat 2023-06-10 19:40:09 CEST; 59min ago
 Main PID: 704 (meyectl)
    Tasks: 10 (limit: 725)
   CGroup: /system.slice/motioneye.service
           ├─ 704 /usr/bin/python3 /usr/local/bin/meyectl startserver -c /etc/motioneye/motioneye.conf
           ├─ 714 /usr/bin/python3 /usr/local/bin/meyectl startserver -c /etc/motioneye/motioneye.conf
           └─2286 /usr/bin/motion -n -c /etc/motioneye/motion.conf -d 5
```

Raspberry Pi Model B Rev 2 (same CPU as Zero W) is too weak for this task however. Updating repos and upgrading/installing packages takes forever, CPU is constantly at 100% usage when not recording (!) but preview video is set at a reasonable 24fps. When recording it 99% of time crashes with:

```
[... cut ...] add mmal videocore camera, stop motion service and voila.
I've tested the most important things of motion and motioneye, and that is motion and recording/taking pictures. So far no problem. 

Things doesn't work after a restart since motion service starts automatically so I believe cron at bootup to stop motion needs in action here. I've couple of weeks linux experience so my knowledge is limited so tell me guys, is there a better option than running cron at every bootup and stop motion service?

FlorianCassayre, 2025-05-09:
Indeed, stopping `motion` seems to have done the trick. Thanks for pointing that out.

JamBalaya56562, 2026-11-12:
As found above, this happens when a separate `motion` service, e.g. from the Debian `motion` package, is running and holds the camera; motionEye starts its own Motion. `sudo systemctl disable --now motion` stops it for good. Closing as answered.

closed 2026-11-12 by JamBalaya56562