motioneye-project/motioneye #2477: Video streaming authentication not working

lukasig, 2022-05-13:
for dev version:
motionEye Version | 0.43.0
Motion Version | 4.3.2
OS Version | Ubuntu 22.04

When I activate "Video Streaming" and choose Authentication mode "Basic", the browser keeps asking for username and password. I have tried using both admin and surveillance username/passwords with no luck.

Also when active, the image in the web interface for that camera does not show anymore.

tunkio, 2022-05-13:
Both of these are infected by invalid str and byte handling. 

https://github.com/motioneye-project/motioneye/blob/11c0f06abd7093fa6e4734e7801e158fb6680c6e/motioneye/utils/__init__.py#L281-L282

One way to fix:
```
def build_basic_header(username, password):
    creds = b'%b:%b' % (username, password)
    return 'Basic ' + base64.encodebytes(creds).decode("ascii").replace('\n', '')
```

Then there is couple places in mjpgclient.py which needs to be fixed:
https://github.com/motioneye-project/motioneye/blob/11c0f06abd7093fa6e4734e7801e158fb6680c6e/motioneye/mjpgclient.py#L185-L188
https://github.com/motioneye-project/motioneye/blob/11c0f06abd7093fa6e4734e7801e158fb6680c6e/motioneye/mjpgclient.py#L244-L247

one option is to replace % auth_header -> % auth_header.encode() and maybe change %s -> %b

And invalid implementation in below locations. Future add_done_callback includes future parameter into method call with isn't implemented properly. 
https://github.com/motioneye-project/motioneye/blob/11c0f06abd7093fa6e4734e7801e158fb6680c6e/motioneye/mjpgclient.py#L249
https://github.com/motioneye-project/motioneye/blob/11c0f06abd7093fa6e4734e7801e158fb6680c6e/motioneye/mjpgclient.py#L270

I can't investigate those implementation problems right now, pretty bad flu is teasing me :)

lukasig, 2022-05-13:
Thanks @tunkio 
but I am too much of a newbie to understand your reply.
I only know it should not behave the way it does, but I have no idea how to fix it :)

tunkio, 2022-05-13:
Ok :) 

I wrote that response hoping that someone have time to check that out, I can also look the problem when this damn fever goes away.

tunkio, 2022-05-17:
If you @lukasig are able to test some changes from draft pull request https://github.com/motioneye-project/motioneye/pull/2486

Maybe easiest is to copy those three changed files from pr and run systemctl restart motioneye

No worries if you can't. I need to do that my self too.

lukasig, 2022-05-19:
hmm... not sure i understand
are 3 files changed or three lines changed in one file (mjpgclient.py)?

tunkio, 2022-05-19:
3 different files
https://github.com/motioneye-project/motioneye/blob/ec6edb6f58cfac573c614dd22406a4029b4b48bf/motioneye/mjpgclient.py
https://github.com/motioneye-project/motioneye/blob/ec6edb6f58cfac573c614dd22406a4029b4b48bf/motioneye/utils/__init__.py
https://github.com/motioneye-project/motioneye/blob/ec6edb6f58cfac573c614dd22406a4029b4b48bf/motioneye/utils/mjpeg.py

raw versions
https://raw.githubusercontent.com/motioneye-project/motioneye/ec6edb6f58cfac573c614dd22406a4029b4b48bf/motioneye/mjpgclient.py
https://raw.githubusercontent.com/motioneye-project/motioneye/ec6edb6f58cfac573c614dd22406a4029b4b48bf/motioneye/utils/__init__.py
https://raw.githubusercontent.com/motioneye-project/motioneye/ec6edb6f58cfac573c614dd22406a4029b4b48bf/motioneye/utils/mjpeg.py

And no problem if you can not test it. I'll try find some time to do more testing. This is not fully fixed version. In my test Basic Auth works.

zagrim, 2024-03-17:
Just figured out the reason for the streaming URL auth not working in 0.43. It's not even about encoding (even though that had been fixed later on), but Motion's incapability to understand hashed streaming password. See my comment [here](https://github.com/motioneye-project/motioneye/issues/2813#issuecomment-2002362864).
There's nothing to be done here, so I'll close this one and if there are ideas on fixing this (either by removing hashing from normal user password altogether or something better), let's continue the discussion in #2813 .

closed 2024-03-17 by zagrim