motioneye-project/motioneye #2493: Not possible to delete files (fotos, videos) from connected remote motioneye camera

sf-206, 2022-05-20:
It is not possible to delete files (fotos, videos) from remote motioneye camera via main motioneye client.

Error message:
Failed to delete movie from http://192.168.2.39:8765/config/2: authentication failed.

MichaIng, 2022-05-20:
Could you give some more details about the issue:
- Which motionEye version are you using, 0.42.0 (Python 2 stable) or 0.43.0 (Python 3 dev)?
- You mean deleting media (stored on the motionEye server) recorded from a remote/network camera fails?

sf-206, 2022-05-23:
Sorry for being so imprecise.
- 0.43.0 (Python 3 dev) latest version
- I have added a remote motion camera to my main client. (add camera - remote motion eye camera). It is possible to access stored videos / fotos via main client. Which means I can view them. But it is not able to delete the files
- When I log in to the remote motion eye client directly I am able to delete the files

I hope this explains the situation

MichaIng, 2022-05-23:
Does the local and/or the remote motionEye show related error logs?
```sh
journalctl -u motioneye
```

sf-206, 2022-05-23:
This is from the Motion Eye Client where the Remote Motion Eye Client was added
May 23 18:13:23 raspberrypi meyectl[419]:    ERROR: failed to delete file 2022-05-22/16-12-56.mp4 of remote camera 2 on http://192.168.2.39:8765/config/2: authe>

This is from the  Remote Motion Eye Client
May 23 18:13:23 raspberrypi meyectl[453]:  WARNING: 403 POST /movie/2/delete/2022-05-22/16-12-56.mp4?_username=admin&_admin=true&_signature=4f0a6d1eca3bed94a7fe>

MichaIng, 2022-05-23:
> 403 POST

An authentication error. I guess this will fix it: #2486